slimbootloader/slimbootloader #1: Doesn't even build in Docker.

AdrianKoshka, 2018-09-18:
```
Applying QEMU FSP patch ...
Applying: Build QEMU FSP 2.0 binaries
error: gpg failed to sign the data
fatal: failed to write commit object
Traceback (most recent call last):
  File "/mnt/storage/tmp/slimbootloader/BootloaderCorePkg/Tools/PrepareFspBin.py", line 193, in <module>
    sys.exit(Main())
  File "/mnt/storage/tmp/slimbootloader/BootloaderCorePkg/Tools/PrepareFspBin.py", line 185, in Main
    BuildFspBins (qemu_repo_dir, sbl_dir, platform, target)
  File "/mnt/storage/tmp/slimbootloader/BootloaderCorePkg/Tools/PrepareFspBin.py", line 141, in BuildFspBins
    Fatal ('Failed to apply QEMU FSP patch !')
  File "/mnt/storage/tmp/slimbootloader/BootloaderCorePkg/Tools/PrepareFspBin.py", line 21, in Fatal
    raise Exception (msg)
Exception: Failed to apply QEMU FSP patch !
Traceback (most recent call last):
  File "BuildLoader.py", line 1199, in <module>
    main()
  File "BuildLoader.py", line 1196, in main
    args.func(args)
  File "BuildLoader.py", line 1155, in cmd_build
    Build(board).build()
  File "BuildLoader.py", line 1016, in build
    self.pre_build()
  File "BuildLoader.py", line 901, in pre_build
    raise Exception  ('Failed to checkout FSP binaries !')
Exception: Failed to checkout FSP binaries !
```

AdrianKoshka, 2018-09-18:
To add to this, at least on the version of docker fedora 28 has, `docker exec -w /work sbl python BuildLoader.py build qemu` doesn't even work...there's no `-w`.

sbl-support, 2018-09-18:
Hi Adrian,

I'd suggest you starting setting up your build environment without using Dockers. The issue you are seeing is known due to the build script is downloading QEMU code outside SBL source directory. We are considering to improve this.

We have not tested build on Fedora 28.

Thanks,
Aaron

AdrianKoshka, 2018-09-19:
Well...perhaps don't write documentation you know that doesn't work next time. I did build it outside of docker and it did work.

closed 2018-09-19 by AdrianKoshka