slimbootloader/slimbootloader #14: A file encode issue in the ApollolakePkg Heci Libray header file.

Eathonhsu, 2018-10-06:
Dear Slim boot loader team,

There is a compile issue when use Visual studio 2015, I tried to bring up an Apollo lake platform.

- Silicon\ApollolakePkg\Include\Library/HeciLib.h: warning **C4819**: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss

I found the root-cause in the _HeciLib.h_. (the change patch is in the attachment)

[HeciLib_Patch.zip](https://github.com/slimbootloader/slimbootloader/files/2452874/HeciLib_Patch.zip)

Sincerely,
Ethan Hsu

mauricema, 2018-10-08:
Hi, Ethan, 

Thank you for reporting this issue and providing a patch.  
Our current process uses pull request to submit code changes. 
Are you willing to submit a pull request so that it can be merged directly ?  

Thanks
Maurice

Eathonhsu, 2018-10-08:
Dear Maurice,

OK, I will submit a pull request soon.

Sincerely,
Ethan

mauricema, 2018-10-08:
Thank you,  Ethan!   :)

huang-jin-intel, 2018-10-12:
Fixed in #16.

closed 2018-10-12 by huang-jin-intel