slimbootloader/slimbootloader #17: Does SLIMBOOT Support Uefipayload and Apollo Lake combination

eddylu99, 2018-10-12:
Reference [](https://slimbootloader.github.io/how-tos/boot-windows.html) for Slimboot booting windows.
and SlimBootloaderIntegration.txt in [](https://github.com/tianocore/edk2-staging/tree/UEFIPayload).

'MinnowBoard3' and 'Qemu' is supported. 
Does Apollo Lake also support integration and how to integrate it?

huang-jin-intel, 2018-10-12:
Yes - Slim Bootloader supports Apollo Lake.  You can integrate UEFIPayload according to the documentation you mentioned above. We have verified on Leaf Hill or similar CRBs (https://slimbootloader.github.io/supported-hardware/apollo-lake-crb.html). If you have further questions, use our mailing list https://lists.01.org/mailman/listinfo/sbl-devel. You are welcome to submit real issues on github 'issues' page too.

eddylu99, 2018-10-12:
When I register on mailing list https://lists.01.org/mailman/listinfo/sbl-devel, I got below messages

Sbl-devel Subscription results
The hidden token didn't match. Did your IP change?

Does the privacy only for Intel IP or any other reason?

thx && br

huang-jin-intel, 2018-10-12:
Interesting. This mailing list is public for anyone. Can you try on a different computer or outside of firewall?

closed 2018-10-16 by eddylu99