slimbootloader/slimbootloader #83: Compilation failed

tomasbw, 2019-02-06:
EfiUtilityMsgs.c:484:9: error: ‘strncat’ output may be truncated copying between 0 and 511 bytes from a string of length 511 [-Werror=stringop-truncation]
         strncat (Line, Line2, MAX_LINE_LEN - strlen (Line) - 1);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
EfiUtilityMsgs.c:469:9: error: ‘strncat’ output may be truncated copying between 0 and 511 bytes from a string of length 511 [-Werror=stringop-truncation]
         strncat (Line, Line2, MAX_LINE_LEN - strlen (Line) - 1);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
EfiUtilityMsgs.c:511:5: error: ‘strncat’ output may be truncated copying between 0 and 511 bytes from a string of length 511 [-Werror=stringop-truncation]
     strncat (Line, Line2, MAX_LINE_LEN - strlen (Line) - 1);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

mauricema, 2019-02-06:
Thank you for reporting issue.
Can you provide more details on your build environment?  For example,  OS, compiler version, etc.
This file is part of the  EDKII base tool,  and it is the same as EDKII repo.  I guess it might be something to do with the supported compiler version.

tomasbw, 2019-02-06:
gcc (GCC) 8.2.1 20181215 (Red Hat 8.2.1-6)
Copyright (C) 2018 Free Software Foundation, Inc.
Fedora29

tomasbw, 2019-02-06:
I've checked EDKII it forces  -Wno-stringop-truncation

tomasbw, 2019-02-06:
../BaseTools/Source/C/Makefiles/header.makefile
ifeq ($(DARWIN),Darwin)
# assume clang or clang compatible flags on OS X
BUILD_CFLAGS = -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g
else
BUILD_CFLAGS = -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g
endif

mauricema, 2019-02-06:
Thank you for the information.  We will take a look into it.

myguitar, 2019-02-06:
I am using gcc 8.2.0 on Ubuntu18, but there is no compile error. Can you give more information here?
gcc -v
uname -a

It looks you are compiling QemuFsp. Can you compile a single file as below?

cd QemuFsp/BaseTools/Source/C/Common
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g -O2  EfiUtilityMsgs.c -o EfiUtilityMsgs.o

Please share your information and the result.

mborgerson, 2019-02-07:
Confirmed this issue building BaseTools can be duplicated on Fedora 29 when building BaseTools @ f9397b39a4aa2bd34fc0f04b1624a1aa72b68edf.

It looks like [upstream EDK2](https://github.com/tianocore/edk2) @ bac50df can build BaseTools without issues. After searching through EDK2 history--appears resolved by this commit: [1d212a8](https://github.com/tianocore/edk2/commit/1d212a83df0eaf32a6f5d4159beb2d77832e0231#diff-81141d6762d92d3c026f14cc02a9d383). Probably best to sync with upstream EDK2 soon.

rprangar, 2019-02-07:
Is it possible to align with the build environment documented here - https://slimbootloader.github.io/getting-started/build-host-setup.html

mauricema, 2019-02-08:
Sure.   Meanwhile we also plan to merge the EDK2 fix into Slim Bootloader.

myguitar, 2019-02-08:
@tomasbw Please feel to apply

myguitar, 2019-02-13:
@tomasbw To follow up this issue, https://github.com/slimbootloader/slimbootloader/pull/91 has been merged. Can you try the latest code?

tomasbw, 2019-02-13:
The compilation works, there are some issues with the quality of the patches, but you probably just brought them from the EDKII. I left some comments.

myguitar, 2019-02-13:
@tomasbw Right, it was brought from EDKII and we keep the original file format. We can use dos2unix, but it will convert entire file and make hard to diff the change.

myguitar, 2019-02-19:
@tomasbw Let me close this issue. Please let us know if you have any other issues or feel free to open your concern. Thanks.

closed 2019-02-19 by myguitar

tomasbw, 2019-02-20:
Ok