slimbootloader/slimbootloader #110: PCIe base not passed from SBL to Payload

wvervoorn, 2019-03-07:
Hello,

At this moment it is important to have identical configuration between SBL and UEFI Payload for the PCIe base. As SBL knows the PCIe base why not add this to the information that is passed between SBL and UEFI Payload. There hardly isn't an effort involved and it prevent configuration issues.

Regards,
Wim

mauricema, 2019-03-07:
Hi, Wim,

Thank you for bringing this question.
What you mentioned is very reasonable.   In SBL, we try to utilize the standard infrastructure to pass information from SBL to payload.  One of them is HOB, which is same as EDKII.   Another one is ACPI table, which is an industry standard already.   
Specifically for PCIe base,  since it is already described in ACPI MCFG table,   we think it should be fine for payload to get the information from there.    In UEFI Payload we should be able to set the PcdPciExpressBaseAddress PCD accordingly using the base passed from ACPI MCFG.

Will this work for you ?   Do you have any other thoughts ? 

Thanks
Maurice

wvervoorn, 2019-03-08:
Hello Maurice,

Thanks for the quick response. This solution sounds good to me. This approach minimizes the changes required in the payload.

Best Regards,
Wim Vervoorn

Eltan B.V.
Ambachtstraat 23
5481 SM Schijndel
The Netherlands

T : +31-(0)73-594 46 64
E : wvervoorn@eltan.com
W : http://www.eltan.com<http://www.eltan.com/>

[cid:image001.jpg@01D4D59F.9A078F10]<https://www.embedded-world.de/en/ausstellerprodukte/embwld19/exhibitor-38826722/eltan-b-v>
"THIS MESSAGE CONTAINS CONFIDENTIAL INFORMATION. UNLESS YOU ARE THE INTENDED RECIPIENT OF THIS MESSAGE, ANY USE OF THIS MESSAGE IS STRICTLY PROHIBITED. IF YOU HAVE RECEIVED THIS MESSAGE IN ERROR, PLEASE IMMEDIATELY NOTIFY THE SENDER BY TELEPHONE +31-(0)73-5944664 OR REPLY EMAIL, AND IMMEDIATELY DELETE THIS MESSAGE AND ALL COPIES."

mauricema, 2019-03-08:
Great!  Thanks for the update.   We will close this issue since this information can be available in Payload with current implementation.

closed 2019-03-08 by mauricema