slimbootloader/slimbootloader #150: Qemu FSP -  IntelFspPkg/Tools/SplitFspBin.py "Fd/Fv/Ffs<->FSPImage" parsing bug

IkeBroflovskiForPresident, 2019-05-07:
The FirmwareDevice class method ParseFsp(self) (IntelFsp2Pkg/Tools/SplitFspBin.py), called when building slim bootloader for Qemu fails to parse the built QemuFsp Fd file correctly.

The Fd (built using "python BuildLoader.py build qemu") correctly contains 3 EFI Fvs, one each to eventually be "split" into FSP_S.bin, FSP_M.bin, and FSP_T.bin. The FSP Information Headers all appear to be in the correct places within the Fd (e.g. Silicon/QemuSocPkg/FspBin/FspRel.bin) "equivalent to" each EFI Fv (S at 0x00, M at 0x15000, T at 0x37000).

Subsequent processing by postbuild in BuildLoader.py expects those 3 FSPImage files, however there are only 2 FSPImage files created by the split, with the M FSPImage improperly containing the T Fv in addition to its' M Fv.

The first indication of trouble is an alignment warning from the BPDG tool:

`VPD input data file:     = Build/QemuFspPkg/RELEASE_GCC5/FV/34686CA3-34F9-4901-B82A-BA630F0714C6.txt
VPD output map file:     = Build/QemuFspPkg/RELEASE_GCC5/FV/34686CA3-34F9-4901-B82A-BA630F0714C6.map
VPD output binary file:  = Build/QemuFspPkg/RELEASE_GCC5/FV/34686CA3-34F9-4901-B82A-BA630F0714C6.bin
BPDG...
Build/QemuFspPkg/RELEASE_GCC5/FV/34686CA3-34F9-4901-B82A-BA630F0714C6.txt(...): warning: The offset value of PCD gQemuFspPkgTokenSpaceGuid.Reserved is not 8-byte aligned!
BPDG...
Build/QemuFspPkg/RELEASE_GCC5/FV/34686CA3-34F9-4901-B82A-BA630F0714C6.txt(...): warning: The offset value of PCD gQemuFspPkgTokenSpaceGuid.Reserved is not 8-byte aligned!
BPDG...
Build/QemuFspPkg/RELEASE_GCC5/FV/34686CA3-34F9-4901-B82A-BA630F0714C6.txt(...): warning: The offset value of PCD gQemuFspPkgTokenSpaceGuid.Reserved1 is not 8-byte aligned!
- Vpd pcd fixed done! -`

I am not sure if I need to investigate this warning "upstream" or if the BPDG tool was trying to tell me it "fixed" the alignment problem with the upstream inputs and it is OK to move forward.

mauricema, 2019-05-07:
The warning message you listed should not be a real concern.  I just tried to build SBL from a fresh clone in Ubuntu, and I was able to build QEMU successfully.   Could you please attach your full build log ?
Yes, SBL expects 3 FSP components (FSP-T/M/S) after splitting.  If your build only generate two of them, then it is going to be an issue.  
To isolate this issue, you can run the script separated ?
python IntelFsp2Pkg/Tools/SplitFspBin.py split -f Silicon/QemuSocPkg/FspBin/FspRel.bin
It should generate d files:  FspRel_S/M/T.bin.
If it is not the case, please run:
python IntelFsp2Pkg/Tools/SplitFspBin.py info -f Silicon/QemuSocPkg/FspBin/FspRel.bin
to parse the FSP info to see if the FSP is generated properly.

mauricema, 2019-05-15:
Do you still have issue with it?   Any updates?   
If there is no update by this Friday,  we plan to close this issue.

mauricema, 2019-05-20:
Close this issue since it has no further updates from submitter.

closed 2019-05-20 by mauricema