slimbootloader/slimbootloader #151: ConfigEditor: Unrecognized variable 'iasImage'

mschuetterle, 2019-05-07:
Fail to change the delta configuration file using the ConfigEditor.

Steps:

1. Start ConfigEditor.
2. Load Config DSC file: Platform/ApollolakeBoardPkg/CfgData/CfgDataDef.dsc
3. Load Config Changes from Delta file: Platform/ApollolakeBoardPkg/CfgData/CfgData_Ext_Gpmrb.dlt
4. Change values in OS Boot Options.

[SBL_bootImage_config.log](https://github.com/slimbootloader/slimbootloader/files/3152228/SBL_bootImage_config.log)

mauricema, 2019-05-07:
Thank you for reporting this issue, Markus!
Current tool uses the default value to determine the data type,  number or string.   For Boot option "image0 name", the default is set to 0 in CfgData_BootOption.dsc, which makes it think the config item is number instead of string.  If you change the default from 0 to empty string '' in CfgData_BootOption.dsc,  the issue will be resolved.

EX:
  # AOS boot from EMMC # |ImgType|flag|DevType|DevNum|HwPart|SwPart|FsType|  image0 name  | sw0 part| sw0 lba|  image1 name  | sw1 part| sw1 lba|  image2 name  | sw2 part| sw2 lba|
  # eMMC boot P1
  # !BSF SUBT:{OS_TMPL:0 :  0    :  0 :   2   :  0   :   0  :    0 :    0 :'iasimage.bin' :       0 :      0 :     0         :     0   :  0     :     0         :     0   :   0    }
  # eMMC boot P2
  # !BSF SUBT:{OS_TMPL:1 :  0    :  0 :   2   :  0   :   0  :    0 :    3 :      ''       :       1 :      0 :     0         :     0   :  0     :     0         :     0   :   0    }
  # SATA boot
  # !BSF SUBT:{OS_TMPL:2 :  0    :  0 :   0   :  0   :   0  :    0 :    0 :'iasimage.bin' :       0 :      0 :     0         :     0   :  0     :     0         :     0   :   0    }
  # USB boot
  # !BSF SUBT:{OS_TMPL:3 :  0    :  0 :   5   :  0   :   0  :    0 :    0 :'iasimage.bin' :       0 :      0 :     0         :     0   :  0     :     0         :     8   :   0    }
  # SPI boot (for fastboot recovery)
  # !BSF SUBT:{OS_TMPL:4 :  4    :  0 :   7   :  0   :   0  :    0 :    3 :      ''        :       0 :      0 :     0         :     0   :  0     :     0         :     0   :   0    }

We will take a look to see how to enhance the tool to make it handle this automatically.

mauricema, 2019-05-07:
Markus, 
Pull request #152 was created to address issue #151.
Could you please verify if this fixes your issue ?

mschuetterle, 2019-05-08:
If have successfully verified the Pull request #152

PR #152 (2019-05-08): [APL] Fix ConfigEditor error on changing Boot Option
To ensure ConfigEditor can handle the configuration data type properly,
the default value needs to be valid. For string type configuration item,
the value should start with single quote. The APL default value for OS
image name is set to number 0 instead of '' to indicate empty string.
This patch corrected this and fixed #151.
commit f0b92d58 (2019-05-08): [APL] Fix ConfigEditor error on changing Boot Option

To ensure ConfigEditor can handle the configuration data type properly,
the default value needs to be valid. For string type configuration item,
the value should start with single quote. The APL default values for OS
image name is set to number 0 instead of '' to indicate empty string.
This patch corrected this and fixed #151.
closed 2019-05-08 by mauricema