slimbootloader/slimbootloader #208: [APL] Boot failure in SBL Stage2

vegnish, 2019-06-21:
Boot process halts on STAGE_2.

Debug log as below:
...
ACPI Ret: Success
Loading Payload ID 0x49464555
Error: Unsupported

STAGE_2: System halted!

I attached the complete log file just in case.

And, on a side note, I reverted to the version before PR #198 and there wasn't an issue.

[2019.06.21-180126-new_sbl.log](https://github.com/slimbootloader/slimbootloader/files/3314194/2019.06.21-180126-new_sbl.log)

mauricema, 2019-06-21:
Thank for reporting this issue.  
There is a recent change to enable GPIO payload ID selection.  And this was enabled for Intel CRB boards by default, including LeafHill, OxbowHill and JuniperHill.   When this is enabled,  the GPIO will control the PayloadId to be either OsLoader (Level Low) or UEFI payload (Level High).   In your case, it seems you only have one payload (OsLoader).  And the GPIO tells SBL to load UEFI payload (Payload ID 0x49464555), but it could not fine it.  So you got "Error: Unsupported".
There are several ways to address it.  
Option1:   Set a jumper (Pin10/Pin12 on J6B2) for your LeafHill board  to let SBL load OsLoader instead of UEFI payload. 
Option2:   Disable AUTO payload ID selection feature by changing the PayloadId directly in your DLT  Platform\ApollolakeBoardPkg\CfgData\CfgData_Int_LeafHill.dlt.   Remove the line  GEN_CFG_DATA.PayloadId  | 'AUTO'.
Option3:   Add UEFI payload in as described 
https://slimbootloader.github.io/tutorials/ex8.html?highlight=uefi.  Then it will try to boot to UEFI payload instead of halt.
Please let us know if these options work for you.  

On the other side, we can also change the default behavior to boot OsLoader instead of UEFI payload.

closed 2019-06-21 by mauricema

shehamb, 2020-05-25:
> Thank for reporting this issue.
> There is a recent change to enable GPIO payload ID selection. And this was enabled for Intel CRB boards by default, including LeafHill, OxbowHill and JuniperHill. [...]

Hello, 
I am having similar issue. And i am using the latest SBL codebase. 
Our hardware is based off Leaf Hill CRB. As per your comments above, i tried option#1 and option#2. But i am getting this error:
--------------------------------------------------------------------------
PlatformName: LeafHill
BootPolicy : 0x00000000
Memory Init
Looking for MRC saving data
MRC data NOT found
Load memory parameters from CfgData.
BRC3 : Setting the profile to (0x0B)
Load general parameter from CfgData.
Call FspMemoryInit ... Success
Loader global data @ 0x7ABBFD70
Load page table from memory @ 0x7A074000
Remapping Stage to 0x7A077000
Switch to memory stack @ 0x7ABFFF00
Stage1 stack: 0x2000 (0xEF0 used)
Stage1 heap: 0xE000 (0x4CD4 used)
Call FspTempRamExit ... Success
CSE: MBP Data HOB not valid!
MRC: [v0.56.89.24]
Memory FSP @ 0x7AC00000
Memory TOP @ 0x7A074000
Loading Stage2 error - Unsupported !
Failed to load Stage2!
STAGE_1B: System halted!
--------------------------------------------------------------------------------

Regards, 
Mahesh.

shehamb, 2020-05-25:
[sbl-stage2-failure.log](https://github.com/slimbootloader/slimbootloader/files/4675569/sbl-stage2-failure.log)

mauricema, 2020-05-26:
From the log, it is totally a different issue.  
The original issue was that it could not find the correct Payload ID and failed to load Payload.  However, in your case, it cannot even load Stage2. 

It is better to start with a new issue instead.

mauricema, 2020-05-26:
From your log attached, the following errors were seen,  Is your STAGE1_DATA_SIZE in BoardConfig.py too small ?
Append public key hash into store: Out of Resources
Load EXT CFG Data @ 0xFEF7BBE4:0xFFFFFFFF ... Not Found

Also how did you stitch your IFWI?  It seems Stage2 could not be located in IFWI image.   
For the verification, did you set  your SIGN_HASH_TYPE in BoardConfig.py? 

Can you post your Platform/ApollolakeBoardPkg/BoardConfig.py file and  BootloaderCorePkg/Platform.dsc ?

shehamb, 2020-05-27:
I haven't changed any settings in BoardConfig.py.  STAGE1_DATA_SIZE is
0xE000 in  BoardConfig.py and  Platform.dsc. I am enclosing these files for
your reference.

And i am using FIT to change few settings in the final SBL BIOS file. These
settings are:

CPU Straps - PUNIT
-------------------
Set Soft SVID Enable to “Enabled”
Set all Rail n Alert Polling Enables to “Enabled”
Set all Rail n SVID IDs to “0x00000000”

Platform Configuration
----------------------
Set PMIC/VR (PmicVrConfig) to “I2C VR – TI TPS650941”
Set Persistent PRTC Backup Power (PrtcBackupPower) to ”None”
Set IAI2CVRRdWrInValidAddrRange_0 to “0x5E275E00”

Platform Protection
----------------------
Clear OEM Public Key Hash to “all zeros (32 bytes)”
Clear Key Manifest ID to “0”
Set Boot Profile to “Boot Guard Profile 0 – Legacy”

Integrated Sensor Hub
----------------------
Set Integrated Sensor Hub Supported to “No”

And then i generate build the image and use it to flash on our hardware,
which is similar to Leaf Hill CRB except for the above mentioned settings.

## @file
# This file is used to provide board specific image information.
#
#  Copyright (c) 2017 - 2020, Intel Corporation. All rights reserved.<BR>
#
#  SPDX-License-Identifier: BSD-2-Clause-Patent
#

##
# Import Modules
#
import os
import sys

sys.dont_write_bytecode = True
sys.path.append (os.path.join('..', '..'))
from BuildLoader import FLASH_MAP, BaseBoard, STITCH_OPS
from BuildLoader import IPP_CRYPTO_OPTIMIZATION_MASK, IPP_CRYPTO_ALG_MASK

#
#    Temporary Memory Layout for APL
#
#  FF000000 +--------------------------+
#           |        Stage1B           |
#           |     (Decompressed)       |
#  FEF80000 +--------------------------+
#           |     Stage1 Heap/Stack    |
#  FEF70000 +--------------------------+
#           |        Not Used          |
#           +-------------+------------+
#           |           Free           |
#           |             |------------|
#           +-------------+   MRC NVS  |
#           |             |            |
#  FEF40000 +-  Stage1B  -+------------+
#           |  Compressed |  FSP Mem   |
#  FEF16000 |             +------------+
#           |             |            |
[... 233 lines cut ...]
                    ),
                    ('Stitch_CFGDATA.bin', [
                        ('CFGDATA.bin',  '',     self.CFGDATA_SIZE,   STITCH_OPS.MODE_FILE_PAD, STITCH_OPS.MODE_POS_TAIL)]
                    ),
                    ('Stitch_KEYHASH.bin', [
                        ('KEYHASH.bin',  '',     self.KEYHASH_SIZE,   STITCH_OPS.MODE_FILE_PAD, STITCH_OPS.MODE_POS_TAIL)]
                    ),
                    ('Stitch_OBB.bin', [
                        ('STAGE2.fd',    'Lz4',  self.STAGE2_SIZE,    STITCH_OPS.MODE_FILE_PAD, STITCH_OPS.MODE_POS_TAIL)]
                    ),
                    ('Stitch_IBBM.bin', [
                        ('STAGE1B.fd',   'Lz4',  self.STAGE1B_SIZE,   STITCH_OPS.MODE_FILE_PAD, STITCH_OPS.MODE_POS_TAIL)]
                    ),
                    ('Stitch_IBBL.bin', [
                        ('STAGE1A.fd',   '',     self.STAGE1A_SIZE,   STITCH_OPS.MODE_FILE_NOP, STITCH_OPS.MODE_POS_TAIL)]
                    ),
                    ('Stitch_EPLD.bin', [
                        ('EPAYLOAD.bin', '',    self.EPAYLOAD_SIZE,   STITCH_OPS.MODE_FILE_PAD, STITCH_OPS.MODE_POS_TAIL)]
                    ),
                    ('Stitch_UVAR.bin', [
                        ('UEFIVARIABLE.bin', '',  self.UEFI_VARIABLE_SIZE,  STITCH_OPS.MODE_FILE_NOP, STITCH_OPS.MODE_POS_TAIL)],
                    ),
                    ])

        return img_list

mauricema, 2020-05-27:
I checked the latest SBL code on CRB board, it worked as expected.  So it might be related to your flow with FIT.   Can you attach the stitched final IFWI image ?   Also can you provide your steps on how you did the FIT modification ?  Can you try to use FIT to modify the settings in base IFWI image first, and then use StitchLoader.py to generate the final IFWI image to flash ?

shehamb, 2020-05-27:
So, i altered the IFWI image first and then used  StitchLoader.py to
generate the final IFWI image. This has worked. SBL boots up all the way
through.
Next, i wanted to get the UEFI shell. So, built the SBL again to include
UEFI payload:
*python3 BuildLoader.py build apl -p
"OsLoader.efi:LLDR:Lz4;UefiPld.fd:UEFI:Lzma"*

With this, SBL boots up with UEFI shell. But i don't see the SD card
mapping (fs0:, fs1: and so on).
-----------------------------------------------------------

UEFI Interactive Shell v2.287477C2-69C7-11D2-8E39-00A0C969723B 78BDC6A0
EDK IIlProtocolInterface: 752F3136-4E16-4FDC-A22A-E5F46812F4CA 78BDBE18
UEFI v2.70 (EDK II, 0x00010000)008-7F9B-4F30-87AC-60C9FEF5DA4E 78A993C0
Mapping table
     BLK0: Alias(s):
          PciRoot(0x0)/Pci(0x1C,0x0)/eMMC(0x0)/Ctrl(0x0)
     BLK1: Alias(s):
          PciRoot(0x0)/Pci(0x1C,0x0)/eMMC(0x0)/Ctrl(0x1)
     BLK2: Alias(s):
          PciRoot(0x0)/Pci(0x1C,0x0)/eMMC(0x0)/Ctrl(0x2)
Press ESC in 2 seconds to skip startup.nsh or any other key to continue.
Shell>

---------------------------------------------------------------------

mauricema, 2020-05-28:
Please ensure your SD card related GPIO are configured properly and the SD card power is applied correctly.  It might also be related to UEFI SD/eMMC driver initialization flow.

shehamb, 2020-05-28:
UEFI bios is able to detect SD card on our new hardware.
And for the same code base of APL platform on SBL, i was able to see SD
card accessible through UEFI shell on the UP2 board. So i suppose SD card
power and UEFI SD/eMMC driver initialization flow related issues can be
ruled out.

thau-my, 2020-05-28:
Is there any error related to SD card initialization such as tuning failure during SBL boot? Could you provide the full SBL log?

shehamb, 2020-05-28:
hello,
i don't see any error as such. enclosing the log for your reference.

mauricema, 2020-05-28:
For SD card power control, it is board specific design.  
EX:  on CRB, it is decided by the GPIO 183. The pin level has to be set to 0 in order to power on the SD CARD.  For your board, you need to see which GPIO will control the power and make sure it is set to correct level.

mauricema, 2020-05-28:
Also just want to be clear, you meant SD card (PCI device 0x1B, function 0) , not eMMC (PCI device 0x1C, functin 0) right ?   UP2 board does not have SD card as far as I know.   It only provides eMMC.

shehamb, 2020-05-29:
Hello Maurice,
I am not sure if it could be a SD card power control issue because on the
same hardware, i am able to boot the UEFI bios (with UEFI shell) and see
the SD card mapping.
And yes, my bad. I meant, eMMC on UP2 board.
Regards,
Mahesh.

mauricema, 2020-05-29:
Since it is a different topic.   For SD card issue, I submitted a new issue to track it.
https://github.com/slimbootloader/slimbootloader/issues/729

mauricema, 2020-05-29:
Can you help clarify on your own board,  which one you are using,   SD card or eMMC ?  Please update on issue #729.

PR #209 (2019-06-21): Change default behavior for GPIO payload ID selection
Current default GPIO payload ID selection is UEFI on APL platform
since the GPIO pin is high without any jumper set. This patch
changed the default paylod ID to OsLoader when the GPIO is set to
high.

It fixed #208.
commit dbc7bd33 (2019-06-21): Change default behavior for GPIO payload ID selection

Current default GPIO payload ID selection is UEFI on APL platform
since the GPIO pin is high without any jumper set. This patch
changed the default paylod ID to OsLoader when the GPIO is set to
high.

It fixed #208.