slimbootloader/slimbootloader #218: Support NHLT table for audio device on APL-UP2

binbinwu1, 2019-06-28:
NHLT table is not integrated in ACPI table by SBL. NHLT table has audio device specific informatinon, which is needed for audio device to work on APL-UP2.
Could SBL intergrate NHLT table on APL-UP2 to support audio?

rprangar, 2019-06-28:
Hi

Yes, this is planned in the near future. Meanwhile, you may consider using SBL support for loading NHLT ACPI from IAS. Will provide an update on NHLT support has been added.

Thanks
Ravi

rprangar, 2019-07-03:
Hi

A PR has been submitted and review is in progress now - https://github.com/slimbootloader/slimbootloader/pull/220

I am planning to close this issue. 

Thanks
Ravi

closed 2019-07-03 by rprangar

binbinwu1, 2019-07-04:
> Hi
> 
> A PR has been submitted and review is in progress now - #220
> 
> I am planning to close this issue.
> 
> Thanks
> Ravi

Thanks