slimbootloader/slimbootloader #343: [WHL] Windows boot stuck on spinning circle with open source UEFI payload

mauricema, 2019-10-03:
Using latest SBL and lastest UEFI payload from open source,  Windows boot stuck on spinning circle screen forever.

closed 2019-10-04 by mauricema

bigchenla, 2020-06-17:
Hi, I use latest SBL, Windows boot still stuck on spinning circle screen, please refer to the log
[putty20200617_141306.txt](https://github.com/slimbootloader/slimbootloader/files/4790660/putty20200617_141306.txt)

mauricema, 2020-06-17:
Please specify which board you used,  and which UEFI payload you used.  How did you build the payload.
Also I notice your SBL tree is "7EDCE89FB182DF0E-dirty".  Please tell what you have modified.

bigchenla, 2020-06-17:
Hi Maurice,
I use WHL customized board, and follow https://slimbootloader.github.io/how-tos/boot-windows.html to build UEFI payload on Windows (edk2 repo: commit 42d8be0eaac5e7e109f487d4e241847e815b077a)
I just config some super IO(ITE8786e) registers in stage1A.

mauricema, 2020-06-17:
Do you have a EC controller for your board ?
If you don't have a EC, you will need to set following in your board dlt file CfgDataInt_Whl.dlt.   SILICON_CFG_DATA.ECEnable | 0

bigchenla, 2020-06-18:
After setting ECEnable=0, I can boot to Windows now. Thanks!!

I have another question about that. After adding the UEFI payload, system always reset one time when call FSP  NotifyPhase ReadyToBoot.
```
Call FspNotifyPhase(40) ... 40000003
FSP Requested Reboot ...
```
But I see the comment in Stage2.c, it should not call FSP NotifyPhase ReadyToBoot if I use open sourced UEFI payload.
```
  } else if ((PayloadId == UEFI_PAYLOAD_ID_SIGNATURE) && (UefiSig != 0)) {
    // Current open sourced UEFI payload does not call any FSP notifications,
    // but some customized UEFI payload will. The 1st DWORD in UEFI payload image
    // will be used to indicate if it will handle FSP notifications.
    CallBoardNotify = FALSE;
```
I try to modify the first byte of UEFIPAYLOAD.fd, let CallBoardNotify = FALSE. System can boot successfully just once.

mauricema, 2020-06-18:
FSP notifications need to be called before OS launch.  Or it will cause security/function related concerns.   Although it can still boot, but lots of settings might not have been set properly if ReadyToBoot and EndOfFirmware notification are skipped.
If FSP requests a reboot, there must be a reason.  EX:  Soft strap in FIT does not match the FSP UPDs. I think you need to figure out why the reset was requested by FSP, and do necessary adjustment.  You might want to contact your Intel representative for FSP related support.  Also you can try to get a debug FSP to collect more information on this reset.

bigchenla, 2020-06-19:
I see, thanks for your help!

mauricema, 2020-06-19:
You are very welcome !  :)

PR #344 (2019-10-04): Fix logic for open source UEFI payload check
There is an issue in current SBL Stage2 code to check if the UEFI
payload is built from open source. It was done by checking the
1st DWORD using Dst[0]. However, during the FV loading, the value
at Dst[0] might have been changed since LoadFvImage() can move the
FV to a new location. This patch cached Dst[0] before calling
LoadFvImage() so that it can always get the oringal value.

It fixed #343.
commit 5571d6d0 (2019-10-04): Fix logic for open source UEFI payload check

There is an issue in current SBL Stage2 code to check if the UEFI
payload is built from open source. It was done by checking the
1st DWORD using Dst[0]. However, during the FV loading, the value
at Dst[0] might have been changed since LoadFvImage() can move the
FV to a new location. This patch cached Dst[0] before calling
LoadFvImage() so that it can always get the oringal value.

It fixed #343.