slimbootloader/slimbootloader #390: [APL] INT34D2 entry is missing in DSDT

umaplehurst, 2019-10-16:
A device entry for `INT34D2` is missing under `Platform/ApollolakeBoardPkg/AcpiTables/Dsdt/Platform.asl`. The ACPI entry is necessary for the `intel_pmc_ipc` driver to load up automatically on Linux.

myguitar, 2019-10-17:
@umaplehurst Thanks for opening this issue. We will check and add the missing device. By the way, what APL board are you using and what Linux are you using for your board?

umaplehurst, 2019-10-17:
Thanks! This is on a custom board, based on Leaf Hill CRB, kernel v4.14.

closed 2019-10-17 by myguitar

umaplehurst, 2019-11-21:
I've done a little bit more testing:

What works: Linux can successfully use the watchdog, all good.
What doesn't work: sending IPC commands to the PMC doesn't work, the error from the Linux driver is "IPC timed out". I don't get this problem with the official IFWI CRB firmware, v227.51, sending commands to the PMC works fine there.

Please let me know if you'd like me to open this as a fresh bug instead.

myguitar, 2019-11-21:
> I've done a little bit more testing:
>
> What works: Linux can successfully use the watchdog, all good.
> What doesn't work: sending IPC commands to the PMC doesn't work, the error from the Linux driver [...]

@umaplehurst Thanks you for more testing. Please open this as a fresh issue and let us know test steps such as Linux sys node command. We will take a look.

PR #402 (2019-10-17): [APL] Expose IPC1 device to OS
This patch adds IPC1 device to DSDT and exposes it to OS.
The behavior can be controlled by Config Data.
- Silicon Setting > Device Enable/Disable > Control 1 > IPC1 Enable/Disable

It also fixed #390.
commit b7df78d4 (2019-10-17): [APL] Expose IPC1 device to OS (#402)

This patch adds IPC1 device to DSDT and exposes it to OS.
The behavior can be controlled by Config Data.
- Silicon Setting > Device Enable/Disable > Control 1 > IPC1 Enable/Disable

It also fixed #390.