slimbootloader/slimbootloader #392: [APL] Update FSP to MR6?

umaplehurst, 2019-10-16:
Using the standard APL FSP MR5 VBT binary, I'm finding that Linux seems to have problems registering one of the DDI outputs on APL. This is seen in the kernel boot log:

```
[drm:lspcon_init [i915]] *ERROR* Failed to probe lspcon
[drm:intel_ddi_init [i915]] *ERROR* LSPCON init failed on port B
```

It looks like the APL FSP MR5 release has a bug in the VBT binary, "OnBoard LSPCON for HDMI 2.0" for EFP1 needs to be set to [Disabled] instead of [Enabled].

I can see that this has been fixed in the MR6 release of FSP for APL, as there is a modified Vbt.bin file, with this changed around, so perhaps slimbootloader should now be updated to use MR6?

mauricema, 2019-10-17:
Yes, we noticed that issue previously.  Since MR6 is available now on github,  we should just switch to use the latest one.   If you are willing to contribute,  you are more than welcome to submit patch to switch to FSP MR6.

umaplehurst, 2019-10-17:
Sounds good. Sure, I can submit a patch. I haven't done any heavy testing but can't see anything obviously wrong, things booted up just fine for me with MR6.

closed 2019-10-17 by umaplehurst