slimbootloader/slimbootloader #393: [APL] Can't stitch image when OEM public key is not present

umaplehurst, 2019-10-16:
I get the following error when giving `StitchLoader.py` an input image which does not have an OEM public key:

```
Patching Slim Bootloader Flash Map table ...
Traceback (most recent call last):
  File "Platform/ApollolakeBoardPkg/Script/StitchLoader.py", line 648, in <module>
    ret = create_ifwi_image (args.ifwi_in, args.ifwi_out, args.bios_out, args.plat_data, args.non_redundant, stitch_dir)
  File "Platform/ApollolakeBoardPkg/Script/StitchLoader.py", line 551, in create_ifwi_image
    patch_flash_map (ifwi_data, platform_data)
  File "Platform/ApollolakeBoardPkg/Script/StitchLoader.py", line 429, in patch_flash_map
    raise Exception("Cannot locate component '%s' in BPDT !" % path)
Exception: Cannot locate component 'IFWI/BIOS/BP0/BPDT/BpdtCseBup/oem.key' in BPDT !
```

My workaround is to comment out the `OEMKEY.bin` entry from `BoardConfig.py`. Given that at the moment things only work anyway with Boot Guard disabled, according to the documentation, this should be a non-fatal issue? Or is the OEM key used for any other important things? FIT seems to be happy enough creating an image that contains no OEM key.

sagraw2, 2019-10-17:
OEMKey manifest is required if OEM wants any of these things:
- enable BootGuard
- use OEM Debug unlock tokens
- replace any of the Intel provided binaries (such as iUni, aDSP, ISH)
If none of the above is used, there is no need to bundle OEMKey manifest within stitched firmware image.

Let me investigate a little bit more on why StitchLoader.py is mandating the presence of it and will update here later.

rprangar, 2019-10-23:
Fixed.

closed 2019-10-23 by rprangar