slimbootloader/slimbootloader #394: [APL] Wake-on-LAN functionality is missing?

umaplehurst, 2019-10-16:
I don't have a Leaf Hill CRB to hand right now to compare behavior, but I know that this works fine in the older `edk2-platforms` (`devel-IntelAtomProcessorE3900`) firmware stack, which has the following code:

```
    IoOr32 (mAcpiBaseAddr + R_ACPI_GPE0a_EN, B_ACPI_GPE0a_EN_PCIE_WAKE0_EN);
    IoOr32 (mAcpiBaseAddr + R_ACPI_GPE0a_EN, B_ACPI_GPE0a_EN_PCIE_WAKE1_EN);
    IoOr32 (mAcpiBaseAddr + R_ACPI_GPE0a_EN, B_ACPI_GPE0a_EN_PCIE_WAKE2_EN);
```

I don't clearly see anything analogous in slimbootloader -- perhaps something should be added to `_PTS` in `Platform.asl` to set the appropriate fields in `GPE0_EN` & make things work out-of-the-box with slimbootloader?

rprangar, 2019-10-18:
WOL was not a planned feature. However it can be enabled by users needing the feature. We welcome any contributions to enable this feature.

closed 2019-10-21 by rprangar

umaplehurst, 2019-11-21:
I've had a look at what the 0.7.1 edk2-platforms Leaf Hill CRB firmware does, where this works, and as far as I can tell, there are only two obvious things in the code: a) set the pinmux for the wake pin, b) set `PCIE_WAKE0_EN` in `ACPI_GPE0a_EN` in the sleep entry callback. I've tried replicating this behavior in SBL (please see attached patch) but it's not working... so either it's wrong somehow, or something else must be missing somewhere in SBL. Any ideas / suggestions?

[lh-crb-wol.txt](https://github.com/slimbootloader/slimbootloader/files/3872280/lh-crb-wol.txt)

mauricema, 2019-11-21:
Thank you for trying to fix it.    I took a look into your patch, and I did see some issue.   
Your current code will do GPIO init in Stage1B GPIO table.  However, that is just a early GPIO table. And the full GPIO table will be programmed in Stage2.
To view/modify the full GPIO configuration, you can try to use ConfigEditor.py in SBL source tree.  You need to load  CfgDataDef.dsc from CFGDATA folder first and then load the CfgData_Int_LeafHill.dlt to see the configuration for LeafHill.   You can make changes to the GPIO config and finally save it back into the same dlt file.   Rebuild SBL, the new configuration should take effect.

To make it simple,  could you please try to add the following at the end of Platform\ApollolakeBoardPkg\CfgData\CfgData_Int_LeafHill.dlt file ?
GPIO_CFG_DATA.GPIO_205_Half1.IoState       | 0x9
GPIO_CFG_DATA.GPIO_205_Half1.WakeCtrl    | 0x1
This will make sure the GPIO to be set correctly as you listed in your patch.

Please let me know if it works.  Thx!

umaplehurst, 2019-11-21:
Many thanks for the review; I spotted the same issue later after posting the previous comment when I double-checked the muxing from Linux. Actually, it looks like the default Stage2 GPIO muxing already in SBL is just fine, there is no need to change the .dlt file; so I just removed the Stage1B GPIO table change in my patch. From Linux, I can confirm that I now have the same GPIO muxing set in the registers when using both the 0.7.1 edk2-platforms firmware & SBL: `pin 0 (PCIE_WAKE0_B) mode 1 0x44000402 0x0003f000 [ACPI]`.

Is there an obvious way to prove whether the modified ACPI _PTS method is working correctly, or not? As another thought, is there something which ScSmiDispatcher does in the edk2-platforms code which changes around some other PMC settings which are necessary in order for the PCIe wake-up event to be generated?

mauricema, 2019-11-21:
One question,  are  you trying to wake from S5 or S3 ?

_PTS will be called just before the system enters sleep state.  If might be hard to check the if you don't have XDP debug.   Normally, GPE enabling will be tied to the _Lxx method.  Since we are trying to set GPE0 BIT3,  we will need to have _L03 in ASL code.   Can you try to add it and try ?

mauricema, 2019-11-21:
Regarding to write G0EN in _PTS, I debugged it with XDP.   The write did take effect.  However,  after the  _PTS() is called,  looks like kernel is zeroing the G0EN out again before entering sleep.  That explained why the G0EN has improper values before the S5.    EDK2 uses SMM handler to ensure the bits are bit before S5.  However, SBL does not enable SMM by default, so it cannot be done in this way.   I am wondering if we can figure out why linux kernel decided to zero out G0EN at the end.

umaplehurst, 2019-11-21:
Trying to wake from S3.

Thank you! That was just the pointer that I needed. I was trying to copy what the previous firmware was doing via _PTS, but the old stack was taking a sledge-hammer approach instead of doing it the "right" ACPI way via _Lxx. Changing it to use _L03, everything works fine from Linux, I think this is the right way forward. Is this something that you'd want to merge? In which case, I can instead submit a pull request for it.

One observation is that WoL works only once on the CRB (it seems that the SoC thinks that WAKE# stays asserted after the first time). This may be a HW issue on the CRB. I'm running further tests to see if the reference BIOS does the same thing...

mauricema, 2019-11-21:
So glad to see it eventually worked for you.   :)   
Thank you for digging into it and trying it.  
Yes, if you can submit a PR for it,  I will be more than happy to merge it.

umaplehurst, 2019-11-21:
Will do! I don't know if there is a longer-term desire to make the wake-up configuration more elegant somehow based on a board-specific property coming from the SBL board config file, but we can begin somewhere :)

On the CRB, I also see the same 'works only once from cold boot' issue with IFWI v229.20 so I am going to assume it's a CRB HW bug rather than any firmware issue (I don't remember seeing that on our non-CRB board, but don't have it to hand just right now to compare against).

mauricema, 2019-11-22:
I have merged your PR and then submitted a new one to make it generic.
If you can help review and test it , that will be great!
https://github.com/slimbootloader/slimbootloader/pull/468