slimbootloader/slimbootloader #395: [APL] Dynamic Platform ID detection not working?

umaplehurst, 2019-10-16:
I'm trying out the following:

1. Create `CfgData_Ext_Test.dlt` in `Platform/ApollolakeBoardPkg/CfgData`:

```
PLATFORMID_CFG_DATA.PlatformId   | 0x10
PLAT_NAME_CFG_DATA.PlatformName  | 'TST'
PID_GPIO_CFG_DATA.PidPin0.Enable | 0x1
PID_GPIO_CFG_DATA.PidPin1.Enable | 0x1
PID_GPIO_CFG_DATA.PidPin2.Enable | 0x1
PID_GPIO_CFG_DATA.PidPin3.Enable | 0x1
[ ... MEMORY_CFG_DATA follows ... ]
```

2. Replace `self._CFGDATA_EXT_FILE` with `['CfgData_Ext_Test.dlt']` in `BoardConfig.py`;
3. Compile, stitch, flash, and observe the following on the serial log:

```
GPIO based PID detection is disabled
```

If I change `PlatformId` to 0x00 (as per the documentation) then I don't see the `GPIO based PID detection is disabled` message, it seems that the GPIO detection takes place correctly, and the BoardID is determined as 0x10. However, I then get `Platform Name config not found` & `Failed to find memory CFGDATA!` as the next errors.

So it seems that:
- have the PlatformId < 16, GPIO PID detection works, but then `PLAT_NAME_CFG_DATA` & `MEMORY_CFG_DATA` isn't picked up from the .dlt file;
- have the PlatformId >= 16, GPIO PID detection fails, but if I hard-code `GetBoardIdFromGpioPins` to return 0x00 in this case, all the relevant CFG_DATA is then correctly picked up from the .dlt file;

After trying that, I then created two separate .dlt files:
- have the PID_GPIO_CFG_DATA settings in a .dlt file with 0x00 as the PlatformId;
- have the real board settings in a .dlt file with 0x10 as the PlatformId;

And things seem to be working OK with this setup. Is this as intended? If so, it would be great if the documentation makes it obvious that two separate files should be created for the feature to work right.

mauricema, 2019-10-17:
What you did should work.  

PlatformId 0 is a reserved ID to carry the default base configurations for all boards.  And the default configuration is listed in the CfgDataDef.dsc.  Each dlt file can be associated with a different PlatformId to specify the modifications on top of the default base configuration file.   A dlt file with PlatformId 0 will actually override the default configuration listed in CfgDataDef,dsc.  It will have similar effect as changing the CfgDataDef.dsc file.   

For GPIO PID detection expects all boards to use the same set of GPIO pins to detect board ID.   That is reason why this configuration is only associated with PlatformId 0.  If you load the dsc/dlt using ConfigEditor.py,  you will not see these options if the PlatformId is not 0.   In case that you do need to change the GPIO pins for board detection, you can either change the  CfgDataDef.dsc directly  or  as what you have done,  add it into a dlt file with PlatformId 0.   This will allow SBL to use the new GPIO info provided to detect boards.   

Yes, we can enhance our documentation to make it more clear.   Thank you for pointing it out to us.

umaplehurst, 2019-10-17:
Sounds great. Thank you for the very clear explanation of the way that things are expected to work.

rprangar, 2019-12-11:
Updated configuration section https://slimbootloader.github.io/developer-guides/configuration.html#platform-id

Closing the issue.

closed 2019-12-11 by rprangar