slimbootloader/slimbootloader #399: [APL] No UART console output with EDK2 UEFI Payload

umaplehurst, 2019-10-17:
Bug doesn't exist with QEMU, but does on physical APL hardware. FWIW, I was also compiling my EDK2 on Linux rather than on Windows. I found that using a newer EDK2 made the issue go away, so I think just a documentation update may be necessary.

[1] suggests to use EDK2 commit `42d8be0eaac5e7e109f487d4e241847e815b077a`; with this commit, I found that I wasn't getting any UART output from the EDK2 (however, it was otherwise working just fine, as it could boot Linux OK, and Linux was then producing output on the UART).

Updating EDK2 to tag `edk2-stable201908` (currently, at commit `37eef91017ad042035090cae46557f9d6e2d5917`) solved the issue for me. However, I have no idea if any other things are broken with that commit; I am booting Linux rather than Windows from UEFI.

[1] https://slimbootloader.github.io/how-tos/boot-windows.html

rprangar, 2019-10-18:
Thanks for letting us know. We will see if we can reproduce the issue. But since there is a stable release, we can update the documentation as well to reflect that.

mauricema, 2019-10-22:
I could not reproduce it on my side.   I have tried both Windows build (VS2015) and Linux build (GCC 5.5) with EDK2 commit 42d8be0eaac5e7e109f487d4e241847e815b077a.  The UEFI payload worked find with SBL  and I can see the uart2 print fine.    
Did you build a release target by any chance ?
Could you please attach your UEFI payload binary so that we can take a look ?

mauricema, 2019-10-28:
@umaplehurst   Could you please provide any updates on this ?  Since we cannot reproduce it on our side ,  we plan to close it if there is no further updates by Nov 1st.

umaplehurst, 2019-10-31:
Apologies for the delay. I've tried this again from a clean build on my side and I can't reproduce this any more, even when going back to the older git commit of EDK2. So, I'm not too sure what went wrong earlier. Thanks for double checking this on your side. Closing the ticket!

closed 2019-10-31 by umaplehurst