slimbootloader/slimbootloader #403: [APL] Documentation for triggering FW update from OS

umaplehurst, 2019-10-17:
[1] says that "triggering firmware update is a platform specific mechanism". [2] says how to do it manually through the shell and that the real OTA mechanism needs to "deposit" a capsule image, but doesn't say what this mechanism needs to do in practice.

[1] https://slimbootloader.github.io/security/firmware-update.html
[2] https://slimbootloader.github.io/how-tos/update-firmware.html

How is the OS (e.g. Linux) meant to request an FW update to be triggered? It would be great if the expected flow for a platform could be clearly documented, with guidance on how it can be easily triggered from the OS for testing purposes.

rprangar, 2019-10-17:
The exact method to trigger FwUpdate is platform specific and **can be customized** based on platform designs.  

There is one example which is implemented on both APL and CFL/WHL platforms...so I will provide that information below.

This example implementation uses CMOS register (accessed through IO ports 0x70 and 0x71) 0x40. If a value of 0x5A is found, then FW Update mode is detected and the Boot Mode is set to FLASH_UPDATE. Please refer IsFirmwareUpdate() function called in Platform\ApollolakeBoardPkg\Library\Stage1BBoardInitLib\Stage1BBoardInitLib.c

The location of the capsule is provided through a configuration option - CAPSULE_INFO_CFG_DATA (Platform\ApollolakeBoardPkg\CfgData\CfgData_CapsuleInformation.dsc & Platform\CoffeelakeBoardPkg\CfgData\CfgData_CapsuleInformation.dsc)

I agree we can provide more information on the example implementations.

rprangar, 2019-10-17:
To add to the previous common implementation between APL and CFL, there is another method that is unique to APL. This method uses the CS ME to pass the FW Update command and the Capsule location to the Slim Bootloader. Please refer to ProcessMbpData function in Platform\ApollolakeBoardPkg\Library\Stage1BBoardInitLib\Stage1BBoardInitLib.c. 

This data passing mechanism is implemented in 
Platform\ApollolakeBoardPkg\Library\ShellExtensionLib\CmdFwUpdate.c.

raghavag, 2019-11-08:
Added required information about firmware update. Changes to the documentation are merged using https://github.com/slimbootloader/documentation/pull/60

closed 2019-11-08 by rprangar