slimbootloader/slimbootloader #409: CFL-H Support

lKARAKAYA, 2019-10-21:
Hello,
For a custom board with Coffee Lake-H Xeon-E 2176m processor and CM246 PCH, I will try to build slimboot. According to documentation, this processor is not supported. Has anybody build SBL for this processor.? Does SlimBoot support this processor.?
Best regards

raghavag, 2019-10-22:
The most critical component to support a platform is the Firmware Support Package(FSP). The FSP available at https://github.com/IntelFsp/FSP/blob/master/CoffeeLakeFspBinPkg/README.md supports both Coffeelake and Coffeelake refresh processors. 

While the FSP is the same for Coffeelake and Coffeelake refresh processors, our validation was done using the Coffeelake refresh processor. 

While theoretically, the SBL supporting Coffeelake Refresh platform may support Coffeelake processor as well, there may be some porting steps like configuration option changes required. But please be aware this has not been validated.

rprangar, 2019-10-23:
Closing this issue. Please open new issues if you run into any in the process of porting SBL for your custom hardware with Coffelake

closed 2019-10-23 by rprangar