slimbootloader/slimbootloader #448: System can NOT boot to android on Slim Bootloader

wt-kevin, 2019-11-11:
1. Download android-x86_64-6.0-r3.iso and make it as USB installation by Refus.
2. Install it on Oxbow CRB with UEFI mode of setup menu(Intel BIOS：intel-atom-e3900-soc-ifwi-205-21-release-package).
3. Remove USB disk, and reset system.
4. Boot to android successfully.
5. Flash Slim bootloader on Oxbow CRB.
6. Boot fail to android.
Slim Bootloader can't support android? if it supports, how can we fix it?

mauricema, 2019-11-11:
Currently Android is not on officially tested by Slim Bootloader for APL platform.

However, I did try out the android-x86_64-6.0-r3.iso. I can install and boot Android image with Slim Bootloader + UEFI payload.   If you did not enable UEFI payload, please do so by following the steps mentioned at 
https://slimbootloader.github.io/tutorials/ex_uefi_payload.html?highlight=uefi%20payload

It seems this Android release supports UEFI boot only, so you will have to use UEFI payload vs OsLoader to boot.   Technically it might be also possible to use OsLoader,  but I believe you need to do some further packaging and customization so that it can follow traditional Linux boot flow to load Android.

mauricema, 2019-11-12:
Some further updates.  I also tried to boot this pre-installed Android image using OsLoader, and I was able to boot it successfully.   

First step is to install Android image into a boot media partition.  This partition needs to be formatted as FAT.  Both GRUB2 and Android file system will be installed into this same partition.   After the installation, the FAT partition should have efi, boot and android-6.0-r3 folders under the root.  

Second step is to modify  /boot/grub/grub.cfg on the FAT partition to add missing boot parameters BOOT_IMAGE=xxxxx.  
Change line from:
	linux /android-6.0-r3/kernel root=/dev/ram0
To:
	linux /android-6.0-r3/kernel BOOT_IMAGE=/android-6.0-r3/kernel root=/dev/ram0
Save it back into the FAT partition.

Third step is to config SBL to boot from that FAT partition by setting the boot options properly.  This can be done in OsLoader Shell using command "boot".

Now, SBL should be able to boot Android directly from the partition.

rprangar, 2019-11-12:
Booting Android requires an Android bootloader. The OsLoader payload of Slim Bootloader can support multiboot compliant Android bootloader, but not the EFI version of Grub used by android x86.

Our recommendation would be to use the UEFI payload which supports the EFI Grub to boot Android with SLim Bootloader.

closed 2019-11-12 by rprangar

wt-kevin, 2019-11-22:
Hi mauricema,
As you said, we tried your steps but still fail, please refer to the following link to download DOC, our steps is descripted on it.
https://www.dropbox.com/s/w4ir2pcomoamytw/Install%20Android%20on%20APL%20CRB%20with%20Slim%20Bootloader.docx?dl=0

mauricema, 2019-11-22:
Hi, Kevin,
In your steps, you need to modify the grub.cfg file on the actual boot media (eMMC, SATA, etc) after the full Android installation.  It looks like you modified the one on USB key even before your installation.  

Also when you  add "BOOT_IMAGE=..." to grub.cfg, please make sure you add it into the correct boot entry.  If you want to boot from option '0',  you should modify the 1st boot entry in  grub.  If you are not sure, you can add "BOOT_IMAGE=..." to all the boot entries in grub.cfg.

wt-kevin, 2019-11-27:
Hi mauricema,
I installed Ubuntu in EMMC and Android6 in external SATA HDD, we can know SATA HDD is /dev/sda as picture(please refer to the following link),
https://www.dropbox.com/s/hvpqw5p33gov1mt/S__126246929.jpg?dl=0
and I changed grub.cfg from
	linux /android-6.0-r3/kernel quiet root=/dev/ram0  
into
	linux /android-6.0-r3/kernel quiet BOOT_IMAGE=/android-6.0-r3/kernel root=/dev/sda0
in SATA HDD,
but the result still failed to boot to SATA HDD as below,
https://www.dropbox.com/s/cksz4sawljxa9ks/messageImage_1574840913572.jpg?dl=0
Do you have any idea? thanks very much for help,

mauricema, 2019-11-28:
Please check your SATA /dev/sda1 does have valid files.  From your log,  SBL could not find grub.cfg on SATA.   All files should be installed on the SATA.
To make it simple, I would recommend you just install Android on eMMC and not use SATA at all to test the whole flow.  I did try that on my side and it worked for me.