slimbootloader/slimbootloader #466: [APL] INT34D2 IPC commands time out

umaplehurst, 2019-11-22:
I'm seeing the following behavior:

Kernel v4.14 & IFWI v227.51, Leaf Hill CRB:
`# echo '231 0' > /sys/devices/platform/INT34D2:00/simplecmd` -- works, returns straight away.
Note: it seems the ASL has changed in IFWI v229.20; I don't see the INT34D2:00 device enumerated in Linux any more.

With SBL:
`# echo '231 0' > /sys/devices/platform/INT34D2:00/simplecmd` -- hangs for a bit, then "IPC timed out" error is reported back in kernel log.

myguitar, 2019-11-22:
Thank your for opening this issue. I will setup the test environment, replicate the issue and get back to you.

myguitar, 2019-11-23:
It looks irq40 is not triggered. Let me take a look more and get back to you.

mauricema, 2019-11-24:
@umaplehurst:
The IRQ for IPC device should have been configured by FSP.   However, it seems to be some issue there. We will look into it.
Meanwhile, the following workaround can be applied in SBL Stage2 platform BoardInit (PreSiliconInit) to make it work.
MmioOr32 (ACPI_MMIO_BASE_ADDRESS + 0x950204, 0x00028002);
Verified this workaround with Ubuntu.

mauricema, 2019-11-24:
It turned out that SBL set IA Untrust bit too early which blocked FSP notification API from making changes to some registers.   Submitted a PR #470 to address it.

umaplehurst, 2019-11-25:
Many thanks for looking into this. Here are my observations:

a) MmioOr32 workaround -- works,
b) PR #470 (without MmioOr32 workaround) -- works, but, it doesn't work any more, once the attached settings are added to CfgData_Int_LeafHill.dlt ... not sure why disabling some peripherals should cause this problem to resurface, but maybe something funky is happening inside FSP with interrupt configuration?

[CfgData.txt](https://github.com/slimbootloader/slimbootloader/files/3884458/CfgData.txt)

mauricema, 2019-11-25:
Short answer,  yes, FSP will request reset if some IP block is disabled.   In this case,  ISH and SATA disabling need to do cold reset.   Currently SBL did not check the FspNotify API return status, and ignored this reset request.  i think we should add that although it is not common to see reset in late boot stage.  If we add the same code as FspSiliconInit handling reset, it will work.  However,  it needs to do reboot on the boot flow.

I think it might be good to keep ISH and SATA controller enabled.  Or maybe consider disabling them through soft straps if possible.  

Anyway, I would say it is totally different issue.   If you are OK, we can merge PR #470 first.   And then try to add code to handle the reset request for FspNotify API in SBL.

umaplehurst, 2019-11-25:
Understood, sounds great to me to merge #470 first!

closed 2019-11-25 by mauricema

umaplehurst, 2019-11-26:
I have tested with #471 and all is great now. Many thanks! I agree, it's good to avoid the additional FSP-requested reboot in the boot cycle, so I will look at changing the straps.

mauricema, 2019-11-26:
Thank you for verification on the fix.  :)

commit abee91f2 (2019-11-25): Delay IA untrust bit set for BSP

SBL currently set IA Untrust bit in MpInit(). It is too early for
BSP. By doing so, it blocked some register access in FSP notification
APIs. This patch moved the IA Untrust bit set for BSP to the end of
boot flow while keeping it at same location for APs.

It also fixed #466.