slimbootloader/slimbootloader #474: Failed to build QEMU FSP binary

jacendi, 2019-11-27:
I'm trying to build Slim Bootloader with UEFI payload for QEMU (using GCC 9.2.1), but an error (unknown symbol FspSecCoreT:_TempRamInitApi) occurs at the post-build stage of QEMUFSP. Here's a part of log:

`Fd File Name:QEMUFSP (/home/jacendi/Download/QemuFsp/Build/QemuFspPkg/RELEASE_GCC5/FV/QEMUFSP.fd)

Generate Region at Offset 0x0
   Region Size = 0x15000
   Region Name = FV

Generating FSP-S FV
###############################
Generate Region at Offset 0x15000
   Region Size = 0x22000
   Region Name = FV

Generating FSP-M FV
#######################
Generate Region at Offset 0x37000
   Region Size = 0x1000
   Region Name = FV

Generating FSP-T FV
######
GUID cross reference file can be found at /home/jacendi/Download/QemuFsp/Build/QemuFspPkg/RELEASE_GCC5/FV/Guid.xref

FV Space Information
FSP-S [16%Full] 86016 total, 14336 used, 71680 free
FSP-M [27%Full] 139264 total, 38584 used, 100680 free
FSP-T [67%Full] 4096 total, 2776 used, 1320 free
Build report can be found at /home/jacendi/Download/QemuFsp/ReportRELEASE.log

- Done -
Build end time: 15:55:18, Nov.27 2019
Build total time: 00:00:05

End of Build...
Start of PostBuild ...
Patch FSP-T Image ...
Patched offset 0x00000000:[00000000] with value 0xFFFFF000  # Temporary Base
Patched offset 0x000370AC:[12345678] with value 0x00001000  # FSP-T Size
Patched offset 0x000370B0:[12345678] with value 0xFFFFF000  # FSP-T Base
Patched offset 0x000370B4:[00000000] with value 0x00000001  # FSP-T Image Attribute
Patched offset 0x000370B6:[56780000] with value 0x56781003  # FSP-T Component Attribute
Patched offset 0x000370B8:[12345678] with value 0x000001F4  # FSP-T CFG Offset
Patched offset 0x000370BC:[12345678] with value 0x00000092  # FSP-T CFG Size
ERROR: Unknown symbol FspSecCoreT:_TempRamInitApi !
Traceback (most recent call last):
  File "/home/jacendi/Download/QemuFsp/BuildFsp.py", line 309, in <module>
    sys.exit(Main())
  File "/home/jacendi/Download/QemuFsp/BuildFsp.py", line 303, in Main
    PostBuild (target, toolchain)
  File "/home/jacendi/Download/QemuFsp/BuildFsp.py", line 268, in PostBuild
    Fatal('Failed to do PostBuild QEMU FSP!')
  File "/home/jacendi/Download/QemuFsp/BuildFsp.py", line 117, in Fatal
    raise Exception(msg)
Exception: Failed to do PostBuild QEMU FSP!
Traceback (most recent call last):
  File "/home/jacendi/slimbootloader/BootloaderCorePkg/Tools/PrepareBuildComponentBin.py", line 261, in <module>
    sys.exit(Main())
  File "/home/jacendi/slimbootloader/BootloaderCorePkg/Tools/PrepareBuildComponentBin.py", line 251, in Main
    BuildFspBins (qemu_repo_dir, sbl_dir, silicon_pkg_name, target)
  File "/home/jacendi/slimbootloader/BootloaderCorePkg/Tools/PrepareBuildComponentBin.py", line 227, in BuildFspBins
    Fatal ('Failed to build QEMU FSP binary !')
  File "/home/jacendi/slimbootloader/BootloaderCorePkg/Tools/PrepareBuildComponentBin.py", line 17, in Fatal
    raise Exception (msg)
Exception: Failed to build QEMU FSP binary !
Traceback (most recent call last):
  File "BuildLoader.py", line 1291, in <module>
    main()
  File "BuildLoader.py", line 1288, in main
    args.func(args)
  File "BuildLoader.py", line 1235, in cmd_build
    Build(board).build()
  File "BuildLoader.py", line 1081, in build
    self.pre_build()
  File "BuildLoader.py", line 965, in pre_build
    raise Exception  ('Failed to prepare build component binaries !')
Exception: Failed to prepare build component binaries !`

What could be the reason for this error? Thanks in advance for help.

mauricema, 2019-11-28:
I think it is caused by the GCC version.  GCC 5.x, 6.x, 7.x have been tested.  
Please use GCC 7.x instead.

The reason for this error could be some more aggressive optimization done at link time in newer GCC version.  Since this API will not be called directly inside FSP itself,  the linker might have removed it from the final binary.  As a result, and the symbol could not be located.

jacendi, 2019-11-28:
Thanks! Yeah, it's compiles fine with GCC 7.x.

closed 2019-11-28 by jacendi

luojia65, 2022-03-30:
Does this issue happen again? I can't build as well for `unknown symbol FspSecCoreT:_TempRamInitApi`