slimbootloader/slimbootloader #477: [APL] Newest SBL Version in Github can't boot to APL

wt-kevin, 2019-11-29:
1. build apl by the following command：
C:\MyWorkSpace\SlimBootloader\python BuildLoader.py build apl -r
2. After build successfully, download LEAFHILD.X64.0070.R01.1805070352.bin from https://slimbootloader.github.io/supported-hardware/apollo-lake-crb.html#stitching
3. Execute the following command to stitch：
C:\MyWorkSpace\SlimBootloader\python Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i LEAFHILD.X64.0070.R01.1805070352.bin -s Outputs/apl/Stitch_Components.zip -o sbl_lfh_ifwi.bin
4. Use SF600 to flash sbl_lfh_ifwi.bin into Oxbow Hill CRB,
5. System hang up as below,
https://www.dropbox.com/s/petmvp8fc1tcjb3/%E6%9C%AA%E5%91%BD%E5%90%8D.jpg?dl=0
6. the previous version before three month ago is OK.
7. I never change my procedure, I don't know why~

mauricema, 2019-11-29:
Not sure what OS your are trying to boot.  But to look into issue, you should use debug build instead of release build '-r'.  Also it is always good to check if you are using the proper payload (OsLoader or UEFI)  and related boot options (Boot option needs to match your boot device settings).

wt-kevin, 2019-12-03:
In previous SBL source of Github, which was used on the same configuration, it can boot to shell successfully without hanging up even I use "-r" to build SBL binary on APL.
For newest SBL source of Github new, if I build SBL "-r" like the previous SBL, it hangs up at  "STAGE_PAYLOAD" as below,
https://www.dropbox.com/s/petmvp8fc1tcjb3/%E6%9C%AA%E5%91%BD%E5%90%8D.jpg?dl=0
but, if I use debug mode(no "-r") to build SBL on APL, it can boot successfully to shell without hanging up at "STAGE_PAYLOAD".
Do you know why? The only difference is build by Debug or Release mode.

mauricema, 2019-12-03:
There are some very basic things that will behave differently between DEBUG and RELEASE build.

For Debug build, we allow to boot to SBL shell, and allow to boot using vmlinuz / grub.cfg /config.cfg.  However, for Release build, we only allow booting signed image using IAS or Container format for security reason.  As a result, it will not allow to boot to SBL Shell.  

You can find the following code in OsLoader.c.   If you really want to boot to Shell in release build, you can remove this check and run RunShell() anyway.  But please understand the security risk since Shell will have full access to memory and MMIO and people can potentially change many things in Shell.

PayloadMain() {
   ...
   if (DebugCodeEnabled ()) {
      RunShell (0);
    } else {
      break;
    }
  ...
}

wt-kevin, 2019-12-04:
Hi Maurice,

Thank you so much for your support, I understood it.

mauricema, 2019-12-04:
You are welcome!  I will close this issue.

closed 2019-12-04 by mauricema