slimbootloader/slimbootloader #478: [APL] Stop three times during first building SBL on Windows10

wt-kevin, 2019-11-29:
1. Execute the following command to download SBL：
C:\MyWorkSpace\git clone https://github.com/slimbootloader/slimbootloader.git
2. Execute the following command to build SBL：
C:\MyWorkspace\SlimBootloader\python BuildLoader.py build apl -r
3. System stop three times, please refer to the following vedio：
https://www.dropbox.com/s/64wef7xv8vshc7c/BuildSlimBootloader.avi?dl=0
4. Anyone knows why? And it impacts function?

rprangar, 2019-12-02:
Hi

We have not seen this behavior on our side. What happens if you dont interrupt the build with "Ctrl +C"?

There were some unicode unrecognizable characters error flagged as well. We suspect that your system setup may be causing some issues. Are you able to build any other EDKII code?

Thanks
Ravi

wt-kevin, 2019-12-04:
Hi Rprangar,
If I don't interrupt the build with "Ctrl +C", it will NOT go to next step, just always stay at this point.
May I know what other EDKII code is? thank you.

rprangar, 2019-12-04:
Hi

Please refer to instructions here - https://github.com/tianocore/tianocore.github.io/wiki/UDK2018-How-to-Build

UEFI Payload also uses EDKII build infrastructure. Have you built UEFI payload before?
https://github.com/tianocore/edk2/blob/master/UefiPayloadPkg/BuildAndIntegrationInstructions.txt

Thanks
Ravi

wt-kevin, 2019-12-04:
Hi Ravi
I built UEFI payload yesterday refer to the following link, and can build successfully.
https://slimbootloader.github.io/how-tos/boot-windows.html
But System can't boot to Windows10 installation USB disk and no UEFI shell is shown.
I will release another question for this.

rprangar, 2019-12-05:
Can you please share the build logs and the serial log if possible?

wt-kevin, 2019-12-05:
Hi Ravi,

For UEFI payload?

rprangar, 2019-12-05:
No, for SBL.

wt-kevin, 2019-12-05:
I mean for this issue that system can't boot to Windows10 installation USB disk and no UEFI shell is shown?

rprangar, 2019-12-05:
No. This is for the original issue where you need to press ctrl+c to get the SBL build going. 

Thanks
Ravi

myguitar, 2019-12-06:
It looks this issue happens on Windows which is using unicode system locale. By any chance, can you change your Windows system locale to US-English and try to re-build SBL?

wt-kevin, 2019-12-06:
Hi myguitar and Ravi,
Thanks for help, the following is log file I built, you can search the following string for stop point.
1. "NMAKE : fatal error U1058: 已被使用者結束"
2. "要終止批次工作嗎 (Y/N)?"
3. "After Stop by Ctrl+C for hanging up at KeyboardInterrupt by Kevin and the rebuild, will not hang up." (Added by Kevin not log shown during building SBL)
Uploaded log file : https://www.dropbox.com/s/bzgsk5zunr7ynba/SBL%20first%20Build.txt?dl=0

myguitar, 2019-12-07:
I am able to reproduce your issue on unicode locale enabled Windows. I recommend to change Language for non-Unicode programs setting. You don't have to change your system locale, but only for non-Unicode program setting.
[image]

FYI, below draft PR would resolve your issue on your system, but still recommend to change your Windows setting to avoid further issues.
https://github.com/slimbootloader/slimbootloader/pull/490

wt-kevin, 2019-12-09:
But I sill built fail(the same as the previous) after changing system locale, please refer to the following picture.
https://www.dropbox.com/s/cyrf82j58ek5az2/Change_Unicode.jpg?dl=0
BTW, could you let me know how we can post picture directly on the Dialog box?

myguitar, 2019-12-09:
> But I sill built fail(the same as the previous) after changing system locale, please refer to the following picture.
> https://www.dropbox.com/s/cyrf82j58ek5az2/Change_Unicode.jpg?dl=0
> BTW, could you let me know how we can post picture directly on the Dialog box?

Use Enligsh (US) for non-unicode programs and uncheck “Beta: ...”

For image, just do drag-and-drop.

wt-kevin, 2019-12-09:
[image]

wt-kevin, 2019-12-09:
fg

wt-kevin, 2019-12-09:
[Celadon Android OS for Intel® Celeron.pdf](https://github.com/slimbootloader/slimbootloader/files/3938226/Celadon.Android.OS.for.Intel.Celeron.pdf)
f

wt-kevin, 2019-12-09:
df

rprangar, 2019-12-09:
Thanks. Can you please try with PR #490 and see if this resolves the issue?

myguitar, 2019-12-09:
I did double-check this with your language pack (Chinese - Traditional, Taiwan).
1. Without any change, this issue is reproducible
2. But not reproducible with Non-Unicode Programs to English-US.
3. And also, it's not reproducible with #490 on apl target (#490 is not applicable for qemu target as draft patch).

If both 2, 3 do not help, you may need try this with system locale English-US.

wt-kevin, 2019-12-11:
Hi 
The following is my results, thank you.
1. Without any change, this issue is reproducible.
2. With Non-Unicode Programs to English-US as below, the issue is still reproducible, I don't why the result between us is different.
[image]
3. Phasing #490 on apl target, the issue is not reproducible.(Fixed)

rprangar, 2019-12-11:
Thanks for the update. We will close this issue tomorrow.

myguitar, 2019-12-12:
@wt-kevin The fix #490 has been merged. Try this on the latest code base.

closed 2019-12-12 by rprangar