slimbootloader/slimbootloader #485: [APL] Fail to boot to windows10 with UEFI Payload

wt-kevin, 2019-12-04:
1. Follow the following link to build UEFI payload.
https://slimbootloader.github.io/how-tos/boot-windows.html
2. Built successfully and go to step 4 to create windows installation USB disk.
3. Try to boot to USB disk and install windows10.
4. Fail to boot to USB disk as below,
https://www.dropbox.com/s/1vo8g7vvjgkq4ur/UEFI_Payload.jpg?dl=0
If UEFI payload install fail? even I build successfully
if UEFI payload install successfully, I should be able to boot to uefi shell?

mauricema, 2019-12-04:
You were still booting OsLoader, not UEFI payload.  
Recently a GPIO based Payload selection was added.  And by default, it will boot OsLoader on LFH.   
Locate connector J6B2 on your board and then short Pin2 and Pin10.  It  should switch to boot UEFI payload then.

mauricema, 2019-12-04:
We will revisit this logic for LFH.   I think it might be better to enable GPIO based payload selection only when PayloadId is set to AUTO.

wt-kevin, 2019-12-05:
Hi mauricema,

After locating connector J6B2 on your board and then short Pin2 and Pin10 on Oxbow Hill,
it is OK to boot to UEFI shell, thank you for help.
mauricema：I think it might be better to enable GPIO based payload selection only when PayloadId is set to AUTO.
you mean when PayloadId is set to AUTO in SBL source code, UEFI payload still works fine even we don't short Pin2 and Pin10 on J6B2?

mauricema, 2019-12-05:
I have submitted a PR to handle this https://github.com/slimbootloader/slimbootloader/pull/487.
Please check the commit message to understand how PayloadId will be decided.
If you set PayloadId to UEFI,  the latest code should not require you to short the pin any more.
But if you set PayloadId to AUTO, then the GPIO will decide to boot OsLoader or UEFI.

mauricema, 2019-12-06:
please let us know if it can be closed.

closed 2019-12-06 by mauricema

wt-kevin, 2019-12-06:
Hi Mauricema,

Thank you, the issue can be closed.
BWT, as you mentioned as below,
Mauricema：But if you set PayloadId to AUTO, then the GPIO will decide to boot OsLoader or UEFI.
Should we be able to add the related GPIO information about payload to the following user guide?
https://slimbootloader.github.io/how-tos/boot-windows.html

mauricema, 2019-12-06:
This guide requested user to set  PayloadId to 'UEFI'.  With the latest code, by doing so, it does not required any GPIO changes any more.

wt-kevin, 2019-12-06:
Hi Mauricema,

Thanks for your information, I got it.