slimbootloader/slimbootloader #492: Celadon Android BSP boots fail in APL OxbowHill CRB

wt-kevin, 2019-12-09:
1. Build BSP and flash the image into eMMC by the following steps in pdf file.
[Celadon Android OS for Intel® Celeron.pdf](https://github.com/slimbootloader/slimbootloader/files/3938751/Celadon.Android.OS.for.Intel.Celeron.pdf)
Image download link：https://www.dropbox.com/s/d5t6zrwc05512p5/cel_apl-flashfiles-eng.wtbuild1604.zip?dl=0
2. Boot fail refer to the following video：
https://www.dropbox.com/s/v9oxl61910aj30z/Celadon_on_SlimBootloader.mp4?dl=0

myguitar, 2019-12-09:
@wt-kevin Can you please describe your issues in detail instead of sharing video? If the external link is deleted, the issue history cannot be tracked. It would be helpful to someone who has the similar experience, or the someone who may know the answer would help you.

As a first look, it looks SBL can boot to Android bootloader. You can see Android bootloader display and message.
At the error message, the Android bootloader is looking for a/b info. This is Android specific feature. You may disable the feature in Android build, or may need to pass proper information in commandline or something like that.

wt-kevin, 2019-12-11:
Can we use adb tool to get log for Android x86?
and do you know how to get android log by the other way(android saved on eMMC? or others?)?

rprangar, 2019-12-11:
Hi

What payload are you using with Celadon?

Thanks
Ravi

wt-kevin, 2019-12-12:
Hi Ravi,
SBL+UEFI payload
Thank you.

rprangar, 2019-12-13:
Hi Kevin

For ADB, you may have to enable device mode in the UEFI payload to be able to interact with the host over USB.

Thanks
Ravi

myguitar, 2019-12-14:
@wt-kevin Have you tried it with Debug UEFI payload build? Can you share UEFI payload logs? It may take a while for me to set up this environment.

wt-kevin, 2019-12-18:
Hi myguitar,
This is SBL of debug version, you mean we need another debug UEFI payload build?
The following is my logs from console port. please refer to it.
[SBL debug log.txt](https://github.com/slimbootloader/slimbootloader/files/3977886/SBL.debug.log.txt)

wt-kevin, 2019-12-18:
Hi rprangar,

SBL doc mentioned how to enable device mode in the UEFI payload? we can't find it~

myguitar, 2019-12-18:
1. UEFI BIOS
I have tried UEFI BIOS first to make sure the installed images work as expect, but the device keeps rebooting after "Call TOS loader entry_addr = 0x74196038". No clue in the log, but I guess this is key mis-match in Android Trusty OS. You may need to get some help from Android or kernelflinger expert to handle TOS properly.

2. A/B partition
In your log, "avb_ab_flow.c:312: ERROR: No bootable slots found.". You need to set proper active slot in misc partition.
fastboot flashing unlock
fastboot set_active a
fastboot flashing lock
fastboot continue

3. EDKII UEFI Payload
It looks kernelflinger is using EFI Variables after rebooting it tries to read the VARs. But unfortunately, UEFI Payload in EDKII repo is using Emulation Variable driver. That means it's valid only during system power on and you cannot get the proper variables after reboot.
You need to use UEFI Payload Intel provides to use Variable service. Please contact your Intel representative to get the UEFI Payload Intel provides.

4. USB device mode
The USB device mode works as expected w/o any change. You can check with "fastboot devices" command in Android Bootloader mode.

wt-kevin, 2019-12-20:
Hi myguitar,

Did you build the image refer to the following link?
https://01.org/projectceladon/documentation/getting_started/build-source

myguitar, 2019-12-20:
> Hi myguitar,
> 
> Did you build the image refer to the following link?
> https://01.org/projectceladon/documentation/getting_started/build-source

@wt-kevin No, I don't have the build environment. Practically, we are not able to replicate every environments all the time. We are trying to answer any technical questions and support issues based on the given information.
SBL supports Yocto and Windows, and we have verified Ubuntu, ACRN, Android and etc. with basic boot support level. So we are looking forward to many contributions from open source community developers for new/missing/advanced features and bug fixes. I think this Android TOS is also one of items community can help. We will more than welcome your contribution as well.

wt-kevin, 2019-12-20:
Hi myguitar,
Thanks for information,
Will you be able to build the environment or android image refer to 576309-intelatom-e3900-apollolake-i-android-o-rn-rev2-0-gold.pdf from RDC?
We also got the built image from Win-River and failed to boot on SBL on APL CRB, but it can boot successfully on UEFI BIOS on it.
So we will also open another issue to trace it. thank you.

rprangar, 2019-12-20:
Hi Kevin

OS boot capability is in the payload and may need porting for a particular platform to support an OS. 

Android is not officially tested on Slim Bootloader for APL platform as indicated in the other Android [issue](https://github.com/slimbootloader/slimbootloader/issues/448).

The open source UEFI payload is platform agnostic and as such lacks certain features like persistent EFI variable support, etc. As indicated earlier in this thread, some of these missing features in the generic UEFI payload may prevent Celadon booting. 

More debugging maybe needed from your side as we don't have a ready recipe for this issue. 

Thanks

wt-kevin, 2019-12-23:
Hi rprangar,

Thanks for your information, I will check it.

closed 2019-12-23 by rprangar