slimbootloader/slimbootloader #503: [APL] intel-spi driver no longer working

umaplehurst, 2019-12-14:
I found a regression while rebasing to the latest SBL for my board. Testing on LH CRB, I can reproduce the same issue:

Test case: check that `hexdump /dev/mtd1ro` works (make sure that intel_spi & intel_spi_platform drivers are loaded -- not sure if Ubuntu has these, I'm using Yocto). You should be able to read the SPI flash contents, but with the newer release, I get "Input/Output error" instead.

I then bisected further. Looks like the commit that's causing the issue is ec8dca50bb16396d80cddadc4bf9a8a0c95151f9. Reverting that change recovers the expected behavior.

mauricema, 2019-12-14:
Thank you for reporting it.   Sure,  we will take a look.  
Meanwhile, could you please roll back to use APL MR5 FSP on latest code and try again ?   We recently found some issues with APL MR6 FSP.

umaplehurst, 2019-12-14:
Sure. I've tried, rolling back FSP to MR5 does not seem to fix it.

mauricema, 2019-12-14:
@umaplehurst
I tried to follow your instructions on the latest code with and without commit ec8dca5.  On my side, the result was always consistent failure. ("Input/Output error").   So it seems that commit might not be the real issue.  Not sure why when you rolled it back, it worked for you.  

I did further debug and found that it should be the intended behavior with current code.  The SBL code sets full SPI flash as Read and Write protect.  If you use the patch I attached below, it will remove the read protect and you then can have read access to SPI flash.  I tried this on top of the latest code,  and it can read the device mtd1ro. 

Please let me know if it works for you.  If so, is there any special reason you need to read SPI ?  If you think it is a valid use case, we can consider removing the SPI RD protection.
  
[disable_spi_read_protect.txt](https://github.com/slimbootloader/slimbootloader/files/3964056/disable_spi_read_protect.txt)

mauricema, 2019-12-14:
Actually, now I might understand why it worked for you.  As I indicated in  commit ec8dca5, when the FSP HOB was cleared too earlier, the FSP might not run the full Notify API well because of HOB dependency.  As a result, the flash protect range might not have been programmed by the FSP in notify phase at all.  So you will have the SPI access.  But that was not intended behavior.

mauricema, 2019-12-18:
Please let us know if this addressed your issue.  
If you want to submit a patch to remove flash read protection for APL,  we will be glad to integrate it into SBL.

umaplehurst, 2019-12-19:
Many thanks for the feedback. I don't have access to the CRB hardware this week to try this out but the patch looks great and I think it will solve the issue for me. Will try it next week.

Actually, I think the default SBL behavior of blocking access entirely is sensible; for my use-case, I'm doing both read & write via intel-spi, this is to allow for firmware updates on non-SBL firmware, and I'd like the same mechanism to also work on SBL.

mauricema, 2019-12-19:
Got it. Thank you for the info.  Please keep us updated once you tested it.

umaplehurst, 2019-12-23:
I can confirm that everything works nicely with master if I tweak `FspsConfig` as per the patch. Many thanks for looking into this!

closed 2019-12-23 by umaplehurst