slimbootloader/slimbootloader #510: [APL] We can NOT use IFWI from RDC for stitching SBL

wt-kevin, 2019-12-19:
1. Refer to the following link：
https://slimbootloader.github.io/supported-hardware/apollo-lake-crb.html
[image]
2. Why must we download IFWI for stitching as above?
3. It seems we can't download IFWI image from RDC.
4. I tried it, and fail.

mauricema, 2019-12-20:
In the tool usage, it is clearly mentioned:
"This script creates a new Apollo Lake Slim Bootloader IFWI image basing
on an existing IFWI base image.  Please note, this stitching method will work
only if Boot Guard in the base image is not enabled, and the silicon is not
fused with Boot Guard enabled."

When you download IFWI image from RDC, I would assume it is Boot Guard enabled.  So it will not work for you.    The link we provided above is not from RDC,  it is EDK2 open source IFWI and Boot Guard is disabled.  
If you want to do full stitching, you should use StitchIfwi.py under the same folder.  Please follow the instruction mentioned in the usage.

wt-kevin, 2019-12-23:
Hi mauricema,

Thanks for your information and sorry for me to miss understanding.
May we close this issue?

umaplehurst, 2019-12-23:
EDK2 IFWI is available here: https://firmware.intel.com/projects/IntelAtomProcessorE3900

wt-kevin, 2019-12-24:
But we have to disable Boot Guard, right?
Because we don't have IFWI source, we we have to ask Intel to check it?

umaplehurst, 2019-12-24:
That's right, I think you do need to disable Boot Guard. But you don't need the IFWI source code to do this. Open up the IFWI firmware binary image in FIT (the tool is available as part of the TXE release kit) and you can disable it there. Then, use the resulting patched firmware image as an input to SBL.

wt-kevin, 2019-12-26:
Hi 
I found a build issue as below, did you meet it? Or do you know how we should fix it?
[image]

umaplehurst, 2019-12-26:
Yes. You need to go to "Platform Protection" (on the left side) and clear the field which says "OEM Public Key Hash". Building the image should then succeed.

wt-kevin, 2019-12-27:
Hi 
Many thanks for help, the problem is fixed.
BTW, can I say SBL don't support this function(Boot Guard)?

rprangar, 2019-12-27:
Hi Kevin

SBL supports enabling boot guard. You would need the IFWI ingredients and stitch an IFWI image using FIT tool when you want to enable Boot Guard. You can use the script to do this - https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ApollolakeBoardPkg/Script/StitchIfwi.py

Please refer to the instructions in the script for more details.

Thanks

wt-kevin, 2019-12-31:
Hi Rprangar,

Thanks for information, I'll try it later, thanks for help.

closed 2020-01-03 by rprangar