slimbootloader/slimbootloader #525: Slimbootloader hangs at FspMemoryInit stage for APL based board

shehamb, 2020-01-09:
Hello, 
We are using Intel Atom E3950 based reference board (by Congatec). I am trying to port Slimbootloader on this board, after following the instructions on the https://slimbootloader.github.io/supported-hardware/apollo-lake-crb.html

But the final BIOS image from this slimbootloader gets stuck up at FspMemoryInit. Can you please share any pointers on this?

I am pasting the console log for the same:
------------------------------------------------------------------------------------------------------------
▒BtGuard: VB : 0, MB : 0

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_APLI
ISVN: 001
IVER: 000.005.001.000.13514
SVER: 8988A7E5F07F9F53-dirty
FDBG: BLD(D) FSP(R)
FSPV: ID($APLFSP$) REV(01040301)
Loader global data @ 0xFEF01D54
Run  STAGE1A @ 0xFEF80000
Load STAGE1B @ 0xFEF88000
No BtGuard verification !

============= Intel Slim Bootloader STAGE1B =============
CPU : APL-D0 [4C @ 1600MHz: premium SKU], uCode rev.2E
Early GpioInit for EMMC
MMC global data init
Use SDMA instead of ADMA2
MMC Phase 1 init
MmcGetOcr Fail Status = 0x80000007
Early MMC Init Fails @ 0xFE700000 - Status(Device Error)
SpiInstance = FEF03490
PchSpiBase at 0xE006A000
Load EXT CFG Data @ 0xFEF03FC4:0x0CA0 ... Success
HASH Verification Success! Component Type (4)
RSA Verification Success!
GPIO based PID detection is disabled
Get Embedded BoardId
Initialize RTC with default values
BOOT: BP0
MODE: 0
BoardID: 0x07
PlatformName: LeafHill
BootPolicy : 0x00000010
Memory Init
Looking for MRC saving data
MRC data NOT found
Load memory parameters from CfgData.
Profile parameter from CfgData - 0x19.
MemorySizeLimit from CfgData - 0x0.
Setting Fspmcfg for LeafHill.
Load general parameter from CfgData.
Call FspMemoryInit ...
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Hardware memory configuration for this board is similar to that of Leaf Hill CRB. 

Regards, 
Mahesh.

myguitar, 2020-01-09:
Hi @shehamb ,

Below link may help your issue.
https://slimbootloader.github.io/how-tos/configure-memory-down.html

As your board is detected as LeafHill, you need to update Platform/ApollolakeBoardPkg/CfgData/CfgData_Int_LeafHill.dlt with proper values.

shehamb, 2020-01-10:
Hi @myguitar , 
Thank you for the pointers. But i have already updated the .dlt file with the values (which i thought would be proper).

myguitar, 2020-01-10:
BoardID: 0x07
PlatformName: LeafHill
Your board is detected as LeafHill, but memory parameters are different from LeafHill's one.

LeafHill (from CfgData_Memory.dsc):
Profile parameter from CfgData - 0xB.
MemorySizeLimit from CfgData - 0x02.

Yours:
Profile parameter from CfgData - 0x19.
MemorySizeLimit from CfgData - 0x0.

Did you override some values in CfgData_Int_LeafHill.dlt delta file?

shehamb, 2020-01-13:
Yes, i did override the values in CfgData_Int_LeafHill.dlt. And the initial information you see in the log, is actually a debug message i have added. I just wanted to see what's the value of Profile and MemorySizeLimit before actually calling the FspMemoryInit(). So just to make sure that my customized values are getting used, i am overriding it once again like this in the function 
UpdateFspConfig(), in file Stage1BBoardInitLib.c

#ifdef CONGATEC_BOARD
	DEBUG ((DEBUG_INFO, "Setting Fspmcfg for LeafHill.\n"));
	/* DRAM Config settings */
	Fspmcfg->Package = 0x1;
	Fspmcfg->Profile = 0xB;
	Fspmcfg->MemoryDown = 0x1;
	Fspmcfg->DDR3LPageSize = 0x0;
	Fspmcfg->DDR3LASR = 0x0;
	Fspmcfg->ScramblerSupport = 0x1;
	Fspmcfg->ChannelHashMask = 0x36;
	Fspmcfg->SliceHashMask = 0x9;
	Fspmcfg->InterleavedMode = 0x2;
	Fspmcfg->ChannelsSlicesEnable = 0x0;
	Fspmcfg->MinRefRate2xEnable = 0x0;
	Fspmcfg->DualRankSupportEnable = 0x1;
	Fspmcfg->RmtMode = 0x0;
	Fspmcfg->MemorySizeLimit = 0x2000;
	Fspmcfg->LowMemoryMaxValue = 0x0;
	Fspmcfg->DisableFastBoot = 0x1;
	Fspmcfg->HighMemoryMaxValue = 0x0;
	Fspmcfg->DIMM0SPDAddress = 0x0;
	Fspmcfg->DIMM1SPDAddress = 0x0;
	Fspmcfg->Ch0_RankEnable = 0x1;
	Fspmcfg->Ch0_DeviceWidth = 0x1;
	Fspmcfg->Ch0_DramDensity = 0x2;
	Fspmcfg->Ch0_Option = 0x3;
	Fspmcfg->Ch0_OdtConfig = 0x0;
	Fspmcfg->Ch0_TristateClk1 = 0x0;
	Fspmcfg->Ch0_Mode2N = 0x0;
	Fspmcfg->Ch0_OdtLevels = 0x0;
	Fspmcfg->Ch1_RankEnable = 0x1;
	Fspmcfg->Ch1_DeviceWidth = 0x1;
	Fspmcfg->Ch1_DramDensity = 0x2;
	Fspmcfg->Ch1_Option = 0x3;
	Fspmcfg->Ch1_OdtConfig = 0x0;
	Fspmcfg->Ch1_TristateClk1 = 0x0;
	Fspmcfg->Ch1_Mode2N = 0x0;
	Fspmcfg->Ch1_OdtLevels = 0x0;
	Fspmcfg->Ch2_RankEnable = 0x1;
	Fspmcfg->Ch2_DeviceWidth = 0x1;
	Fspmcfg->Ch2_DramDensity = 0x2;
	Fspmcfg->Ch2_Option = 0x3;
	Fspmcfg->Ch2_OdtConfig = 0x0;
	Fspmcfg->Ch2_TristateClk1 = 0x0;
	Fspmcfg->Ch2_Mode2N = 0x0;
	Fspmcfg->Ch2_OdtLevels = 0x0;
	Fspmcfg->Ch3_RankEnable = 0x1;
	Fspmcfg->Ch3_DeviceWidth = 0x1;
	Fspmcfg->Ch3_DramDensity = 0x2;
	Fspmcfg->Ch3_Option = 0x3;
	Fspmcfg->Ch3_OdtConfig = 0x0;
	Fspmcfg->Ch3_TristateClk1 = 0x0;
	Fspmcfg->Ch3_Mode2N = 0x0;
	Fspmcfg->Ch3_OdtLevels = 0x0;
	Fspmcfg->RmtCheckRun = 0x0;
	Fspmcfg->MrcDataSaving = 0x0;
	Fspmcfg->MrcFastBoot   = 0x0;
	Fspmcfg->PeriodicRetrainingDisable   = 0x1;

	CopyMem(Fspmcfg->Ch0_Bit_swizzling, &Ch0_Bit_swizzling,
		sizeof(Ch0_Bit_swizzling));
	CopyMem(Fspmcfg->Ch1_Bit_swizzling, &Ch1_Bit_swizzling,
		sizeof(Ch1_Bit_swizzling));
	CopyMem(Fspmcfg->Ch2_Bit_swizzling, &Ch2_Bit_swizzling,
		sizeof(Ch2_Bit_swizzling));
	CopyMem(Fspmcfg->Ch3_Bit_swizzling, &Ch3_Bit_swizzling,
		sizeof(Ch3_Bit_swizzling));
#endif

mauricema, 2020-01-13:
Several comments/questions here:
- Is your memory configuration exactly the same as LeafHill ?   or it is different ?  If it is the same, please just try the default without any MRC parameter overriding.
- Do you have any UEFI BIOS that can boot on your board ?  If so, is there any way that you can capture the MRC parameter used for your board?
- Since it hangs in MRC code.  Mostly, it indicates the MRC parameter does not match your hardware platform.  You might want to check with your hardware engineers on these parameters.
- If you have directly Intel support,  you can also request a DEBUG FSP binary from them to get more useful information to analyze the memory training failure.

mauricema, 2020-01-22:
Do you have any further question ?  If no, I plan to close this pretty soon.

shehamb, 2020-01-22:
Hello @mauricema , 
Apologies for delay in replying. I wanted to try out few things before i post something here. 

Yes, memory configuration is same on both the boards (at least that's what i could make out by referring to the specs):
(4x 32bit LPDDR4 channel, 8 GB, 2400 MT/s). 
One difference being, in the SoC version:
Congatec board CPU : APL-D0
LeafHill board CPU : APL-B1 

So, i tried a fresh build and tried it on the board. But it hangs at the same place: 

*******************************************************************
============= Intel Slim Bootloader STAGE1A =============
SBID: SB_APLI
ISVN: 001
IVER: 000.005.001.000.14666
SVER: 0000000000000000
FDBG: BLD(D) FSP(R)
FSPV: ID($APLFSP$) REV(01040301)
Loader global data @ 0xFEF01D4C
Run  STAGE1A @ 0xFEF80000
Load STAGE1B @ 0xFEF88000
No BtGuard verification !

============= Intel Slim Bootloader STAGE1B =============
CPU : APL-D0 [4C @ 1600MHz: premium SKU], uCode rev.2E
Early GpioInit for EMMC
MMC global data init
Use SDMA instead of ADMA2
MMC Phase 1 init
MmcGetOcr Fail Status = 0x80000007
Early MMC Init Fails @ 0xFE700000 - Status(Device Error)
SpiInstance = FEF033B0
PchSpiBase at 0x00000D02
ScSpiBar0 at 0xD2000000
Append public key hash into store: Unsupported
Load EXT CFG Data @ 0xFEF03BE4:0x0CD8 ... Success
HASH verification for usage (0x00000200): Success
RSA  verification for usage (0x00000200): Success
GPIO based PID detection is disabled
Get Embedded BoardId
Initialize RTC with default values
BOOT: BP0
MODE: 0
BoardID: 0x07
PlatformName: LeafHill
BootPolicy : 0x00000010
Memory Init
Looking for MRC saving data
MRC data NOT found
Load memory parameters from CfgData.
Load general parameter from CfgData.
Call FspMemoryInit ...
*******************************************************************

Yes, Congatec board has a UEFI BIOS and in fact i am able to boot up VxWorks kernel on it. But i didn't quite figure out about how to dump the DRAM configuration.

I am not getting much support from the Congatec side. So i just have to keep changing the DRAM configuration and keep trying. Meantime, i have requested Intel to provide me the debug FSP library. 

Regards, 
Mahesh.

mauricema, 2020-02-19:
Do we have any update on this ?  Can we close it if it is a memory FSP UPD configuration issue ?

shehamb, 2020-02-20:
Yes, it's a memory FSP UPD configuration issue. So closing this.

closed 2020-02-20 by shehamb

andreyv1978, 2020-04-09:
Hi shehamb

Are you managed to sort out the memory init issue? I have a similar problem with our board

Andrey

shehamb, 2020-04-09:
Hi Andrey,

No, i couldn't resolve the issue on the Congatec board. However, when i
quickly checked your issue, it looks similar. I just received UP2 board few
days ago. And i quickly tried SBL on it. But it didn't work. I also faced
blank screen issue. From the issue comments, it seems like you were able to
resolve this issue?

Regards,
Mahesh.

andreyv1978, 2020-04-09:
yes, use updated version it should work
but, now I am having the same issue with our company board , and the memory setting are correct according to the schematics

andreyv1978, 2020-04-09:
if you need changes to the UP2 board SKU=3 let me know

shehamb, 2020-04-09:
i am using sku=2. but thank you.

Getting stuck up with the FSP memory settings can be very frustrating. Our
board is going to be ready in another month or so. So, just to be on the
safer side, i am exploring the UEFI bios as well. And so far, it has worked
consistently on two boards, Leaf Hill CRB and UP2 board.

Have you tried the UEFI bios on your company board?

andreyv1978, 2020-04-09:
i still have time to sort out slimboot, i am trying to build myself FSP binaries, but it is a big project with a lot of dependencies 
can you send me a link to uefi bios to try out ? i've seen coreboot should work on LeafHill

shehamb, 2020-04-09:
Please check for this text file attached herewith. I used it as a reference
to set up the UEFI on my linux machine.

I was planning to use Coreboot, and started off it with. But for the final
stitching part, I could never get the required binaries. It needs special
access to get it from Intel's platform software kit.
Did you try Coreboot?

shehamb, 2020-04-09:
Sorry, forgot to attach the file.

===============================================================================
UEFI Firmware Project for Intel Atom(R) Processor E3900 Series platforms
Build Instructions for Open Source Release, Version 0.71, Aug 30th, 2018
================================================================================

================================================================================
                           INDEX 
================================================================================
1.  HOW TO CREATE A FULL SOURCE TREE
2.  HOW TO BUILD (WINDOWS ENVIRONMENT)
3.  HOW TO BUILD (LINUX ENVIRONMENT)

NOTE REGARDING UP SQUARED PLATFORM: The firmware provided by this project for the
UP Squared maker board is not based on the official manufacturer's firmware. This
project is for experimentation and is not supported by the manufacturer (Aaeon).
Flashing this firmware on the UP Squared will void the manufacturer's warranty.
Thanks to Aaeon for their cooperation and providing platform porting information. 

================================================================================
1.                           HOW TO CREATE A FULL SOURCE TREE 
================================================================================
1) Create a new folder (directory) on the root of your local development machine
   for use as your workspace (this example uses "C:\MyWorkspace").

   NOTE: Some code paths are very long, and placing the working directory too
         deep below the ROOT directory may cause the path to be longer than the 
         operating system or file system's maximum path length.

   NOTE: Below git commands are verified under Git Bash environment of Git for 
         Windows.

2) Get devel-IntelAtomProcessorE3900 branch from GitHub* with the following command.
   a. git clone https://github.com/tianocore/edk2-platforms.git -b devel-IntelAtomProcessorE3900
   b. Enter folder edk2-platforms by running "cd edk2-platforms"
   c. git checkout 2c15135d5d766b276bc1a6d41385230d4aeeb7ea
   
3) Get EDK II packages
   a. git clone -b vUDK2018 https://github.com/tianocore/edk2.git
      Once the download completes, all EDK II packages and files will be found
      under the "C:\MyWorkspace\edk2" directory.

4) Get the Intel(R) Firmware Support Package(Intel(R) FSP)
   a. git clone https://github.com/IntelFsp/FSP.git
   b. Enter folder FSP by running "cd FSP".
   c. git checkout 667eb3edc5d5cd562c33d9fbe89b9bee6a8a80ae
   d. Copy the ApolloLakeFspBinPkg to the folder
      "C:\MyWorkspace\edk2-platforms\Silicon\BroxtonSoC\BroxtonFspPkg".

5) Install UEFI UNDI driver:

5.1) Install UEFI UNDI driver for Intel i210 Ethernet Controller
   Download PREBOOT.EXE Version 21.1 from the "Intel® Ethernet Connections 
   Boot Utility, Preboot Images, and EFI Drivers" page.
     https://downloadcenter.intel.com/download/19186/Intel-Ethernet-Connections-Boot-Utility-Preboot-Images-and-EFI-Drivers
   Windows user:
   a. Install PREBOOT.EXE into the default folder (C:\Intel21.1).
   b. Copy the UEFI x64 PCI-E gigabit driver (C:\Intel21.1\APPS\EFI\EFIx64\E7320X3.EFI) 
      to the platform package directory below, create the folder if it does not exist:
      "C:\MyWorkspace\edk2-platforms\Platform\BroxtonPlatformPkg\Common\Binaries\UNDI\I210PcieUndiDxe"
   Linux user:
   a. Use command "unzip PREBOOT.EXE" to extract UEFI x64 PCI-E gigabit driver 
      (preboot/APPS/EFI/EFIx64/E7320X3.EFI)
   b. Copy E7320X3.EFI to platform package directory below, create the folder
      if it does not exist:
      "~/src/MyWorkspace/edk2-platforms/Platform/BroxtonPlatformPkg/Common/Binaries/UNDI/I210PcieUndiDxe"

5.2) Install UNDI driver for AX88179 USB-to-LAN adapter
   a. Download AX88179_178A_UEFI_V2.8.0.zip from http://www.asix.com.tw/FrootAttach/driver/AX88179_178A_UEFI_V2.8.0.zip.
[... 186 lines cut ...]
           ./BuildBIOS.sh /MX /C /L Broxton Release

        Leaf Hill CRB FAB D:  
          To build release version 64-bit BIOS for Leaf Hill CRB,
            ./BuildBIOS.sh /LH /D Broxton Release

          To build debug version 64-bit BIOS for Leaf Hill CRB,
            ./BuildBIOS.sh /LH /D Broxton Debug
     
          To build BIOS which disables SPI Descriptor Region and TXE region access permission for host CPU.
            ./BuildBIOS.sh /LH /D /L Broxton Release

        UP Squared:    
          To build release version 64-bit BIOS for UP Squared,
            ./BuildBIOS.sh /UP /A Broxton Release

          To build debug version 64-bit BIOS for UP Squared,
            ./BuildBIOS.sh /UP /A Broxton Debug
     
          To build BIOS which disables SPI Descriptor Region and TXE region access permission for host CPU.
            ./BuildBIOS.sh /UP /A /L Broxton Release

   3) After the build successfully completes, the 8MB firmware binary image will
      be located in the following location on your local hard drive:
      "~/src/MyWorkspace/edk2-platforms/Platform/BroxtonPlatformPkg/Common/Tools/Stitch/"

andreyv1978, 2020-04-09:
hm, exactly what i am building right now, and it is a bit of nightmare on my Linux VM and Windows PC, you managed to build binary according to the build instructions ?
coreboot is my plan B, never tried it

shehamb, 2020-04-09:
Yes, i followed the build steps for linux.

But here are my 2 cents:

provided you've created the codebase at ~/Myworkspace

open a new terminal window

cd Myworkspace
echo $WORKSPACE //make sure it's empty. if not, close all the open terminal
windows and start again.
make -C edk2/BaseTools

cd edk2
. edksetup.sh BaseTools/
source edksetup.sh

cd ../edk2-platforms/
./BuildBIOS.sh /LH /D Broxton Debug

andreyv1978, 2020-04-09:
I still have the same silly error from python:
Processing meta-data .... 

build.py...
 : error C0DE: Unknown fatal error when processing [/home/andreyv/project/FSP/edk2/MdePkg/MdePkg.dec]
	
(Please send email to edk2-devel@lists.01.org for help, attaching following call stack trace!)

(Python 2.7.17 on linux2) Traceback (most recent call last):
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2496, in Main
    MyBuild.Launch()
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2229, in Launch
    self._MultiThreadBuildPlatform()
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/BinWrappers/PosixLike/../../Source/Python/build/build.py", line 2004, in _MultiThreadBuildPlatform
    self.Progress
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/AutoGen/AutoGen.py", line 184, in __new__
    if not AutoGenObject._Init(Workspace, MetaFile, Target, Toolchain, Arch, *args, **kwargs):
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/AutoGen/AutoGen.py", line 417, in _Init
    for key in BuildData.Pcds:
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Workspace/InfBuildData.py", line 876, in _GetPcds
    self._Pcds.update(self._GetPcd(MODEL_PCD_FEATURE_FLAG))
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Workspace/InfBuildData.py", line 991, in _GetPcd
    Value = GuidValue(TokenSpaceGuid, self.Packages, self.MetaFile.Path)
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Common/Misc.py", line 833, in GuidValue
    GuidKeys = P.Guids.keys()
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Workspace/DecBuildData.py", line 270, in _GetGuid
    RecordList = self._RawData[MODEL_EFI_GUID, self._Arch]
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Workspace/MetaFileParser.py", line 241, in __getitem__
    self.Start()
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Workspace/MetaFileParser.py", line 1742, in Start
    self._SectionParser[self._SectionType[0]](self)
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Workspace/MetaFileParser.py", line 44, in MacroParser
    Parser(self)
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Workspace/MetaFileParser.py", line 2009, in _PcdParser
    IsValid, Cause = CheckPcdDatum(ValueList[1], ValueList[0])
  File "/media/sf_vmshared/projects/FSP/edk2/BaseTools/Source/Python/Common/Misc.py", line 1865, in CheckPcdDatum
    Printset = set(string.printable)
AttributeError: 'module' object has no attribute 'printable'

Not sure if you seen it on your PC
I am using Ubuntu 18.04 but i hope it should not be a problem, it might have different python packages

shehamb, 2020-04-09:
Couple of points to revisit:

- i am using Ubuntu 16.04 (VM). it shouldn't matter, i think.
- i have noticed that you're using the "shared" folder from windows?
perhaps, it'll be a good idea to create a folder locally and then set up
the codebase in that folder.
- did you set up the tools first? so these are the steps i followed:
1. open a new terminal window.
2. cd Myworkspace (folder where you've edk2 and edk-platforms checked out)
3. echo $WORKSPACE //make sure it's empty. if not, close all the open
terminal windows and start again.
4. make -C edk2/BaseTools
5. cd edk2
6. . edksetup.sh BaseTools/
7. source edksetup.sh
8. cd ../edk2-platforms/
9. ./BuildBIOS.sh /UP /A Broxton Debug

andreyv1978, 2020-04-09:
i am afraid the problem in python 2.7.17 instead of 2.5.x 
will sort it out first

mauricema, 2020-04-10:
For new board enabling, unfortunately FSP memory parameter is always required.  And it has to match the board design.  Otherwise, the FSP might hang.  This same FSP memory configuration step will be required for all boot solutions using FSP including open sourced APL EDKII, coreboot, SlimBoot, etc.   

Regarding to UP2 board, did you try the latest code ?  There are some related PR merged for the board to address the display port issue and DDR ID detection issue.  Maybe you can give a try.

shehamb, 2020-04-13:
Thank you Maurice. I have just checked out latest sbl and am getting some
build error. I have created issue for the same.

andreyv1978, 2020-04-14:
Hi shehamb 

I've managed to create UEFI build on the clean VM Ubuntu 16.04. However, still hang in FspMemoryInit. 
You mentioned the UEFI build passed FspMemoryInit on your board ?

Andrey

shehamb, 2020-04-14:
So, i checked out and built the latest sbl for Up2 platform. But i am not
able to boot it up. I don't see anything on the console and the monitor.

andreyv1978, 2020-04-14:
and the UEFI bios build working on your board not on the UP2 board ?

shehamb, 2020-04-14:
UEFI BIOS ->  works on Leaf Hill and UP2

SBL -> works on Leaf Hill, doesn't on UP2

andreyv1978, 2020-04-14:
'UEFI BIOS ->  works on Leaf Hill and UP2'
LeafHill it is your customized board or Intel dev board?

shehamb, 2020-04-14:
Leaf Hill is Intel dev board.

andreyv1978, 2020-04-14:
Thank you, now everything makes sense 
I should look into my memory settings again

mauricema, 2020-04-14:
With SBL on UP2,  can you share your debug log ?

Thanks
Maurice

shehamb, 2020-04-14:
I don't see anything on the console.

mauricema, 2020-04-14:
Did you miss the stitching parameter by any change  ?
python Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i <BIOS_IMAGE_NAME> -s Outputs/apl/Stitch_Components.zip -o <SBL_IFWI_IMAGE_NAME> -p 0xAA00000E

The “-p 0xAA00000E” is very important.

Thanks
Maurice

<mailto:notifications@github.com%0b>> <mailto:notifications@github.com>>

shehamb, 2020-04-14:
Sorry, i had missed out the option  -p 0xAA00000E.

With this option, i can see the board is trying to boot up. But it's
getting stuck up at:
ASSERT_EFI_ERROR (Status = Not Found)
ASSERT [Stage2]
/home/mahesh/up2/SBL/slimbootloader/Platform/ApollolakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c(1913):
!EFI_ERROR (Status)

STAGE_2: System halted!

I am enclosing the log for your reference.

BtGuard: VB : 0, MB : 0

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_APLI
ISVN: 001
IVER: 000.005.001.000.16665
SVER: 359A9465DCC12A18-dirty
FDBG: BLD(D IA32) FSP(R)
FSPV: ID($APLFSP$) REV(01040301)
Loader global data @ 0xFEF71D44
Run  STAGE1A @ 0xFEF00000
Load STAGE1B @ 0xFEF80000
No BtGuard verification !

============= Intel Slim Bootloader STAGE1B =============
CPU : APL-D0 [4C @ 1600MHz: premium SKU], uCode rev.3C
Early GpioInit for EMMC
MMC global data init
Use SDMA instead of ADMA2
MMC Phase 1 init
SpiInstance = FEF7B340
PchSpiBase at 0x00000D02
ScSpiBar0 at 0xD2000000
Append public key hash into store: Unsupported
Load EXT CFG Data @ 0xFEF7BBE4:0x0CD8 ... Success
HASH verification for usage (0x00000200) with Hash Alg (0x1): Success
SignType (0x2) SignSize (0x100)  SignHashAlg (0x1)
RSA verification for usage (0x00000200): Success
MemSkuId 0x05, FabId 0x00
BOOT: BP0
MODE: 0
BoardID: 0x0E
PlatformName: UP2
BootPolicy : 0x00000010
Memory Init
Looking for MRC saving data
MRC data NOT found
Load memory parameters from CfgData.
UP2 memory SKU ID is 0x5
Load general parameter from CfgData.
Call FspMemoryInit ... Success
Loader global data @ 0x7ABBFD70
Load page table from memory @ 0x7A074000
Remapping Stage to 0x7A077000
Switch to memory stack @ 0x7ABFFF00
Stage1 stack: 0x2000 (0xFF4 used)
Stage1 heap: 0xE000 (0x4CD4 used)
Call FspTempRamExit ... Success
CSE: MBP Data HOB not valid!
MRC: [v0.56.89.24]
Memory FSP @ 0x7AC00000
Memory TOP @ 0x7A074000
HASH verification for usage (0x00000002) with Hash Alg (0x1): Success
Loaded STAGE2 @ 0x7AB32000

============= Intel Slim Bootloader STAGE2 =============
Unmapping Stage
[... 149 lines cut ...]
VbtFileBuffer->HeaderVbtSize = 0x1506
Not extended VBT
IgdOpRegion ended
  IPCBar0Address Address = 0xFE042000, Len = 0x00002000
  SSRAMBar0Address Address = 0xFE900000, Len = 0x00002000
  P2SBBaseAddress Old = 0xFFFFFFFF, New = 0xD0000000
  BIOS MAIL-BOX Data= 0xFED17080
  BIOS MAIL-BOX Interface= 0xFED17084
  GPIO0Addr Address = 0xD0C50000, Len = 0x0000076C
  GPIO1Addr Address = 0xD0C40000, Len = 0x00000764
  GPIO2Addr Address = 0xD0C70000, Len = 0x00000674
  GPIO3Addr Address = 0xD0C00000, Len = 0x00000654
  RpAddress Port[0] Address = 0x00140000
  RpAddress Port[1] Address = 0x00140001
  RpAddress Port[2] Address = 0x00130000
  RpAddress Port[3] Address = 0x00130001
  RpAddress Port[4] Address = 0x00130002
  RpAddress Port[5] Address = 0x00130003
ACPI Init
Updated Hpet BaseAddressLower32Bit = 0xFED00000, EventTimerBlockId = 0x8086A701

ASSERT_EFI_ERROR (Status = Not Found)
ASSERT [Stage2] /home/mahesh/up2/SBL/slimbootloader/Platform/ApollolakeBoardPkg/Library/Stage2BoardInitLib/Stage2BoardInitLib.c(1913): !EFI_ERROR (Status)

STAGE_2: System halted!

mauricema, 2020-04-14:
SVER: 359A9465DCC12A18-dirty
Looks like it is the modified code.  I cannot map it to the SBL open source code.
Can you do a clean clone for the latest code from github and then build and try ?

shehamb, 2020-04-14:
Clean clone did the trick! Thank you.

Is there any way to get the uefi shell on sbl? my final aim is to boot the
VxWorks kernel through the VxWorks EFI Loader Module. I was able to do so
in UEFI bios, through UEFI shell.

mauricema, 2020-04-14:
Yes, you can.  
To do this, you need to build UEFI payload binary and integrate it into SBL.  
Please follow instructions below:
https://slimbootloader.github.io/how-tos/boot-windows.html?highlight=uefi%20payload
BTW, you can use the latest EDK2 repo instead of the mentioned commit for UEFI Payload build.

shehamb, 2020-04-15:
Thank you. I am able to boot VxWorks now.
I have few more questions though:
How do i change BIOS settings? Like, i wish to change few settings:

   - Ensure that failed PCI enumeration for the FPGA will not cause its
   upstream port to be disabled and powered down permanently.
   - Ensure that the PMIC options are set for the TI PMIC (I2C instead of
   SVID ).
   - Allow option for enabling Direct Connect Interface (DCI) Debug via the
   USB3 port .
   - Disable Secure Boot features

Where do i need to make changes for these things?

Also, how do i get 'usual' BIOS settings (which i am through UEFI and  up2
vendor provided BIOS). I am using UPA1AM50.bin as input for generating the
final sbl. With this, when i get into the BIOS settings, i see nothing
under device manager.

Regards,
Mahesh.

mauricema, 2020-04-15:
Bootloader is designed for fixed configuration. It does not offer the same like BIOS to provide setup screen to change option at runtime.   SBL allows you to change options at build time, or change options at run time through capsule update mechanism.  You can use "ConfigEdit.py" in SBL to explore the available config items. 
For the options you mentioned, it needs to be customized to handle those cases.

andreyv1978, 2020-04-17:
> Thank you. I am able to boot VxWorks now. I have few more questions though: How do i change BIOS settings? Like, i wish to change few settings: - Ensure that failed PCI enumeration for the FPGA will [...]

Hi Mahesh

Are you managed to boot VxWorks  using UEFI payload build from EDKII or OsLoader ?
I am trying to build UEFI Payload on Ubuntu 16.04 and it is failing all the time

Andrey

shehamb, 2020-04-20:
Hi Andrey,

Yes, i was able to boot VxWorks using UEFI payload. I followed the
instructions as mentioned by Maurice. It's available on the sbl webpage.

and i am also using 16.04.

Regards,
Mahesh.

andreyv1978, 2020-04-21:
I am failing to build on Ubuntu 16.04 using a following command
build -a IA32 -a X64 -p UefiPayloadPkg/UefiPayloadPkgIa32X64.dsc -b DEBUG -t GCC48 -D BOOTLOADER=SBL

not really helpfull errors, and not sure what else is missing

GNUmakefile:930: recipe for target '/home/andreyv/FSP/edk2/Build/UefiPayloadPkgX64/DEBUG_GCC48/IA32/MdePkg/Library/BaseLib/BaseLib/OUTPUT/Ia32/LongJump.obj' failed
make: *** [/home/andreyv/FSP/edk2/Build/UefiPayloadPkgX64/DEBUG_GCC48/IA32/MdePkg/Library/BaseLib/BaseLib/OUTPUT/Ia32/LongJump.obj] Error 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/andreyv/FSP/edk2/Build/UefiPayloadPkgX64/DEBUG_GCC48/IA32/MdePkg/Library/BaseLib/BaseLib]

build.py...
 : error F002: Failed to build module
	/home/andreyv/FSP/edk2/MdePkg/Library/BaseLib/BaseLib.inf [IA32, GCC48, DEBUG]

shehamb, 2020-04-22:
I tried this build again, and i was able to do without any errors. Did you
set the build environment ? Take a look at this screenshot:
[image: image.png]

and here is the screenshot of the beginning of the build:
build -a IA32 -a X64 -p UefiPayloadPkg/UefiPayloadPkgIa32X64.dsc -b RELEASE
-t GCC5 -D SBL

[image: image.png]

andreyv1978, 2020-04-22:
bingo, thank you 
your build configuration are working :) 

works:
build -a IA32 -a X64 -p UefiPayloadPkg/UefiPayloadPkgIa32X64.dsc -b RELEASE
-t GCC5 -D SBL

not:
build -a IA32 -a X64 -p UefiPayloadPkg/UefiPayloadPkgIa32X64.dsc -b DEBUG -t GCC48 -D BOOTLOADER=SBL

maybe the GCC compiler makes a difference. Thank you for your help