slimbootloader/slimbootloader #537: Build error for apl platform

shehamb, 2020-01-20:
Hello, 
i just checked out the slimbootloader repository. And i did a build for apl platform using:
python BuildLoader.py build apl

But it throws this error:
- Done -
Build end time: 21:42:45, Jan.19 2020
Build total time: 00:00:30

Patching STAGE1A
Patched offset 0x00007FFC:[90909090] with value 0xFFFFA000  # Patch BFV
Patched offset 0x00002000:[00000000] with value 0xFFFFCBD0  # Patch Stage1A Entry
Patched offset 0x00002004:[00000000] with value 0xFFFFA380  # Patch Module Base
Patched offset 0x00005AC8:[FF000000] with value 0xFFFFA064  # Patch VerInfo
Patched offset 0x00005AC0:[FF000000] with value 0xFFFFA2AC  # Patch PcdBase
Patched offset 0x00007FF8:[90909090] with value 0xFFFFA0A4  # Patch FlashMap
Patched offset 0x00005AC4:[FF000000] with value 0xFFFFA1B4  # Patch HashStore
Patching STAGE1B
Patched offset 0x00000000:[00000000] with value 0xFEF94740  # Patch Stage1B Entry
Patched offset 0x00000004:[00000000] with value 0xFEF88070  # Patch Stage1B Base
Patched offset 0x0000FE20:[FF000000] with value 0xFEF9897C  # Patch Internal CfgDataBase
Patching STAGE2
Patched offset 0x00000000:[00000000] with value 0x01013A30  # Patch Stage2 Entry
Patched offset 0x00000004:[00000000] with value 0x01000070  # Patch Stage2 Base
Patched offset 0x0001A114:[FF000000] with value 0x01022A2C  # Patch VBT
Patched offset 0x0001A118:[FF000000] with value 0x0101BEBC  # Patch ACPI
Patched offset 0x0001A110:[FF000000] with value 0x0102444C  # Patch Logo
Traceback (most recent call last):
  File "BuildLoader.py", line 1306, in <module>
    main()
  File "BuildLoader.py", line 1303, in main
    args.func(args)
  File "BuildLoader.py", line 1250, in cmd_build
    Build(board).build()
  File "BuildLoader.py", line 1125, in build
    self.post_build()
  File "BuildLoader.py", line 1204, in post_build
    self.create_bootloader_image (layout_name)
  File "BuildLoader.py", line 883, in create_bootloader_image
    align_pad_file(src_path, dst_path, val, mode, pos)
  File "/home/mahesh/congatec/slimbootloader-master/BootloaderCorePkg/Tools/BuildUtility.py", line 638, in align_pad_file
    (os.path.basename(src), srclen, val))
Exception: File 'PAYLOAD.lz' size 0x240F3 is greater than padding size 0x23000 !

Build for qemu platform is successful. Is there something i am missing here?

Regards, 
Mahesh.

myguitar, 2020-01-20:
Hi Mahesh,

The generated binary size varies depending on build toolchain version. I think some of your build toolchain might have been updated - mainly gcc version. Can you share you toolchain version info? You can find the version info at the beginning of your build log.

To resolve the size exceed issue, you can manually adjust the region size in BoardConfig.py file.
In your case, you can increase the PAYLOAD region size in Platform/Apollolake/BoardConfig.py according to build mode.
self.PAYLOAD_SIZE         = 0x0001F000
if self.RELEASE_MODE == 0:
  self.PAYLOAD_SIZE    += 0x00004000

Best Regards,
Aiden

shehamb, 2020-01-20:
Hi Aiden,

Here is the toolchain version info:

mahesh@ubuntu:~/congatec/slimbootloader-master$ python BuildLoader.py build
apl
WARNING: Python version 2.7.12 is unsupported, potential build issue might
encounter !
         Please consider installing and using Python 3.6 or above to launch
build script !

Using /usr/bin/python, Version 2.7.12
Using GCC5, Version 5.4.0
Using /usr/bin/openssl, Version OpenSSL 1.0.2g  1 Mar 2016
Using /usr/bin/nasm, Version NASM version 2.11.08
Using /usr/bin/git, Version git version 2.7.4
Build [apl] ...

And increasing the PAYLOAD region size did resolve the issue!

Thank you.
Regards,
Mahesh.

myguitar, 2020-01-20:
Hi Mahesh,

Thanks for your information. The gcc 5.4 is somewhat old version. I recommend you to use gcc 7 or 8. And, please use python 3.6 or higher version as well.

Best Regards,
Aiden

shehamb, 2020-01-20:
Sure Aiden. Thanks for the pointers. I will close this issue now.

closed 2020-01-20 by shehamb