slimbootloader/slimbootloader #549: how to control PMU_PLTRST_N in slimbootloader

Michael-Park, 2020-01-30:
Hello,

Please refer to plots below.
The first one was captured from PMU_PLTRST_N (pin# : AG57) of Leaf Hill CRB (Apollo Lake-I) with Intel BIOS.
After power on, PMU_PLTRST_N is released (high), then 4 seconds later, it is asserted (low) again for 4 seconds and then it is released again.
[image]

The second one was captured from same pin with FSP + Slimbootloader.
As you can see, the low/high transition occurs only 1 time.
[image]

With the FSP + Slimbootloader, PCIe device such as PCIe-to-SATA multiplier was not detected by Slimbootloader sometimes. This problem didn't happen with Intel BIOS.
So my customer manipulated the pin, PMU_PLTRST_N, to occur low/high transition twice time using external HW switch array by force, then this problem didn't happen even in FSP + Slimbootloader.

Based on this test result, I need to know how to manipulate PMU_PLTRST_N in Slimbootloader.
Would you please help me to get a sample code or guide how to implement it to Slimbootloader?

Best Regards,
Michael Park

mauricema, 2020-01-30:
Hi, Michael,

According to the EDS:
PMU_PLTRST_N Platform Reset: This signal is used to reset devices on the platform (such as SIO,LAN, processor, and so forth.). This signal is asserted during power-up and when S/W initiates a hard reset sequence through the Reset Control register (I/O port CF9h).  The SoC drives PLTRST# active a minimum of 1ms when initiated through the Reset Controlregister (I/O port CF9h).  

So I think it will be possible to issue a CF9 reset to assert PMU_PLTRST.   

The reason why you saw PMU_PLTRST asserted after the boot in CRB BIOS might be due to the actual reset request from BIOS.  Slimboot loader does not issue reset in boot flow unless FSP requests so or in some other special conditions.

Also if your issue is related to PCIe device detection, it might be related to the PFET.  This is a FSP UPD called: StartTimerTickerOfPfetAssert.  It will be used to determine how long the FSP will wait before starting the PCIe RP init.  

For experiment, you can try to change the following in SBL code to see if it helps:
Fspmcfg->StartTimerTickerOfPfetAssert = 20000;
to: 
Fspmcfg->StartTimerTickerOfPfetAssert = 0;
It will force 100ms wait.

Thanks
Maurice

Michael-Park, 2020-02-05:
Hi Maurice,

My customer tested as you recommended.
They changed 2 settings as below.
1. "FSP MemoryInit Settings > PCIE SLOT Power Enable Assert Time -PFET" using BCT tool
2. Fspmcfg->StartTimerTickerOfPfetAssert in "\slimbootloader\Platform\ApollolakeBoardPkg\Library\Stage1BBoardInitLib\Stage1BBoardInitLib.c"

They set the value to '0', '20000', '1000000000' and checked the PCIe detection issue.
But there was no difference between them. The PCIe detection issue was not improved.

As I mentioned, the PCIe detection issue didn't happen when my customer assert PMU_PLTRST using external HW switch array by force. So they want to know how to manipulate this pin (assert PMU_PLTRST or make some wait) in slimbootloader.

Best Regards,
Michael Park

mauricema, 2020-02-06:
OK.   Did they try to issue CF9 reset as I mentioned to trigger a reset to change PMU_PLTRST ?   EX:  IoWrite8 (0xCF9, 0x06),  or even try IoWrite8 (0xCF9, 0x0E) ?

I don't believe SBL has direct control over the PMU_PLTRST pin.

mauricema, 2020-02-19:
Any update on this ?

Michael-Park, 2020-02-21:
Sorry for my late update.
I got a feedback from my customer that they fixed this issue by modifying the firmware of PCIe to SATA multiplier. After updating the firmware, the PCIe detection issue has not been reproduced anymore.
Thank you for your help.

closed 2020-02-21 by Michael-Park

mauricema, 2020-02-21:
Thank you for the update!  Glad to know it was resolved.  :)