slimbootloader/slimbootloader #577: [APL] SBL stuck in FWU mode

umaplehurst, 2020-02-18:
I don't have a way to reproduce this, so this is somewhat hypothetical:

a) we have a board that was working just fine, we powered it off,
b) some time later, it was powered on, it refused to boot, and it looks like it was "stuck" in SBL FW update mode -- please see attached log,
c) it was powered off again, left alone for some time, and then on subsequent power on, it was fine again, booting normally,

Our board doesn't have an RTC battery. So, I guess that the RTC NVRAM will contain defaults or maybe junk after power has been removed for some time. I am wondering whether there is the possibility that RTC register 0x40 (which is currently used as the SBL APL FWU mode flag) was by purely random chance at the 0x5A value in b) above ... and after removing power again, RTC NVRAM was then reset to some other value, and the boot worked.

I can't find anything in the Apollo Lake EDS which says what the "default" RTC NVRAM values are meant to be after power loss. I am wondering if a suitable solution here is, if 0x40 is found to be 0x5A, it should be cleared immediately with a write by the SBL code to some other value; this way, even if FW update request might randomly happen due to corrupt RTC, after a single FW update failure, the next boot will work normally, and the board will not instead be "stuck" forever in FW update mode until power has been drained.

[noBoot.log](https://github.com/slimbootloader/slimbootloader/files/4219169/noBoot.log)
[reset-marker.txt](https://github.com/slimbootloader/slimbootloader/files/4219170/reset-marker.txt)

raghavag, 2020-02-18:
Hi,

Yes, the risk of weak RTC coin cell not sustaining the values in the CMOS registers is present. One way to handle that is to check RTC power status bit before consuming the CMOS register contents.
Please see the attached image as an example to implement above suggested solution.
[image]
 
Because of these reasons, we have switched over to sticky chipset register for signaling FW update. A power cycle will set the register to a known default. During boot, SBL will check this register and initialize the state machine which will further guide the FW update flow.

Also, there is an issue in the older code, Firmware update code will not clear the trigger in case of firmware update failure. This issue is also resolved in the latest code.

umaplehurst, 2020-02-19:
Thank you for confirming! I'll merge in the latest code, I think this will resolve all the issues :)

closed 2020-02-19 by umaplehurst