slimbootloader/slimbootloader #587: Error building qemu exercise using Docker

mgmillsa, 2020-03-14:
**From "Miscellaneous: Docker", I ran:**

`
>$ docker build -t sbl --network=host .
>$ docker run -it --rm --network=host --name=sbl -d -v $PWD:/work sbl
>$ docker exec -w /work sbl python BuildLoader.py build qemu
`

**This ended with the following error:**

...
`
make[1]: Entering directory '/work/BaseTools/Tests'
make[1]: Leaving directory '/work/BaseTools/Tests'
make: Leaving directory '/work/BaseTools'
fatal: could not create leading directories of '/Download/QemuFsp': Permission denied
Building QEMU FSP binaries from EDKII repo (Base Tag: edk2-stable201911)
Cloning EDKII repo ...
Traceback (most recent call last):
  File "/work/BootloaderCorePkg/Tools/PrepareBuildComponentBin.py", line 255, in <module>
    sys.exit(Main())
  File "/work/BootloaderCorePkg/Tools/PrepareBuildComponentBin.py", line 245, in Main
    BuildFspBins (qemu_repo_dir, sbl_dir, silicon_pkg_name, target)
  File "/work/BootloaderCorePkg/Tools/PrepareBuildComponentBin.py", line 167, in BuildFspBins
    Fatal ('Failed to clone FSP repo to directory %s !' % fsp_dir)
  File "/work/BootloaderCorePkg/Tools/PrepareBuildComponentBin.py", line 17, in Fatal
    raise Exception (msg)
Exception: Failed to clone FSP repo to directory /Download/QemuFsp !
Traceback (most recent call last):
  File "BuildLoader.py", line 1329, in <module>
    main()
  File "BuildLoader.py", line 1326, in main
    args.func(args)
  File "BuildLoader.py", line 1273, in cmd_build
    Build(board).build()
  File "BuildLoader.py", line 1131, in build
    self.pre_build()
  File "BuildLoader.py", line 1000, in pre_build
    raise Exception  ('Failed to prepare build component binaries !')
Exception: Failed to prepare build component binaries !
`

mgmillsa, 2020-03-14:
Is this the same issue from a couple years ago where it's trying to download qemu outside of the container? If so, let's get the documentation fixed. I'll even do it and create a pull request and escalate internally.

mgmillsa, 2020-03-14:
Downloading outside the container seems to be the issue. I got it working by copying the entire repo into the container. It's a little less elegant, but does the trick. It also has the side effect of the Output directory being owned by root. I'll continue to try to remedy that. Below are the Docker commands I used and the Dockerfile.

>$ docker build -t sbl --network=host .
>$ docker run -it --rm --network=host --name=sbl -d -v $PWD/Outputs:/Slim/Outputs sbl
>$ docker exec sbl python BuildLoader.py build qemu

```
FROM ubuntu:18.04

# Install build dependencies
RUN apt-get update && apt-get -y install sudo build-essential python python3 \
    uuid-dev nasm openssl gcc-multilib git m4 bison flex qemu-system-x86

# Install ACPICA Utilities
ADD http://ftp.br.debian.org/debian/pool/main/a/acpica-unix/acpica-tools_20160831-1+b1_amd64.deb /tmp
RUN cd /tmp && \
    dpkg -i acpica-tools_20160831-1+b1_amd64.deb && \
    apt-get install -f && \
    apt-get -y install python3-distutils

# Add a user account, give sudo access
# RUN useradd -m docker && echo "docker:docker" | chpasswd && adduser docker sudo
# USER docker

# Create working directory and copy in Slim Bootloader
WORKDIR /Slim
COPY . .
VOLUME ["/Slim/Outputs"]

RUN git config --global user.name "User" && git config --global user.email "user@example.com"
CMD /bin/bash
```

mauricema, 2020-03-16:
Thank you for investigating this issue.  
I believe it is the same issue.  SBL tries to clone EDKII QemuFsp repo and put it outside.  
Can you create a work space /work, and then clone sbl into /work/sbl ?  Then QemuFsp will be under /work/Download/QemuFsp.

mauricema, 2020-03-17:
Since this issue was addressed by documentation changes below:
https://github.com/slimbootloader/documentation/pull/67
Can we close this now ?

mgmillsa, 2020-03-17:
Yes, thank you.

closed 2020-03-17 by mgmillsa