slimbootloader/slimbootloader #612: Slim Bootloader failed to boot on UP2 Squared

andreyv1978, 2020-03-26:
Hi

I am a newbee in the customization of the bios and fw. Started relatively easy task to use SlimBootloader on UP2 Squared board. I've made a following steps:
1. Confirmed I can flash UPA1AM50.bin with SF100 and boot board with customized Yocto build or Ubuntu 18.04
2. I've followed the steps from https://slimbootloader.github.io/supported-hardware/up2.html and used UPA1AM50.bin, made stitching as instructed
3. The board reboot in the black screen, no access to bios or anything

Can anybody help what I am doing wrong ?

Andrey

mauricema, 2020-03-26:
Hi, Andrey,

Thank you for reporting this issue.

Can you hook up the serial port so that we can see more detailed info ?

Also is it possible for us to try your stitched image?   Since UP2 has many different memory configurations,  the default memory config might not match your board SKU.  It might be required to adjust the memory parameters in the DLT file.  Do you know which model you have ?  How much memory size ?

Thanks
Maurice

andreyv1978, 2020-03-26:
Hi Maurice

Thank you for the fast reply, I hope to receive serial UART tomorrow, and maybe would have a better idea what is wrong
I am using UP2  Squared with N4200  

How can i upload the original binary from the board ?

Andrey

andreyv1978, 2020-03-26:
[sbl_test.zip](https://github.com/slimbootloader/slimbootloader/files/4388483/sbl_test.zip)

jgutbub, 2020-03-26:
Hi Andrey,

Can you clarify what you mean by "board reboot in the black screen", does this mean that the display did turn on and just has a black screen or the display did not even turn on?

Thanks,
James

andreyv1978, 2020-03-26:
Hi James

The board appears to be ON, but there is no signal to the screen, because I do not have a serial cable can not check what is wrong.

Andrey

andreyv1978, 2020-03-26:
closed by mistake

mauricema, 2020-03-26:
I have looked into your image, and did find something wrong.
Did you following the stitching process for UP2 ? 
python Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i <BIOS_IMAGE_NAME> -s Outputs/apl/Stitch_Components.zip -o <SBL_IFWI_IMAGE_NAME> -p 0xAA00000E
Did you miss "-p 0xAA00000E" part?   I did not see that happen from your final stitched IFWI image.

andreyv1978, 2020-03-26:
yes,  might be uploaded the wrong image after some customizations
I would make a clean build tommorow and will upload it

andreyv1978, 2020-03-27:
clean build:

1. BuildLoader.py build apl
2. Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i <BIOS_IMAGE_NAME> -s Outputs/apl/Stitch_Components.zip -o <SBL_IFWI_IMAGE_NAME> -p 0xAA00000E

Attached stitched image

Result:
Black screen, no signal to monitor output

[sbl_ifwi.zip](https://github.com/slimbootloader/slimbootloader/files/4392580/sbl_ifwi.zip)

mauricema, 2020-03-27:
I tried your attached image and it did boot fine on my UP2 board.   And I can see serial port msg and SlimBoot logo.  Maybe you just need to wait for a while ?  The 1st boot will do full memory training and will it is going to take a while.

mauricema, 2020-03-27:
Also when you flash it, please do verification to ensure the image is fully updated into the flash.  I have seen case that DediProg did not program the image properly.

andreyv1978, 2020-03-27:
hm, can you tell me what the UP2 board spec you have ?
I've verified image and even downloaded it and verified again
The board is in ON with always blue light near the reset button, but the screen has no signal. I've waited more than 5 minutes, made a reset a few times, but the symptoms are the same (screen have nothing)
If I am flashing an original bios the board takes about a minute to get to the normal restart and boot
Unfortunately, the serial cable is in transit already 2 weeks, and not sure when I can get it :( 
Andrey

mauricema, 2020-03-27:
The board label says "UP - APL01 A1 0_0_0".  It is also 8GB memory.

andreyv1978, 2020-03-27:
Looks like I should wait for the serial cable to continue this
Do you have any other ideas? ?

mauricema, 2020-03-27:
I should take less than 1min for 1st boot.  I am not quite sure if your board is booting or not without serial.   There is one more thing might be we can try:
In Platform\ApollolakeBoardPkg\BoardConfig.py, change the following  to enable USB KB
        # BIT0:Serial  BIT1:USB KB
        # Support serial port input console by default
        self.CONSOLE_IN_DEVICE_MASK  = 0x00000003
        # BIT0:Serial  BIT1:GFX
        self.CONSOLE_OUT_DEVICE_MASK = 0x00000003
Then rebuild/stitching/flash.
Let it boot, wait for 1-2 mins
then you try to hit the keyboard "Caps Lock" key.  If the LED is ON, it means the SBL is running correctly.  Maybe just the video is not up...

mauricema, 2020-03-27:
Alos you can try to disable GRUB support to prevent it from booting existing OS image.
Changing below in BoardConfig.y
self.ENABLE_GRUB_CONFIG  = 0

andreyv1978, 2020-03-27:
tried following:
self.CONSOLE_IN_DEVICE_MASK = 0x00000003
self.CONSOLE_OUT_DEVICE_MASK = 0x00000003
self.ENABLE_GRUB_CONFIG = 0

"Caps Lock" key makes no light on
I have UART1 USB cable, also nothing from there. Unfortunately, UART1 can not give anything from the BIOS boot

Have to stop for today, will continue on Monday

mauricema, 2020-03-27:
So it looks like it did not boot on your board some how.

andreyv1978, 2020-03-30:
have a serial wired, but the log is bit short
Looks like stuck on 'Call FspMemoryInit..'

tried both configurations the output is the same

[fspmemoryinit-failed.txt](https://github.com/slimbootloader/slimbootloader/files/4402990/fspmemoryinit-failed.txt)

mauricema, 2020-03-30:
You log shows "UP2 memory SKU ID is 0x03" .  This SKU is not yet supported in SBL.  
If you look into Stage1BBoardInitLib.c,  you will see the code that prints this. We have support for SKU 0/1/2, but not 3.   I think  you can try to force the MemSkuId to be 2 (since you mentioned your board is 8GB), and then give a try.

andreyv1978, 2020-03-30:
added in Stage1BBoardInitLib.c
    PlatformData->ModuleIdInfo.Bits.MemSkuId = 2; (line 624)
before sittings of Fspmcfg

Output:
MemSkuId 0x03, FabId 0x00
BOOT: BP0
MODE: 0
BoardID: 0x0E
PlatformName: UP2
BootPolicy : 0x00000010
Memory Init
Looking for MRC saving data
MRC data NOT found
Load memory parameters from CfgData.
UP2 memory SKU ID is 0x2
Load general parameter from CfgData.
Call FspMemoryInit ...

It looks like the setting should work, but still stuck with FspMemoryInit 

How the PlatformData knows about MemSkuId ?

mauricema, 2020-03-30:
It reads it from the board GPIO 214 and 215.  
If it does not work, it means SKU 3 has different memory parameters and needs to be adjusted accordingly.  I am not sure what are the memory parameters for your board SKU.    Maybe you can try SKU 0 or 1 as well.

mauricema, 2020-03-30:
There is a very similar issue like this:
https://github.com/slimbootloader/slimbootloader/issues/525
It was addressed by using the proper memory parameters.

andreyv1978, 2020-03-30:
Thank you for updating, I would play around it tomorrow 
At least we know that the UPS-APLP4-A20-0864 board is not currently supported

andreyv1978, 2020-03-31:
Played around the settings of SKU 3 overriding some of them with SKU 0, it moved to the next stage
However, it looks I require correct schematics to sort it out
I feel the issue can be closed, because it required UP2 support to sort it out

[bios-pass-with-sku3.txt](https://github.com/slimbootloader/slimbootloader/files/4409690/bios-pass-with-sku3.txt)

mauricema, 2020-03-31:
Thank you for the updates!
:)  You board booted already!
If you put a Ubuntu 18.04 on USB driver, it should boot directly now!

andreyv1978, 2020-03-31:
hm, it was yocto 2.7.1 there, would try it tomorrow with Ubuntu
what about message?
'Failed to get Graphics Info HOB from FSP '
also it was no display output, which actually match the message

mauricema, 2020-03-31:
the GFX message indicates FSP did not initialize the SOC GFX properly.  
Did you change anything in config to disable GFX ?

andreyv1978, 2020-04-01:
I've used a following settings to boot it up:
        Fspmcfg->DualRankSupportEnable = 1;
        Fspmcfg->Ch0_RankEnable        = 1;
        Fspmcfg->Ch0_DramDensity       = 2;
        Fspmcfg->Ch1_RankEnable        = 1;
        Fspmcfg->Ch1_DramDensity       = 2;

However, this error message is still presents 
CSE: MBP Data HOB not valid!

I've not managed to use any of the Ch2 and Ch3, enabling them make the boot hang on FspMemoryInit  
        Ch2_RankEnable
        Ch2_DramDensity 
        Ch3_RankEnable 
        Ch3_DramDensity

I've tried to boot Ubuntu from MMC or USB, they are valid, but nothing can be booted, looks the file system is not recognized

andreyv1978, 2020-04-01:
So adding:
        Fspmcfg->Ch2_RankEnable        = 1;
        Fspmcfg->Ch2_DramDensity       = 2;
        Fspmcfg->Ch3_RankEnable        = 1;
        Fspmcfg->Ch3_DramDensity       = 2;
        Fspmcfg->RmtCheckRun           = 3;
makes board to init FspMemoryInit, but still have error 'CSE: MBP Data HOB not valid!'
no graphics and fat system recognized :(

mauricema, 2020-04-01:
"CSE: MBP Data HOB not valid!" is a known issue with FSP MR6.   But it is not that critical for boot.   This can be addressed by using APL FSP MR5 release instead.
Also, memory parameter is very important.  If it is not correct, might result in system stability issue or other weird issue.
For GFX, not sure which port you used (HDMI or DP) ?  I tried HDMI.  Also sometime, the monitor connected also make a difference.

andreyv1978, 2020-04-01:
Bingo, thank you HDMI output is working, can see Slim Bootloader image :)
so, how can i boot from MMC or USB now , i have Ubuntu 18.04 or Yocto
Ubuntu already installed on MMC

andreyv1978, 2020-04-01:
bios output

[sku3-memory-sorted.txt](https://github.com/slimbootloader/slimbootloader/files/4415884/sku3-memory-sorted.txt)

mauricema, 2020-04-01:
that is good to know.
To boot Ubuntu, you just need to download the official Ubuntu 18.04 iso image and write it into a USB key.  Then you just plug in this key into USB port, reboot your system.   It will try to boot from the USB automatically.

mauricema, 2020-04-01:
Or you can follow instructions here to boot Yocto.
https://slimbootloader.github.io/supported-hardware/up2.html#booting-yocto-linux-from-usb

andreyv1978, 2020-04-01:
great news, it works !!!!!!! at least Ubuntu 18.04. I do not need to sort out Yocto, it was a learning exercise before making Apollo Lake Bios on the different board
Looks all peripheral works, video, USB, mouse, keyboard, ethernet 
Do you want my changes for the MemSKU=3 ?

mauricema, 2020-04-01:
Sure.  If you can contribute back, that will be perfect.

mauricema, 2020-04-01:
BTW, the UP2 HAT40 interfaces might not work properly because it needs some FPGA binary which we don't have yet.

andreyv1978, 2020-04-01:
do you want me to send file for review ? it is a small addons for the memsku=3

mauricema, 2020-04-01:
maybe just submit a code github review for it if you like.  
If you do not prefer that,  you can also post the changes here and we can submit a PR for it.

andreyv1978, 2020-04-01:
what I do not see is a Bios menu to edit on the fly, save changes and restart, it is not available ? or i need to build every time if  I want changes to make?

andreyv1978, 2020-04-01:
here my changes, not found an option to submit code review sorry

[Stage1BBoardInitLib.zip](https://github.com/slimbootloader/slimbootloader/files/4416204/Stage1BBoardInitLib.zip)

andreyv1978, 2020-04-01:
would be able to test with my board if you make more changes to the file

mauricema, 2020-04-01:
Thanks, we will take a look and upstream.
For BIOS menu, since it is a bootloader with fixed configuration,  we don't offer setup option.  but we do allow capsule update to do configuration update at run time.  We also have ConfigEdit.py and other scripts to modify an existing IFWI with CFGDATA change.  It does not need rebuild.

closed 2020-04-01 by andreyv1978

mauricema, 2020-04-01:
BTW, submit a PR with your code change.
https://github.com/slimbootloader/slimbootloader/pull/623
If you want to update your name mentioned in the commit message, please let me know.