slimbootloader/slimbootloader #661: Build failed for latest sbl on apl platform

shehamb, 2020-04-13:
Hi,
I just checked out the SBL and i am build getting this build error:

Fd File Name:STAGE1B (/home/mahesh/up2/SBL/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/STAGE1B.fd)

Generate Region at Offset 0x0
   Region Size = 0x14000
   Region Name = FV

Generating STAGE1B FV
###### ['GenFv', '-F', 'TRUE', '-a', '/home/mahesh/up2/SBL/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/Ffs/STAGE1B.inf', '-o', '/home/mahesh/up2/SBL/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/STAGE1B.Fv', '-i', '/home/mahesh/up2/SBL/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/FV/STAGE1B.inf']
Return Value = 2
GenFv: ERROR 3000: Invalid
  the required fv image size 0x145a8 exceeds the set fv image size 0x14000

build.py...
 : error 7000: Failed to generate FV

build.py...
 : error 7000: Failed to execute command

- Failed -
Build end time: 00:46:32, Apr.13 2020
Build total time: 00:00:24

mauricema, 2020-04-13:
I think it is related to your compiler.  
The travis build on github confirmed it can build successfully with GCC7.  In your case, it produces a little bit bigger image with GCC5 and cannot fit into the flash layout.   You can adjust the layout by changing Platform\ApollolakePlatformPkg\BoardConfig.py:
From:
  self.STAGE1B_FD_SIZE = 0x0006B000
To:
  self.STAGE1B_FD_SIZE = 0x0006C000

shehamb, 2020-04-14:
Thank you for your pointers. Build issue is resolved now. Closing this
ticket now.

closed 2020-04-14 by shehamb