slimbootloader/slimbootloader #673: FspMemoryInit  hang forever in APL platform

andreyv1978, 2020-04-14:
Hi

We are using Intel Atom E3950 based reference board, LeafHill CRB. We are trying to port Slimbootloader on our board following instructions from 
https://slimbootloader.github.io/supported-hardware/apollo-lake-crb.html
The FSP binaries We are using taken from https://software.intel.com/en-us/articles/uefi-firmware-project-for-intel-atom-processor-e3900-series-processor-platforms
We finalized memory settings as instructed according to our schematics. Unfortunately, the boot is hanging on  FspMemoryInit, see the attached output log.

We got FSP Debug build from Intel MR5, but there is still the same message on the screen '
Call FspMemoryInit ...'

[sqx2-fspmemoryinit-hang-mr5-debug.txt](https://github.com/slimbootloader/slimbootloader/files/4474439/sqx2-fspmemoryinit-hang-mr5-debug.txt)

How can we investigate the root of the problem using FSP debug binaries ? What else can we do to understand what is wrong with memory settings?

Andrey

mauricema, 2020-04-14:
Hi, Andrey,

Did you get the FSP debug binary ?   If so,  you need to do the following:
1. Copy FSP debug binary into Silicon\ApollolakePkg\FspBin\FspDbg.bin
2.  Build SBL using Debug FSP
     python BuildLoader.py build apl -fd
3.  Stitch and flash

You should see something like  "FDBG: BLD(D) FSP(D)" from the UART print.  If you see "FDBG: BLD(D) FSP(R)",  it indicates you are still using release FSP binary.  With DEBUG FSP, you should see much more debug message from MRC code.

andreyv1978, 2020-04-14:
updated, but still no messages from FspMemoryInit function

[sqx2-fspmemoryinit-hang-mr5-debug.txt](https://github.com/slimbootloader/slimbootloader/files/4475901/sqx2-fspmemoryinit-hang-mr5-debug.txt)

mauricema, 2020-04-14:
If so, the FSP you got might be an release FSP then.  Where did you get this debug fsp binary from ?   
Run following command , it will tell you it is a  real debug FSP.
python IntelFsp2Pkg\Tools\SplitFspBin.py info -f Silicon\ApollolakePkg\FspBin\FspDbg.bin
In the print out, you should see BIT0 to be 0 to indicate DEBUG FSP bianry.  EX:
ComponentAttribute         = 0x1000

andreyv1978, 2020-04-14:
Got it from Intel support:

Found the following 3 Firmware Volumes in FSP binary:
FV0:
  GUID   : 1B5C27FE-F01C-4FBC-AEAE-341B2E992A17
  Offset : 0x00000000
  Length : 0x0002B000
FV1:
  GUID   : F32DB7DF-C5BC-423C-9416-08F1FA26A282
  Offset : 0x0002B000
  Length : 0x00059000
FV2:
  GUID   : 52F1AFB6-78A6-448F-8274-F370549AC5D0
  Offset : 0x00084000
  Length : 0x00002000

FSP_S contains FV0
<FSP_INFORMATION_HEADER>:
  Signature                  = 0x48505346 ('FSPH')
  HeaderLength               = 0x48
  Reserved1                  = 0x0
  SpecVersion                = 0x20
  HeaderRevision             = 0x3
  ImageRevision              = 0x1040301
  ImageId                    = 0x245053464C504124 ('$APLFSP$')
  ImageSize                  = 0x2B000
  ImageBase                  = 0x200000
  ImageAttribute             = 0x1
  ComponentAttribute         = 0x3003
  CfgRegionOffset            = 0x124
  CfgRegionSize              = 0x3B0
  Reserved2                  = 0x0
  TempRamInitEntryOffset     = 0x0
  Reserved3                  = 0x0
  NotifyPhaseEntryOffset     = 0x588
  FspMemoryInitEntryOffset   = 0x0
  TempRamExitEntryOffset     = 0x0
  FspSiliconInitEntryOffset  = 0x592

FSP_M contains FV1
<FSP_INFORMATION_HEADER>:
  Signature                  = 0x48505346 ('FSPH')
  HeaderLength               = 0x48
  Reserved1                  = 0x0
  SpecVersion                = 0x20
  HeaderRevision             = 0x3
  ImageRevision              = 0x1040301
  ImageId                    = 0x245053464C504124 ('$APLFSP$')
  ImageSize                  = 0x59000
  ImageBase                  = 0xFEF71000
  ImageAttribute             = 0x1
  ComponentAttribute         = 0x2003
  CfgRegionOffset            = 0x124
  CfgRegionSize              = 0x200
  Reserved2                  = 0x0
  TempRamInitEntryOffset     = 0x0
  Reserved3                  = 0x0
  NotifyPhaseEntryOffset     = 0x0
  FspMemoryInitEntryOffset   = 0x3F0
  TempRamExitEntryOffset     = 0x3FA
  FspSiliconInitEntryOffset  = 0x0

FSP_T contains FV2
<FSP_INFORMATION_HEADER>:
  Signature                  = 0x48505346 ('FSPH')
  HeaderLength               = 0x48
  Reserved1                  = 0x0
  SpecVersion                = 0x20
  HeaderRevision             = 0x3
  ImageRevision              = 0x1040301
  ImageId                    = 0x245053464C504124 ('$APLFSP$')
  ImageSize                  = 0x2000
  ImageBase                  = 0xFFFFF000
  ImageAttribute             = 0x1
  ComponentAttribute         = 0x1003
  CfgRegionOffset            = 0x124
  CfgRegionSize              = 0x58
  Reserved2                  = 0x0
  TempRamInitEntryOffset     = 0x40D
  Reserved3                  = 0x0
  NotifyPhaseEntryOffset     = 0x0
  FspMemoryInitEntryOffset   = 0x0
  TempRamExitEntryOffset     = 0x0
  FspSiliconInitEntryOffset  = 0x0

Looks like not a Debug ?

mauricema, 2020-04-14:
Yes, it is still a release FSP.   A debug APL FSP dump is like this:
....
FSP_T contains FV2
<FSP_INFORMATION_HEADER>:
  Signature                  = 0x48505346 ('FSPH')
  HeaderLength               = 0x48
  Reserved1                  = 0x0
  SpecVersion                = 0x20
  HeaderRevision             = 0x3
  ImageRevision              = 0x1040301
  ImageId                    = 0x245053464C504124 ('$APLFSP$')
  ImageSize                  = 0x2000
  ImageBase                  = 0xFFFFF000
  ImageAttribute             = 0x1
  ComponentAttribute         = 0x1000
  CfgRegionOffset            = 0x124
  CfgRegionSize              = 0x58
  Reserved2                  = 0x0
  TempRamInitEntryOffset     = 0x42D
  Reserved3                  = 0x0
  NotifyPhaseEntryOffset     = 0x0
  FspMemoryInitEntryOffset   = 0x0
  TempRamExitEntryOffset     = 0x0
  FspSiliconInitEntryOffset  = 0x0

andreyv1978, 2020-04-14:
good would need to ask Debug version again, thank you

andreyv1978, 2020-04-15:
The Intel unfortunately can not provide a different Debug FSP from what I've already got
Can you help me to use what I have to investigate the problem of the FspMemoryInit failure ?

mauricema, 2020-04-15:
What you got is not a debug FSP image at all.  For release FSP, it is already on public github,  For debug FSP, it is not available publicly.  You will have to contact your Intel representative to request it.
For a customized board, without debug FSP and board info, it will be difficult to guess what are those memory parameters.

andreyv1978, 2020-04-15:
can you help me to understand what I should receive from Intel, I've got FSP MR5 Debug files: fsp.bsf and fsp.fd , not sure if this is enough without some bios binary file ?

mauricema, 2020-04-15:
Yes, you will get fsp.bsf and fsp.fd.   But the fsp.fd should be a debug version.  What you got is a regular release version FSP.  If you did request debug version FSP,  maybe they gave the wrong images to you ?

andreyv1978, 2020-04-16:
> * 
> * Build SBL using Debug FSP

not really followed your recommendations "Copy FSP debug binary into Silicon\ApollolakePkg\FspBin\FspDbg.bin", used the old bin instead of renaming Fsp.fd to FspDbg.bin
I am receiving now a full debug log from FSP, and managed to find the correct mem settings
Closing an issue because managed to get to the boot menu, thank you for your help

closed 2020-04-16 by andreyv1978

mauricema, 2020-04-16:
That is really great news!  :)
Thank you for updating!