slimbootloader/slimbootloader #680: Failed to set GPIO_200 GpioDerection to Native

andreyv1978, 2020-04-17:
I am trying to edit GPIO setting using ConfigEditor.py for ApolloLake platform using my already created .dlt file with memory settings
Unfortunately, a section of the GPIO GPIO_200 --> GpioDirection can not be set to Native, the log output from the python shows 'Update GPIO_200_Half1_Direction from 0x3 to 0xFF !', but when I am trying to save changes to the file I have an exception:
File "C:\Python37\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "w:/datapath/shared/project/slimbootloader/BootloaderCorePkg/Tools/ConfigEditor.py", line 786, in SaveFullToDelta
    self.SaveDeltaFile(True)
  File "w:/datapath/shared/project/slimbootloader/BootloaderCorePkg/Tools/ConfigEditor.py", line 778, in SaveDeltaFile
    self.UpdateConfigDataOnPage()
  File "w:/datapath/shared/project/slimbootloader/BootloaderCorePkg/Tools/ConfigEditor.py", line 1043, in UpdateConfigDataOnPage
    self.SyncConfigDataFromSubRegion()
  File "w:/datapath/shared/project/slimbootloader/BootloaderCorePkg/Tools/ConfigEditor.py", line 863, in SyncConfigDataFromSubRegion
    self.CfgDataObj.UpdateBsfBitFields(SubItem, Value, ValArray)
  File "w:\datapath\shared\project\slimbootloader\BootloaderCorePkg\Tools\GenCfgData.py", line 917, in UpdateBsfBitFields
    ValueArray[:]  = bytearray.fromhex('{0:0{1}x}'.format(int(BitsValue[::-1], 2), Blen * 2))[::-1]
ValueError: non-hexadecimal number found in fromhex() arg at position 9

I am able to save anything else for the Direction, 0x0, 0x1, 0x2, 0x3 but not 0xFF.  

Regards
Andrey

andreyv1978, 2020-04-17:
The build also fails if I am using the following settings:
GPIO_CFG_DATA.GPIO_200_Half1.Direction   | 0xF
 not really sure what setting should be there for the Direction = Native

mauricema, 2020-04-17:
What are you trying to set for GPIO 200?

If you want to change it to native functions,  you need to set GpioAltFunc instead.
0 is GPIO, 1 is function 1, 2 is function 2, …

Valid GpioDirection option is 0 to 3.    0:GPIO, 1:GPI, 2:GPO, 3:HI_Z

In general, to modify GPIO,  you can use BootloaderCorePkg\Tools\ConfigEditor.py to load the CfgDataDef.dsc file and your board DLT file located in your platform CFGDATA folder.   Then you can make changes in the GUI, and then save it back into the board DLT.  This does not need manually modification for DLT file.

Thanks
Maurice

andreyv1978, 2020-04-18:
I am trying to set following settings:
GPIO_CFG_DATA.GPIO_200_Half1.Direction   | 0xF             <<<<Don't know if this is correct for Native setting
GPIO_CFG_DATA.GPIO_200_Half1.AltFunc     | 0x1              < #1 Alternative function is HPD
GPIO_CFG_DATA.GPIO_200_Half1.PuPdEnable  | 0x3           < Enable Internal Pullup
GPIO_CFG_DATA.GPIO_200_Half1.PuPdValue   | 0xC            < 20k PU

Andrey

mauricema, 2020-04-19:
The Direction will only be used when AltFunc is 0 (GPIO).
Since you have AltFunc = 1,  the Direction value does not really matter.  I believe you can set it to 0.
GPIO_CFG_DATA.GPIO_200_Half1.Direction | 0

andreyv1978, 2020-04-20:
I am trying also set some of the PCIE RP with Config Editor, but I am missing some of the setting there or can not see how can I do that. For example
PCIE RP0-3 need to be set to x4 Gen2
PCIE RP4-5 need to be set to x1 Gen1
Also the LCTL2_LSTS2 bit 6 need to be 1 (which is 3.5db). I see LCTL2_LSTS2 setting in the code, but how can i do that in Config Editor ?

mauricema, 2020-04-20:
PCIe root port bifurcation is part of the silicon softstraps.  It needs to be configured using FITC tool.
PCIe speed Gen1/Gen2 and De-emphasis selections are available in FSP-S UPD.  But they are not exposed in CFGDATA to be configured by ConfigEditor.  So you might want to customize it into UpdateFspConfig() in Stage2BoardInitLib.c    PCieRpSpeed/PcieRpSelectableDeemphasis
EX:
if (PlatformId == YourBoardId) {
  FspsConfig->PcieRpSpeed[RpIndex] = ??;
  FspsConfig->PcieRpSelectableDeemphasis[RpIndex] = ??;
}
Please refer to Fsp.bsf for the detailed FSP options and values.

andreyv1978, 2020-04-21:
I've added a following code:
  if (GetPlatformId() == PLATFORM_ID_SQX2)
  {
    DEBUG ((DEBUG_INFO, "Makes an extra configurations of the PCIE for the SQX2\n"));

    PcieMaxRootPorts = ARRAY_SIZE (FspsConfig->PcieRpHotPlug);
    DEBUG ((DEBUG_INFO, "PCIE ports connected = 0x%02X\n",\
        PcieMaxRootPorts));

    // Configure PCIe Speed. 0:Auto(Default), 1:Gen1, 2:Gen2, 3:Gen3.
    // Gen1 and x1
    FspsConfig->PcieRpSpeed[0] = 1;
    FspsConfig->PcieRpSpeed[1] = 1;

    // Gen2  and x4
    FspsConfig->PcieRpSpeed[2] = 2;
    FspsConfig->PcieRpSpeed[3] = 2;
    FspsConfig->PcieRpSpeed[4] = 2;
    FspsConfig->PcieRpSpeed[5] = 2;

    /* When the Link is operating at 5.0 GT/s speed, this bit 
      selects the level of de-emphasis
      for an Upstream component.
      1b:-3.5 dB 0b:-6 dB. 0:Disable, 1:Enable(Default). */
    FspsConfig->PcieRpSelectableDeemphasis[0] = 0;
    FspsConfig->PcieRpSelectableDeemphasis[1] = 0;

    // Gen2 and -3.5db
    FspsConfig->PcieRpSelectableDeemphasis[2] = 0x2;
    FspsConfig->PcieRpSelectableDeemphasis[3] = 0x2;
    FspsConfig->PcieRpSelectableDeemphasis[4] = 0x2;
    FspsConfig->PcieRpSelectableDeemphasis[5] = 0x2;
  }

How can I set a following:
PcieRp [0] --> set as CLKREQ0 and PCIE_CLKOUT0 ?
PcieRp [1] --> set as CLKREQ1 and PCIE_CLKOUT1  ?
PcieRp [2-5] --> set as CLKREQ2 and PCIE_CLKOUT2 ?

and grouping:
PcieRP[2-5]  -->grouping of  x4 ?
PcieRP[0-1]  -->grouping of  x1 ?

Can you help ne with that ?

mauricema, 2020-04-21:
btw, since it is a different topic than the original submitted.  Can we close the original issue ?

andreyv1978, 2020-04-21:
Thank you for you help
Closing an issue

closed 2020-04-21 by andreyv1978

andreyv1978, 2020-04-23:
Hi mauricema 

I've received a FIT tool today, it looks like would generate the binary output for me
How I should integrate it with SBL build ? 

Andrey

mauricema, 2020-04-23:
Hi, Andrey

A simple way is to use FIT to modify your base IFWI image that you used for the stitching.   Once the base IFWI image is modified, everything else is exactly the same.

To modify your base IFWI image, you need to run the FIT,  then open the base IFWI image you have.   Then you can go to “Flex I/O” tab to find “Root Port Configration” and change to the desired mode 2x1 and 1x4.  Then you can build the image again.  It will generate a new IFWI with your modified settings.    Then you can use this new IFWI as your base image for SBL stitching.

Thanks
Maurice

andreyv1978, 2020-04-23:
It looks now much better with your recommendations than I've tried today
However i still have a number of errors:

Building SMIP data.
Executing pre-build actions
Building objects
Processing attribute: SmipData
Executing post-build actions

Full Flash Image written to W:\datapath\sqx2\FIT\SmipSubPartitionRaw.bin
Writing map file W:\datapath\sqx2\FIT\SmipSubPartitionRaw.map
Error 7: [SMIP Controller] Invalid Manifest Extension Utility path. Manifest Extension Utility path set to: ''
Error 6: [SMIP Controller] Failed to sign SMIP data. 
Error 37: [Ifwi Actions] Failed to build SMIP data. 
Error 19: [Ifwi Actions] Failed to prepare OEM SMIP data. 
Error 12: [Ifwi Actions] Failed to prepare one or more IFWI components. 
Error 15: Failed to build. 
Failed to build!

Is anything else I should set to make a build ?

andreyv1978, 2020-04-23:
the SMIP version config all zeroes, is it making a build fail ?

andreyv1978, 2020-04-23:
Looks like I know what is a problem, I am missing private.pem key
Would sort it tomorrow

andreyv1978, 2020-04-28:
Hi Maurice

I am using Intel Fit tool to rebuild LeafHill binary image with our settings
Unfortunately, It looks like some of the signing settings aren’t right to generate binaries
I've used a following tools to set up FIT:
1.	OpenSSL tool placed in the C:\ directory and PATH is set
     a.	RSA key is generate as following: openssl genrsa -des3 -out private.pem 2048
2.	Manifest Extension Utility downloaded and set to use OpenSSL as signing tool with key: private.pem 
3.	FIT tool setting set as in attached picture <fit-settings.PNG> 

Unfortunely. The build is failing with a following:
Executing pre-build actions
Executing pre-build actions
Building objects
Processing attribute: GpioProfile
Executing post-build actions

GPIO Profile 'Profile 0' saved to file:
W:\datapath\sqx2\FIT\Int\gpio_Profile_0.bin

Committing ME configuration.

Building SMIP data.
Executing pre-build actions
Building objects
Processing attribute: SmipData
Executing post-build actions

Full Flash Image written to W:\datapath\sqx2\FIT\SmipSubPartitionRaw.bin
Writing map file W:\datapath\sqx2\FIT\SmipSubPartitionRaw.map
Signing SMIP data with default Manifest Extension Utility key.
(Exit code: 0)

Command:
W:\datapath\sqx2\FIT\Manifest Extension Utility\WIN32\meu.exe  -resign all -f  "W:/datapath/sqx2/FIT/SmipSubPartitionRaw.bin" -cfg  "W:\datapath\sqx2\FIT\Manifest Extension Utility\WIN32\meu_config.xml" -o  "W:/datapath/sqx2/FIT/SmipSubPartitionSigned.bin" -st  "OpenSSL" -stp  "C:\OpenSSL\bin\openssl.exe"

Error 11: [SMIP Controller] An error ocurred while running the Manifest Extension Utility to sign the SMIP data. 
Error 6: [SMIP Controller] Failed to sign SMIP data. 
Error 37: [Ifwi Actions] Failed to build SMIP data. 
Error 19: [Ifwi Actions] Failed to prepare OEM SMIP data. 
Error 12: [Ifwi Actions] Failed to prepare one or more IFWI components. 
Error 15: Failed to build. 
Failed to build!

Can you instruct me what else should I set to generate IFWI build ?
Should I have a special HASH key to sign IFWI image in FIT tool? I've seen in the document statement which is saying that I should have an original private key which was used to sign an image, is it right ?

Andrey

thau-my, 2020-04-28:
hi, Andrey,

Did you specify a) SMIP Signing Key b) OEM Key Manifest Binary ,  under FIT Windows -> Platform Protection -> Platfrom Integrity ?

If you are not familiar with the FIT configuration, I would suggest you use the StitchIfwi.py script available in Platform/ApollolakeBoardPkg/Script to build the IFWI image, 
It is more straighforward, you just need to refer to the help note in the script to setup your Stitching folder.

Regards,
Hau, Tze-ming

andreyv1978, 2020-04-28:
It looks like SMIP Signing Key is null, where can I get SMIP key ? or how can I generate it ?

[image]

andreyv1978, 2020-04-28:
Looks like the key from BootloaderCorePkg/Tools/Keys makes a build, thank you

thau-my, 2020-04-28:
Yes, It contains TestSigningPrivateKey.pem which is Intel debug private key for testing purpose.

andreyv1978, 2020-04-28:
It is too early for me,  but for the release build, i would require different key ?

thau-my, 2020-04-28:
Not necessary for release build. 
If your system is designed with Verified boot enabled in BootGuard policy, then for security reason, the SMIP key should be a highly secure key, you also need to have a OEM key manifest with all other key hashes within it and signature attached.

andreyv1978, 2020-04-28:
Thank you for your update