slimbootloader/slimbootloader #689: GraphicsPpiInit failed on the APL platform

andreyv1978, 2020-04-21:
Hi

I am trying to configure DisplayPort in MR5 debug FSP, but the following message indicates the video is not working:
'GraphicsPpiInit failed. 
Failed to Call Ppi and fill frame buffer '

or after
'Failed to get Graphics Info HOB from FSP'

Attaching the output log, maybe you can help me to sort it out
[graphics-failed.txt](https://github.com/slimbootloader/slimbootloader/files/4509667/graphics-failed.txt)

Andrey

mauricema, 2020-04-21:
This is part of the FSP behavior and SBL might not be able to help resolve it.
FSP depends on a VBT binary to initialize the GFX ports.  The VBT binary needs to be configured properly to match your board design.  SBL used the same VBT bianry downloaded from FSP package. That is for CRB boards.  You might need to contact your Intel representative to get a Intel Binary Modification Program (BMP) utility to config it if your GFX design is different from CRB.

andreyv1978, 2020-04-21:
Thank you for update, I will contact our board designer

andreyv1978, 2020-04-27:
Hi

I've received Intel Display Configuration Tool (DisCon), because BMP is deprecated. The new tool works only with .bin files and .json files instead of .bsf. Also I've received VBT binaries ( for different fabs, ports and types ) for Apollo Lake Platform (Broxton).  I may not need to reconfigure anything just replace the current Vbt.bsf and .Vbt.dat from Platform\ApollolakeBoardPkg\VbtBin with a new one ? But how can I change the .bin from DisCon to .dat in SBL ?
Can you help me with that ?

I've seen you have a fix for multiple Vbt tables that I can use for my platform
https://github.com/slimbootloader/slimbootloader/commit/94a62590368e2cb5c73d52b9195f6fee7c894c5f

Andrey

jgutbub, 2020-04-27:
Hi Andrey,

The '.dat' extension is not different from the '.bin' extension when it comes to VBT binaries, you should not have any issue to change '.dat' to '.bin' in this instance. Thank you for mentioning the DisCon tool, it seems to be similar to BMP but uses JSON instead of BSF as you mentioned.

Thanks,
James

andreyv1978, 2020-04-29:
Looks like with a VBT table where Display port is enabled is working. I am not a specialist in all graphic settings, there are too many there.
Closing the ticket

closed 2020-04-29 by andreyv1978