slimbootloader/slimbootloader #696: Spread-Spectrum Clockings (SSC) settings missing in APL

andreyv1978, 2020-04-29:
Hi

I am not be able to find settings for the Spread-Spectrum Clocking (SSC) on its PCIE_CLKOUT outputs
I should set not to use it in the Apollo Lake settings 

Can you help me to find it out how to configure in SBL ?

Andrey

mauricema, 2020-04-29:
Did you check FIT soft strap in Flex I/O ?
There are "Pcie Port Non-Common Clock with SSC mode enable" soft strap under Flex I/O.  Is this what you want ?

andreyv1978, 2020-04-29:
Yes looks like all defaults disabled and LCPLL_CR_RW_CONTROL_1 disabled as well, closing issue
thank you for your help

andreyv1978, 2020-05-20:
Hi

We used a default configuration of  LCPLL_RW_CONTROL_1_DEFAULT.SSC_EN as Disable in FIT tool. However, our suspicions that https://github.com/slimbootloader/slimbootloader/issues/715 caused becuaes SSC is applied or not disabled at the right time. 
Is any option available to check it from SBL where it is set ?

Andrey

thau-my, 2020-05-20:
Hi Andrey,

Programming of both LJ1PLL and LCPCC SSC in bootloader requires IPC CMD interface for PMC. The IPC CMD library is not available in SBL.
As for the FIT configuration, the SSC configuration will stay in default (disable) 
If you want to enable SSC via FIT, you also need to set SSC_EN_OVR.
But that will cause a 1~2 sec delay in boot time due to the PMC FW reset after overriding the SSC configuration.

andreyv1978, 2020-05-21:
Had to reopen an issue because we've found that the setting from the FIT tool have no any effect. The SSC is enabled from the boot start

andreyv1978, 2020-05-21:
SSC is enabled in SlimBoot:
[image]

this is a picture of different BIOS where SSC is disabled:
[image]

rprangar, 2020-05-22:
Hi

As thau-my mentioned earlier, SSC can be controlled via FIT settings. In addition to that, there are some IPC/SideBand Interfaces available to control it during boot. For reference please look at [EDK2 support for APL](https://github.com/tianocore/edk2-platforms/blob/devel-IntelAtomProcessorE3900/Platform/BroxtonPlatformPkg/Common/PlatformSettings/PlatformPreMemPei/PlatformInitPreMem.c)

I believe Intel reference BIOS enables SSC and SBL does not. One suggestion I would have is for you to flip the current FIT setting and see if it makes a difference.

andreyv1978, 2020-05-22:
Hi

Regarding to the changes that I can make in edk2-platform, see the answer on my question from https://github.com/slimbootloader/slimbootloader/issues/704.
mauricema wrote that all my changes in edk2-platform would be replaced with SBL, is that correct ?

Can you please instruct how to bring 'IPC CMD interface for PMC' into SBL and I will do it? I am relatively familiar with your code already

Andrey

andreyv1978, 2020-05-22:
Hi 

I've made a different flip flops to change settings like (made a lot of different permutations):
Platform Configuration:
LJ1PLL_RW_CONTROL_1_DEFAULT.SSC_EN --> Disable
LJ1PLL_RW_CONTROL_1_DEFAULT.SSC_EN_OVR --> Disable_Override
LCPLL_RW_CONTROL_1_DEFAULT.SSC_EN --> Disable
LCPLL_RW_CONTROL_1_DEFAULT.SSC_EN_OVR --> Enable_Override

So, SSC actually never disabled !!!!!!
FIT generates IFWI binaries, all are different in the regions 0x21e0 - 0x22d0. Also the SBL have no effect on those places, confirmed comparing SBL IFWI image with FIT generated. I am not sure, but maybe we need to configure SSC disable somewhere else ? 

Andrey

andreyv1978, 2020-05-22:
see for example the settings:
[image]
Image showing Downspreading still happening even when Upspreading is set
[image]

Can you explain it?
All the permutations in FIT tool for LJ1PLL and LCPLL settings makse the same output after SBL stitching

andreyv1978, 2020-05-26:
Hi

Can you update if the SBL has a bug with a FIT tool for the specific settings?

Andrey

rprangar, 2020-05-26:
Hi

FIT manipulates the soft strap settings in the descriptor region and SBL has no direct interaction with these settings. So I don't think this is an SBL bug.

Regarding your other question, I provided the link to a reference on how to control SSC. Pasting it here again. Please check PlPeiHighSpeedSerialInterfaceSSCInit and PeiDDRSSCInit  functions in this link [EDK2 Reference] (https://github.com/tianocore/edk2-platforms/blob/devel-IntelAtomProcessorE3900/Platform/BroxtonPlatformPkg/Common/PlatformSettings/PlatformPreMemPei/PlatformInitPreMem.c)

Thanks.

rprangar, 2020-05-26:
Hi

Would it be possible for you to file an IPS ticket with Intel to get some hardware inputs regarding your requirements/issue. 

Meanwhile, if you would like to disable SSC through SBL, you have to port the code I referred above to SBL. BoardInit (Phase - PostConfigInit) in https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ApollolakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c may be a good place to implement this.

andreyv1978, 2020-05-27:
Hi rprangar

Thank you for an update, it looks like I got how to port sideband and IPC, I will update a status after porting and testing

Andrey

andreyv1978, 2020-05-28:
Hi rprangar

I've completed sideband interface porting to SBL can send you code to review and checkout 
Unfortunately, it looks something wrong
1. Reading from LCPLL_CTRL_1 always returns 0, even if I am changing in FIT tool
2. Writing to LCPLL_CTRL_1 looks working, but have no effect on SSC
Do you know what could be an issue ? the place (0x99, 0x9910) is a correct one ?
I've not finished yet IPC interface

Andrey

andreyv1978, 2020-05-28:
I've completed IPC interface porting also, and only now reading from LCPLL_CTRL_1  receives the correct FIT settings. Have to test it a bit more tomorrow 

So, I would like to submit 2 changesets for review 
1. IPC interface
2. SideBand Interface
How can I do that ? Would you like to have it in your code ?

rprangar, 2020-05-28:
Hi

Yes, your contributions are very welcome. Please refer https://slimbootloader.github.io/developer-guides/contributions.html for the github pull request flow.

Thanks.

andreyv1978, 2020-05-29:
bit lost in creation of pull request (I am a new bee in using git). It returns an error : Creating pull requests for 'slimbootloader:SideBand' failed: Validation Failed
Can you help out what can be wrong ?

I've created a patch with a SideBand interface only 
[SideBandInterface.zip](https://github.com/slimbootloader/slimbootloader/files/4701381/SideBandInterface.zip)

andreyv1978, 2020-05-29:
and the second patch for the IPC interface
[IPCinterface.zip](https://github.com/slimbootloader/slimbootloader/files/4701500/IPCinterface.zip)

andreyv1978, 2020-05-29:
Unfortunately, I can read from FIT setting and see the changes I am making. But setting everything disable has no effect to SSC (it is always enabled)
The setting of SetSSCDisable() you can see in patch added to the issue, it applied in PostConfigInit at the end (tried different places in PostConfigInit)

What else can invalidate SSC settings to make it always enabled ?

rprangar, 2020-05-29:
Any reset has to be conditional - in this case, it would translate to boot-check setting -apply setting-reset-boot-check setting-continue boot. Otherwise the boot will get stuck in a reset cycle.  

At this point, I would recommend to file an IPS ticket with Intel to get inputs from the HW PAE. There are certain other implications of disabling SSC on which the HW PAE team would be able to provide inputs.

andreyv1978, 2020-06-03:
can't really understand why SSC is enabled, reading from FIT setting using IPC interface can verify LJ1PLL  and LCPLL SSC is set disabled. However, the board is starting up and continue booting with SSC enabled

my patch to the IPC interface is here:
https://github.com/andreyv1978/slimbootloader/pull/1

thau-my, 2020-06-03:
Did you verify that **both** DDR SSC (LJ1PLL) and HSIO SSC (LCPLL) enable even though set to disable?
At which point during boot did you measure it? SBL shell or OS?

andreyv1978, 2020-06-03:
I am verifying it before DDR training (FSP init) https://github.com/andreyv1978/slimbootloader/pull/1/commits/eca86812ad2fd4ca9992d3ad3b84f7a9c31d7527#diff-991058d987bad5212a26703fab2aa2bd

After it finishes we are measuring it in SBL shell, can see the attached images previously
Intel response:
 "The information that may answer your clarification request and give more details, which should be reviewed with the assistance of your BIOS vendor, can be found in sections 23.1 and 23, on page 213 of the Intel Pentium and Celeron Processor N- and J- Series Formerly Apollo Lake Intel Architecture Firmware Specification Volume 2 of 2 BIOS Specification document # 559811."

rprangar, 2020-06-03:
I think there may be some confusion. I believe the UEFI reference code that I pointed out earlier, changes the SSC setting through a register. It does not change the soft strap setting in the descriptor set by FIT. 

I would recommend to continue the follow up through IPS.

andreyv1978, 2020-06-08:
I've got an answer from Intel

"Disabling SSC in Apollo Lake is not supported / not POR (Plan of Record), not validated and not recommended. It is not possible to disable SSC for PCIe only. There are other IPs that will be impacted if SSC is disabled.

If you still want to use this, for own testing, the process for CRB is to leave SSC enabled in FIT and set the override enabled strap, then make according changes in BIOS menu.
FIT example:

For own BIOS/Bootloader, the implementation needs to follow the example provided in the comment on Github.

Example for UEFI BIOS can be found in the Open Source version of Apollo Lake reference code:
https://github.com/tianocore/edk2-platforms/blob/devel-IntelAtomProcessorE3900/Platform/BroxtonPlatformPkg/Common/PlatformSettings/PlatformPreMemPei/PlatformInitPreMem.c

The fragment that controls whether SSC is enabled or disabled is based on this parameter: SystemConfiguration.HSSIOSSCEnable and the method called is PeiHighSpeedSerialInterfaceSSCInit.

As the file says the SSC enable/disable will affect the following IPs: USB3, PCie, SATA, eDP, DP, eMMC, SD and SDIO SSC

Our hardware team is checking if we can add some more details/clarification in our collateral around this topic.
"
I would add the SSC enabling code to what I've already completed, The issue might be, to set SSC disable after enabling it  and trigger restart. The SBL does not have BIOS menu, all would have to work in a single flow with  a multiple restarts

Andey

andreyv1978, 2020-06-11:
We have an old BIOS partially made for use by someone else, it looks like uses a full edk2 source code with  UEFI BIOS features (i can see all of that when reading debug logs). The first step they are doing is enabling SSC, which are clearly indicated by printing messages after checking   HSSIOSSCEnable. The next step is a manual selection of SSC disable and the restart is triggered. At the end the SSC is disabled and the board always boot with SSC disable.

I am trying to do the same, but without UEFI BIOS functioanlities. I can trigger SSC enable and wait to the paylod stage (but without having HSSIOSSCEnable, I wouldn't know it applied or not ). The next I can trigger SSC disable and make the cold restart, but I need HSSIOSSCEnable to know that now I am on stage of SSC disable and should not enable it again.

Maybe the different approach:
1. First boot: DDR training I am setting SSC enable, waiting to payload and triggers SSD disable and after a cold restart 
2. How can I let to the BIOS to know that this is a second boot and it should not trigger SSC enable any more ?

mauricema, 2020-06-11:
HSSIOSSCEnable is a setup option.  So once it is set in UEFI BIOS, it is static and will not change unless you go to the setup screen to change it again.  EX: if you have it enabled in setup, it will always have a value 1.    
Do don't  have to wait till UEFI payload in order to do these.   All can be done in SBL instead.  Why do you need the input from HSSIOSSCEnable ?  It will not tell you if it is the 1st or 2nd boot. 
In your case, I think you will need to have HSSIOSSCEnable=0 to disable SSC.   To sort this out, you need to know what is the working BIOS flow, and then we can align with that flow for your case.  Also maybe you should be able to read current SSC status  to know if it is always disabled ?

andreyv1978, 2020-06-11:
The following flow is working and recommended by Intel (as a pure test :) ):
1. enable SSC as described in https://github.com/tianocore/edk2-platforms/blob/devel-IntelAtomProcessorE3900/Platform/BroxtonPlatformPkg/Common/PlatformSettings/PlatformPreMemPei/PlatformInitPreMem.c
2. first Init, and after enable HSSIO only with LCPLL_CTRL_1 writing in 0x3
3. follow a full BIOS boot to the payload
4. disable SSC writing into LCPLL_CTRL_1 0x1
5. a full FSP restart
6. it should boot with SSC disable

This flow Intel recommends, and it's what our previous BIOS is doing. The FIT setting are not valid because Intel is always have SSC enable by default, but for some reason it should be set enabled as well. I've tried only booting with SSC disable option set in multiple stages Stage 1, or 1B or Stage 2, but it is never set at the end.

andreyv1978, 2020-06-11:
Adding the Intel response:
' .... I see 2 options.
One if to make it fixed in code if you don't want to have options to test either way or use CfgData.
https://github.com/slimbootloader/slimbootloader/tree/master/Platform/ApollolakeBoardPkg/CfgData

CfgData would be alternative to UEFI variables.
Your firmware engineering team should be able to follow up through the thread you have in the community.

With CfgData you could have 2 images, with and without SSC configuration for your testing.

So, the FIT would not change the behavior of SSC, keeping it enabled at first during boot, then during pre-memory initialization SBL could turn it off.
Just a reminder as I've mentioned, this (disabling SSC for Apollo Lake) is not supported by Intel.  ...'

mauricema, 2020-06-11:
For the recent code you added into SBL, shouldn't we set ssc_en_over to SSC_ENABLE for both DDR and HSIO SSC ?  Without overriding,  I guess the actual ssc_en setting will not take effect.

WBuf.LJ1PLL_CTRL_1.Fields.ssc_en_ovr = SSC_DISABLE;  => SSC_ENABLE ?
LCPLL_CTRL_1.Fields.ssc_en_ovr = SSC_DISABLE;  => SSC_ENABLE ?

Your original code:
SetSscDisable (
  VOID
  )
{

  ...
  
  //
  // set the ssc_en to Disable!
  //
  WBuf.LJ1PLL_CTRL_1.Fields.ssc_en = SSC_DISABLE;
  WBuf.LJ1PLL_CTRL_1.Fields.ssc_en_ovr = SSC_DISABLE;
  Status = IpcSendCommandEx (IPC_CMD_ID_EMI_RFI_SUPPORT, IPC_SUBCMD_ID_SSC_APPLY_NOW, &WBuf, BufferSize);
  if (EFI_ERROR (Status)) {
      DEBUG ((DEBUG_ERROR, "\nFailed to disable LJ1PLL_CTRL SSC \n\r"));
  }

  ...
  
  // disable SSC
  LCPLL_CTRL_1.Fields.ssc_en = SSC_DISABLE;
  LCPLL_CTRL_1.Fields.ssc_en_ovr = SSC_DISABLE;
  DEBUG ((DEBUG_INFO, "LCPLL_CTRL_1 write data: 0x%02X\n", LCPLL_CTRL_1.Data));
  SideBandWrite32 (0x99, 0x9910, LCPLL_CTRL_1.Data);
}

andreyv1978, 2020-06-12:
Hi mauricema 

I've made a few additons to my SBL bios to test a multiple options
1. library SscLib with a following functionalities
1.1 SetSscInit
-> initiate SSC LJ1PLL_CTRL_1 to enable ssc_en + ssc_en_ovr
-> nothing to be changes in LCPLL_CTRL_1 and LCPLL_CTRL_2
1.2 SetSscSetDisable
-> LCPLL_CTRL_1 set with 0x1
1.3 SetSscSetEnable
-> LCPLL_CTRL_1 set with 0x3
-> LCPLL_CTRL_2 set with 0x32b (UP_SPREAD_ONLY)
-> LJ1PLL set to ssc enable

So, i've made experiment and set SSC disable according to the flow:
1.	Stage 1B before FSP Memory Init: 
    a.	Set SSC Init with following
       i.	 LJ1PLL_CTRL_1.ssc_en  = SSC_ENABLE
     ii.	LJ1PLL_CTRL_1.ssc_ovr  = SSC_ENABLE
  b.	Set SSC HSSIO disable
      i.	LCPLL_CTRL_1 = 0x1
2.	Stage 2 Post Silicon Init: 
  a.	Set SSC HSSIO disable
    i.	LCPLL_CTRL_1 = 0x1
3.	Stage 2 PrePayloading: 
  a.	Set SSC HSSIO disable
    i.	LCPLL_CTRL_1 = 0x1

[image]

**The result even after multiple board resets the SSC still enabled** 

Experiment 2:
the flow is the same, but instead SSC disable initiated SSC enable with the setting of ssc_mode (UP_SPREAD_ONLY)

**The result still the same SSC enabled and DOWN_SPREAD_ONLY**

What am I doing wrong ? Where should I initiate SSC settings ? It looks like nothing applied at all
If you have an idea how to experiment more would welcome ? The code is trivially simple, but have no any effect to the FSP settings

Andrey

mauricema, 2020-06-12:
Andrey, 
I did not see any issue with your steps.   
When you did multiple board seset,  what kind of reset did you use ?   How did you trigger the reset ?

- Maurice

mauricema, 2020-06-12:
BTW, did you try to use your working BIOS as a base image to stitch SBL IFWI using StitchLoader.py ?

andreyv1978, 2020-06-12:
the working scenario is our old BIOS from the external company, I do not have source code, only can use DEBUG BIOS to get a full  log 
This is a log from Phoenix:
[phoenixputty.log](https://github.com/slimbootloader/slimbootloader/files/4771722/phoenixputty.log)

Line 81:
Setup Variable is not ready for SSC setting! Used default value!!
HSSIO : Default value of LCPLL_CTRL_1 register: 0x0
 LCPLL_CTRL_2 register: 0x7D9C12B

HSSIO : Setup Variable is not ready for SSC setting! Leave the default system HSSIO SSC settings!!

Line 31340 : I've added a marker at the point that the BIOS menu is accessed and F10 is pressed to save settings

Line 31372: After a restart the correct value is written to disable SSC
HSSIO : Default value of LCPLL_CTRL_1 register: 0x0
 LCPLL_CTRL_2 register: 0x7D9C12B
HSSIO disable : write LCPLL_CTRL_1 register: 0x1

After a power cycle the log phoenixputtynewboot is captured:
Line 222 : Correct value for SSC disable is set
HSSIO : Default value of LCPLL_CTRL_1 register: 0x0
 LCPLL_CTRL_2 register: 0x7D9C12B
HSSIO disable : write LCPLL_CTRL_1 register: 0x1

[phoenixputtynewboot.log](https://github.com/slimbootloader/slimbootloader/files/4771731/phoenixputtynewboot.log)

andreyv1978, 2020-06-12:
> BTW, did you try to use your working BIOS as a base image to stitch SBL IFWI using StitchLoader.py ?

hm, not tried to be honest
What can it give me ?

mauricema, 2020-06-12:
We want to know if some of the FIT settings have impact on the register settings.

Aslo in your working BIOS, what is the setup variable value for DDR and HSIO SSC when you enter the setup screen ?

andreyv1978, 2020-06-12:
good question, let me check everything on Monday

mauricema, 2020-06-12:
Also please try the following instead:
The code you added was to use SideBandWrite32 to set LCPLL_CTRL.  I am not sure if it is correct.
I believe it should use IpcSendCommandEx()

Could you try following ?

SSC_LCPLL_IPC_BUFFER   WBuf;
UINT8                  BufferSize;

BufferSize = sizeof (UINT32) * 2;
WBuf.LCPLL_CTRL_1.Data = 0;
WBuf.LCPLL_CTRL_2.Data = 0x7D9C12B;

//High Speed Serial IO SSC disable
WBuf.LCPLL_CTRL_1.Fields.ssc_en = SSC_DISABLE;
WBuf.LCPLL_CTRL_1.Fields.ssc_en_ovr = SSC_ENABLE;
IpcSendCommandEx(IPC_CMD_ID_EMI_RFI_SUPPORT, IPC_SUBCMD_ID_LCPLL_APPLY_NOW, &WBuf, BufferSize);

andreyv1978, 2020-06-15:
> IPC_SUBCMD_ID_LCPLL_APPLY_NOW

Looks like my source code is missing a following:
SSC_LCPLL_IPC_BUFFER  --> i can create it myself is not a problem
IPC_SUBCMD_ID_LCPLL_APPLY_NOW --> do not have this at all 
will try to use instead IPC_SUBCMD_ID_SSC_APPLY_NOW, but if you have this code somewhere can you let me know where i can get it

andreyv1978, 2020-06-15:
just making a new experiment from the previously uploaded log: phoenixputty.log line 12657, they set SSC enable first before SiInitPrePolicy()

I hope, i am doing similar experiment but setting SSC disable: Stage2BoardInitLib.c line 1046 PreSiliconInit: adding:

PeiDDRSsc:
    WBuf.LJ1PLL_CTRL_2.Fields.ssc_cyc_to_peak_m1 = SSC_Select_Table[SSC_DEFAULT_SETTING].Ssc_Cyc_To_Peak;
    WBuf.LJ1PLL_CTRL_2.Fields.ssc_frac_step = SSC_Select_Table[SSC_DEFAULT_SETTING].Ffs_Frac_Step;
    //
    // Set default value of Clock bending
    //
    WBuf.LJ1PLL_CTRL_5.Fields.pll_ratio_frac = CLK_Bending_Table[CLK_BENDING_DEFAULT_SETTING].Pll_Ratio_Frac;
    WBuf.LJ1PLL_CTRL_5.Fields.pll_ratio_int = CLK_Bending_Table[CLK_BENDING_DEFAULT_SETTING].Pll_Ratio_Int;

    WBuf.LJ1PLL_CTRL_2.Fields.ssc_mode = 0x1;

    //
    // send the IPC command for SSC
    //
    Status = IpcSendCommandEx (IPC_CMD_ID_EMI_RFI_SUPPORT, IPC_SUBCMD_ID_SSC_APPLY_NOW, &WBuf, BufferSize);

    //
    // Delay for 1ms to avoid the SSC doesn't set correctly sometimes
    //
    MicroSecondDelay (1000);

    //
    // set the ssc_en to Disable!
    //
    WBuf.LJ1PLL_CTRL_1.Fields.ssc_en = SSC_DISABLE;
    WBuf.LJ1PLL_CTRL_1.Fields.ssc_en_ovr = SSC_DISABLE;
    Status = IpcSendCommandEx (IPC_CMD_ID_EMI_RFI_SUPPORT, IPC_SUBCMD_ID_SSC_APPLY_NOW, &WBuf, BufferSize);
    return Status;

And immediately:
    WBuf.LCPLL_CTRL_1.Fields.ssc_en = SSC_DISABLE;
    WBuf.LCPLL_CTRL_1.Fields.ssc_en_ovr = SSC_ENABLE;
    IpcSendCommandEx(IPC_CMD_ID_EMI_RFI_SUPPORT, IPC_SUBCMD_ID_SSC_APPLY_NOW, &WBuf, BufferSize);

The boot is stuck on IpcSendCommandEx(IPC_CMD_ID_EMI_RFI_SUPPORT, IPC_SUBCMD_ID_SSC_APPLY_NOW, &WBuf, BufferSize); --> what is wrong ?

mauricema, 2020-06-15:
SSC_LCPLL_IPC_BUFFER is a structure containing just two DWORD.    IPC_SUBCMD_ID_LCPLL_APPLY_NOW is 0x03.   IPC_SUBCMD_ID_SSC_APPLY_NOW should not be used for SSC_LCPLL_IPC_BUFFER since the buffer size does not match.
If it still does not work for you, you will need to consult your Intel support to find more information on the proper flow since disabling SSC is not officially supported by APL POR.  Or if you have your working BIOS source code, you can try to see how they did it.

andreyv1978, 2020-06-15:
I only have this with 0x03
#define IPC_SUBID_TELEMETRY_TRACE_READ              0x03  ///<  Read MBB_TELEM_TRACE_MASK register
How to you get IPC_SUBCMD_ID_LCPLL_APPLY_NOW  ?

mauricema, 2020-06-15:
Both are correct.  These are sub commands.   They have different primary command.
Here is the simplified code for LCPLL change:
{
  SSC_LCPLL_IPC_BUFFER WBuf;
  UINT32  WBuf[2];
  UINT8    BufferSize;
  EFI_STATUS  Status;
 
  BufferSize = 8;
  WBuf[0] = 0;
  WBuf[1] = 0x7D9C12B;
  Status = IpcSendCommandEx (0xE8, 0x03, WBuf, BufferSize);
}

mauricema, 2020-06-15:
WBuf[0] BIT0 is ssc_en_ovr, and BIT1 is  ssc_en.  You might want to set
ssc_en_ovr = 1,  ssc_en = 0 to disable HSIO SCC.   So  you can try  WBuf[0] = 1 instead;

andreyv1978, 2020-06-16:
Hi mauricema

Made a variety of tests today using your recommendation as:
Status = IpcSendCommandEx (0xE8, 0x03, WBuf, BufferSize);

also tried to use 0x4 instead of 0x3 
also tried to pass WBuf as &WBuf
also used it in PreSiliconInit and in 'Stage 1B before FSP Memory Init'

all tests are passed to the shell and SSC was enabled 

Andrey

Andrey

mauricema, 2020-06-16:
If that is the case,  I have no further idea on what to try.   Since disabling SSC is not POR,  not sure if it has been fully validated.  Maybe you can follow up through the IPS ticket.

andreyv1978, 2020-06-16:
thank you for your help

rprangar, 2020-06-19:
I am closing this issue. If there is any new information regarding this issue, please feel free to reopen it.

Thanks.

closed 2020-06-19 by rprangar

andreyv1978, 2020-07-09:
Hi rprangar 

I've been away for some time, but now back online and need to sort out SSC.
I've started to look into coreboot usage and found that it might be the same issue as with SBL on our board
So, looking again into both coreboot and SBL :) 
Can you please advise me about the method of the Stitchifwi.py script, I've not used previously and it requires extra settings and packages from Intel. Would it be any different if I would use Stitchifwi.py instead of FIT tool and generate image directly ?
I am not sure what Phoenix made to make SSC disable, but they have a full UEFI BIOS menu to play around SSC. I would like to make the same magic still. The board redesign might be an option but not a fast one 

Andrey

andreyv1978, 2020-07-10:
Hi rprangar

coreboot have nothing to configure for SSC, therefore parked it
I've started to build a full edk2 image as explained : https://software.intel.com/content/dam/develop/external/us/en/documents/uefi-firmware-porting-guide-for-the-intel-atom-processor-e3900-series-820187.pdf
Are you familiar with these binaries? How can I configure UART2 port to see what is going on (failed to see anything on UART2)?
I will appreciate if you can help me
Andrey

andreyv1978, 2020-07-22:
Just a quick update from Intel

"....I did some testing Friday on APL CRB with F1 stepping and latest BIOS and couldn't see changes in the behavior using the knobs on my side.
I'm getting together with my FW engineering (development) counterparts to reassess. I will also be performing some joint testing with our HW PAE team tomorrow.... "