slimbootloader/slimbootloader #704: [APL] PLX chip goes OFF because of FSP reset while trained

andreyv1978, 2020-05-07:
Hi

We've developed the first BIOS on our board long time ago with external supplier, and during the process asked them to disable FSP reset. Because we had a problem with PLX chip.
The PLX chip is relying on the Apollo Lake to provide the PCIE_CLKOUT and is using this clock to enable it to train up to the host PC.
They made it possible for us by disabling it from Silicon\BroxtonSiPkg\SouthCluster\Library\Private\PeiScInitLib\ScInit.c (maybe, cannot be sure for certain)
Unfortunately, I do not have this code in the SBL to play around or see where the reset is happening  while FSP is trained.
Can you help me to identify resetting during a stage of training? Can the training process be controlled from SBL ?

Andrey

thau-my, 2020-05-08:
Are you referring the the global reset vs cold reset when the function change from disable to enable vs enable to disable?

thau-my, 2020-05-08:
The global reset/cold reset will only be performed after as a call-back from PCI enumeration complete. If you can remove the call-back registration, then there would be the calll-back event causing the reset.

andreyv1978, 2020-05-08:
yes, I am referring to the cold reset during FSP training

mauricema, 2020-05-11:
ScInit.c you mentioned is part of the FSP code.  SBL will not have any control over that.  It seems your vendor modified the ScInit.c to make it work for you.  I am not sure what has been modified by them, but if that is the case, then mostly the similar change will be required inside FSP in order to address your issue.

thau-my, 2020-05-13:
About the reset happen during FSP training, if it is caused by change of FUNCTION DISABLE of internal SoC devices, you maybe try to comment out the below function call in BootloaderCorePkg/Stage2/Stage2.c to experiment to see if the code reset still happen.

//       BoardNotifyPhase (PostPciEnumeration);

andreyv1978, 2020-05-13:
> BoardNotifyPhase

Thank you for update, will get back to this after resolving PMIC instability: https://github.com/slimbootloader/slimbootloader/issues/715

andreyv1978, 2020-05-14:
Hi

Can you help me with a flow I am doing:
1. Generate IFWI binary from edk2-platform, adding changes to the ScInit.c
2. using FIT tool load IFWI from the previous stage and add extra configuration, generate a new IFWI 
3. Build SBL
4. stitch SBL + IFWI from the stage 2

I can't see my debug messages added in stage 1, therefore not sure if any of my changes applied. 
The FIT tool overrides what's been added in stage 1? 

Andrey

mauricema, 2020-05-14:
In step 1, the change in ScInit.c will not have any any impact on SBL and FSP.  Since the whole IFWI BIOS region will be replaced by SBL image in step 4. 

FIT tool normally will not change SBL Stage1.  If you added debug info in Stage1A, please make sure they are added after  "BoardInit (PostTempRamInit);".  This is the place where the UART will be initialized and available for use.

andreyv1978, 2020-05-15:
> About the reset happen during FSP training, if it is caused by change of FUNCTION DISABLE of internal SoC devices, you maybe try to comment out the below function call in [...]

The commented code like:
        DEBUG ((DEBUG_INFO, "Cold reset removed\n"));
        //BoardNotifyPhase (PostPciEnumeration);
        Status = EFI_SUCCESS;
  
Got the following log:
Cold reset removed
System BIOS ID is Intel
VbtAddress =0x7AB555C4 
VBT data found
VbtFileBuffer->HeaderVbtSize = 0x1BD3 
extended VBT supported
IgdOpRegion ended
  IPCBar0Address Address = 0xFE042000, Len = 0x00002000
  SSRAMBar0Address Address = 0xFE900000, Len = 0x00002000
  P2SBBaseAddress Old = 0xFFFFFFFF, New = 0xD0000000
  BIOS MAIL-BOX Data= 0xFED17080 
  BIOS MAIL-BOX Interface= 0xFED17084 
  GPIO0Addr Address = 0xD0C50000, Len = 0x0000076C
  GPIO1Addr Address = 0xD0C40000, Len = 0x00000764
  GPIO2Addr Address = 0xD0C70000, Len = 0x00000674
  GPIO3Addr Address = 0xD0C00000, Len = 0x00000654
  RpAddress Port[0] Address = 0x00140000
  RpAddress Port[1] Address = 0x00140001
  RpAddress Port[2] Address = 0x00130000
  RpAddress Port[3] Address = 0x00130001
  RpAddress Port[4] Address = 0x00130002
  RpAddress Port[5] Address = 0x00130003
ACPI Init
DMAR table update.
DmarTableUpdate() - start
McD0BaseAddress 0xE0000000, MchBar 0xFED10000, IgdMode 0x0002
GttMode 0x0003, GttMemSize 0x00800000
RMRR Base  address IGD 000000007B800000
RMRR Limit address IGD 000000007FFFFFFF
RMRR Base  address HECI 0000000000000000
RMRR Limit address HECI 00000000FFFFFFFF
VTD base address1 0x00000000FED64000
VTD base address2 0x00000000FED65000
 VtdLib: P2sbMmbase = 0xE0068000, IBDF = 0xFAF8
 VtdLib: IoApicId 0x1
DmarTableUpdate()  - end
Updated Hpet BaseAddressLower32Bit = 0xFED00000, EventTimerBlockId = 0x8086A701
ACPI Ret: Success
HdaNhltAcpi Addr 0x7A00A610, Length 0x0000053E
ACPI Table 0x544C484E - Appended
Loading Payload ID 0x49464555
Registering container EPLD
HASH verification for usage (0x00001000) with Hash Alg (0x1): Success
SignType (0x2) SignSize (0x100)  SignHashAlg (0x1)
RSA verification for usage (0x00001000): Success
HASH verification for usage (0x00000000) with Hash Alg (0x1): Success
Load Payload ID 0x49464555 @ 0x00800000
FV Format Payload
MP Init (Done)
Call FspNotifyPhase(40) ... NotifyPhaseApi() - Begin  [Phase: 00000040]
Unsupported FSP Notification Value
NotifyPhaseApi() - End  [Status: 0x80000003]
Unsupported

ASSERT_EFI_ERROR (Status = Unsupported)
ASSERT [Stage2]
 ..\slimbootloader\BootloaderCorePkg\Stage2\Stage2Support.c(54): 
!EFI_ERROR (Status)

STAGE_2: System halted!

Anything else I should Notify for the correct  Status ?

mauricema, 2020-05-15:
if you really want to ignore the reset request from FSP and do some tests,   you need to comment out all three instances for reset handling code in SBL.  You don't need to comment out "BoardNotifyPhase".
//  FspResetHandler (Status);
//  ASSERT_EFI_ERROR (Status);

But I have to point out that ignoring the reset request from FSP might cause potential functionality issue or security issue.  It should not be done in the final product.  The correct way is to figure out what caused the reset request from FSP and address that accordingly.

andreyv1978, 2020-05-19:
It looks like removing only reset in Stage2Support.c make a difference we need 
Thank you for your support, closing ticket

closed 2020-05-19 by andreyv1978