slimbootloader/slimbootloader #709: PMIC/VR configuration, the way it's done in Flash image tool

shehamb, 2020-05-12:
Hello, 
Our reference board for the hardware development is Leaf Hill Board. But we have changed the PMIC to TI's IC. From the FIT tool, i can see that there is a way to select which PMIC to use. I am wondering if there is any way to make the change in the SBL source code? If yes, where can i make this change?
Here is the snapshot of the PMIC configuration on the FIT:
[image]

Thanks in advance. 
Regards, 
Mahesh.

mauricema, 2020-05-12:
Hi, Mahesh,
I am not aware of any way to do it in SBL.   
PMIC needs to be enabled at very early stage even before SBL starts to execute,  So I guess it needs to be determined at platform power sequence phase.

thau-my, 2020-05-13:
Hi, Mahesh,

Do you intent to re-configure some registers in PMIC via I2C bus during SBL boot?

shehamb, 2020-05-13:
Hi,

i don't think so. of course, it depends on once we boot up the board, after
checking whether the new PMIC has properly initialized.

do you have any suggestions on this?

thau-my, 2020-05-14:
Do you want to check the internal register of PMIC from SBL during boot?
which interface in the SoC is the PMIC I2C connected to? If it is PMIC_PMC_I2C on the SoC, then you can't access directly from SBL. It need to go through IPC, I checked in SBL, IPC library is not included.

shehamb, 2020-05-19:
Hello Thau-my, 
Apologies for late reply. 
Yes, PMIC is connected to SoC through PMIC_PMC_I2C. So, for time being, we're using the FIT for modifying this setting.

closed 2020-05-19 by shehamb