slimbootloader/slimbootloader #715: [APL] PMIC instability

andreyv1978, 2020-05-13:
Hi

Our PMIC introduces instability while FSP is trained, it sometimes goes OFF and all suck in FspMemoryInit 'PHASE 3: FINAL SL CENTERING', but sometimes not. I have to restart board multiple times to finish FSPMemoryInit training. 
I've investigated an issue and found that, our board design is using PMIC through SMBus instead of I2C.  Also I've noticed you have a code update adding SMBus access library https://github.com/slimbootloader/slimbootloader/pull/126/commits/cc28746b7654ed8dc16c49bd3097ac35c1b0ed37
I've found the recent LeafHill design is not using SMBus anymore, but the SMBus maybe can be configured with FIT tool MISC_PMC_ENABLE.ALLOW_I2C_CONNECTION_WITH_SVID. Unfortunetely, the instability is not resolved using FIT tool with my board.
Do you know what I can configure in SBL to let to PMIC use SMBus ?
Because an issue is happening in Stage1B I am not certain what else can be set and where.

Andrey

mauricema, 2020-05-13:
SBL currently does not do anything on PMIC control today for APL platform.  We do have SMBUS library to communicate with devices on SMBUS.  It is more for SPD data read purpose only.   

If you have different PMIC chip,  I think you will need to get your support from your Intel representative to help on the design review and enabling.  There might be some settings (EX: straps, FIT configs, etc) that need to be configured properly for your board.  The instability might also be caused by board issues, (EX: PMIC did not provide the required voltage level for some reason, or PMIC timing is not configured properly to satisfy the required timing, etc.)

andreyv1978, 2020-05-14:
We have the same PMIC as in Leaf Hill CRB, the only difference is still uses SMBus interface instead of I2C.

mauricema, 2020-05-14:
I only saw SVID and I2C interfaces used for APL PMIC communication.  I have not seen any SMBUS based PMIC design.  And I am not sure if it is supported by APL SOC from board design perspective.  Please consult your board design team to confirm it.

andreyv1978, 2020-05-19:
I am still investigating PMIC instability, but I am not certain if this is SBL issue. 
PMIC is going OFF on the first FSP training (the very first boot). However, sometimes the FSP training pass, but PMIC is going off during yocto boot.
I don't see any PMIC configuration or anyhting that SBL can influence. Therefore, closing for now

andreyv1978, 2020-05-22:
Had to reopen this because of the issue found in #696

rprangar, 2020-05-26:
Hi

I have provided some inputs related to the SSC in its own thread.

I just wanted to provide a note on the use of I2C for PMIC - Private I2C is a secure interface for PMIC. Please contact your hardware representative for more guidance on this issue.

Thanks.

andreyv1978, 2020-05-27:
closing this based on the rprangar comment

closed 2020-05-27 by andreyv1978