slimbootloader/slimbootloader #721: Failure in FspMemoryInit on our own hardware (based off Leaf Hill CRB)

shehamb, 2020-05-19:
Hello,
We have based our hardware on the Intel's Leaf Hill CRB (Apollo Lake platform). between these two boards, everything is same except:
- we're using external PMIC (TI TPS650941)
- We are using a two-core SKU of the Apollo Lake SOC and the Leaf Hill CRB uses a four-core SKU of the SOC. 
- The Leaf Hill CRB uses Micron DRAM with a part number of MT53B512M32D2GZ-062 AIT:B (FBGA code of D9SWV). And our hardware uses Micron DRAM with a part number of MT53E512M32D2NP-046 WT:E (FBGA code D9WGB). Both of these parts are 2 GiB (0.5 G-word x 4 bytes per word) components: 512 Mebibit by 32 bits per word, dual rank. Both components are rated for use at 2400 MT per second. Both boards have a total of 8 GiB.

So, i built the slimbootloader for the Leaf Hill CRB and made the changes to the PMIC using the FIT. and i use this BIOS file generated using FIT and program it using the DediProg. 
When i boot up the board, it's getting stuck up at the FspMemoryInit(). I got the FSP debug library from you guys and have been using it. So i can see that it dumps lots of information. the last few lines of the console messages are:

Signal          CH RK ST HZ  192  96   1  M CC0 CC1 CC2 Vref Dly
--------------  -- -- -- --  --- --- ---  - --- --- --- ---- ----
RecEnPiCode_dq  03 00 00 00   --  --  01  -   -   -   -   -- 0001
RecEnPiCode_dq  03 00 01 00   --  --  01  -   -   -   -   -- 0001
RecEnPiCode_dq  03 00 02 00   --  --  01  -   -   -   -   -- 0001
RecEnPiCode_dq  03 00 03 00   --  --  01  -   -   -   -   -- 0001
RecEnPiCode_dq  03 01 00 00   --  --  01  -   -   -   -   -- 0001
RecEnPiCode_dq  03 01 01 00   --  --  01  -   -   -   -   -- 0001
RecEnPiCode_dq  03 01 02 00   --  --  01  -   -   -   -   -- 0001
RecEnPiCode_dq  03 01 03 00   --  --  01  -   -   -   -   -- 0001
CP 0xA5, Status[0]
In a Dead Loop: Acting Dead!
  
May i know when this dead lock kind of condition will happen?

I am enclosing the complete console log for your reference. 
[teraterm-brc3.log](https://github.com/slimbootloader/slimbootloader/files/4649475/teraterm-brc3.log)

Regards, 
Mahesh.

andreyv1978, 2020-05-19:
I've had the same problem at the beginning, took me some time to properly configure memory bit swizzling according to the board schematics.

shehamb, 2020-05-19:
This was on Leaf Hill board or your own hardware? if it was on Leaf hill, i
thought we don't need to make any changes to  configure memory bit
swizzling. At least i didn't have to do it on Leaf Hill.

andreyv1978, 2020-05-19:
no  we have our own board, but based Leaf Hill with a few changes
the memory settings were a first step to go forward and pass FSPmemoryInit

shehamb, 2020-05-19:
ok. is there any documentation available to figure out how to configure
memory bit swizzling according to schematics?

rprangar, 2020-05-19:
Hi

Memory parameters are heavily dependent on the hardware design. 

There is a Slim Bootloader Application Note for Whiskey Lake / Coffee Lake Refresh S / Coffee Lake Refresh H Platform that provides some information on how to configure memory parameters. This document (#615394) is available on https://www.intel.com/content/www/us/en/design/resource-design-center.html.

On a different note, this kind of question may be more appropriate for the mailing list rather than filing as an SBL issue. Mailing list is here - https://lists.01.org/hyperkitty/list/sbl-devel%40lists.01.org/

Please let us know if we can close this issue.

Thanks
Ravi

shehamb, 2020-05-20:
Hello Ravi,

Thanks for the links. Yes, i will close this issue.

But i still couldn't find information about how do i cross verify swizzle
settings. Is there any application note for that?

Regards,
Mahesh.

andreyv1978, 2020-05-20:
I've got it from someone who designed a board and looked at the schematics to reverify settings

thau-my, 2020-05-20:
If swizzling is configured incorrectly, the FspMemoryInit will halt at very early stage of MemoryInit due to mismatch of DQ signal. 
The swizzling configuration base on Leafhill board can be found in CfgData_Memory.dsc
If you need to explanation on how to configure the byte swizzling, you can refer to doc#559810 Apollo Lake Platform IA Firmware Spec volume 1 section 6.1.3 DQ mapping for CH0 to CH3.

thau-my, 2020-05-20:
I don't have MT53E512M32D2NP-046 WT:E datasheet from Micron. But if it is a newer LP4 module, typically it will have a higher density die (die shrink from 2 to 1), If that is the case, you might want to change a) the dram density for all channel (Ch0~3) to 0x4 (16Gb),  b) Rank Enable (for Ch0~Ch3) to 0x1 (SingleRank).

thau-my, 2020-05-20:
a) dram density (Ch0 to Ch3): 0x4 (16Gb)
b) Rank Enable (Ch0 to Ch3) to 0x1

shehamb, 2020-05-20:
Hello Thau-my,

Thank you for your suggestions. These settings have worked and SBL is no
more stuck up at fspMemoryInit()

Regards,
Mahesh.

closed 2020-05-21 by rprangar