slimbootloader/slimbootloader #729: [APL] SD card boot does not work on Intel CRB platform with SBL

mauricema, 2020-05-29:
On Oxbow Hill board,  change boot option to boot from SD card,  the OsLoader could not initialize the SD card successfully.

rprangar, 2020-05-29:
copying old thread...

@shehamb

shehamb commented 2 days ago
So, i altered the IFWI image first and then used StitchLoader.py to
generate the final IFWI image. This has worked. SBL boots up all the way
through.
Next, i wanted to get the UEFI shell. So, built the SBL again to include
UEFI payload:
python3 BuildLoader.py build apl -p
"OsLoader.efi:LLDR:Lz4;UefiPld.fd:UEFI:Lzma"

With this, SBL boots up with UEFI shell. But i don't see the SD card
mapping (fs0:, fs1: and so on).
UEFI Interactive Shell v2.287477C2-69C7-11D2-8E39-00A0C969723B 78BDC6A0
EDK IIlProtocolInterface: 752F3136-4E16-4FDC-A22A-E5F46812F4CA 78BDBE18
UEFI v2.70 (EDK II, 0x00010000)008-7F9B-4F30-87AC-60C9FEF5DA4E 78A993C0
Mapping table
BLK0: Alias(s):
PciRoot(0x0)/Pci(0x1C,0x0)/eMMC(0x0)/Ctrl(0x0)
BLK1: Alias(s):
PciRoot(0x0)/Pci(0x1C,0x0)/eMMC(0x0)/Ctrl(0x1)
BLK2: Alias(s):
PciRoot(0x0)/Pci(0x1C,0x0)/eMMC(0x0)/Ctrl(0x2)
Press ESC in 2 seconds to skip startup.nsh or any other key to continue.
Shell>
…
@mauricema

Member
mauricema commented 2 days ago
Please ensure your SD card related GPIO are configured properly and the SD card power is applied correctly. It might also be related to UEFI SD/eMMC driver initialization flow.

@shehamb

shehamb commented 2 days ago
UEFI bios is able to detect SD card on our new hardware.
And for the same code base of APL platform on SBL, i was able to see SD
card accessible through UEFI shell on the UP2 board. So i suppose SD card
power and UEFI SD/eMMC driver initialization flow related issues can be
ruled out.
…
@thau-my

thau-my commented 2 days ago
Is there any error related to SD card initialization such as tuning failure during SBL boot? Could you provide the full SBL log?

@shehamb

shehamb commented yesterday
hello,
i don't see any error as such. enclosing the log for your reference.
…
@mauricema

Member
mauricema commented yesterday
For SD card power control, it is board specific design.
EX: on CRB, it is decided by the GPIO 183. The pin level has to be set to 0 in order to power on the SD CARD. For your board, you need to see which GPIO will control the power and make sure it is set to correct level.

@mauricema

Member
mauricema commented yesterday
Also just want to be clear, you meant SD card (PCI device 0x1B, function 0) , not eMMC (PCI device 0x1C, functin 0) right ? UP2 board does not have SD card as far as I know. It only provides eMMC.

@shehamb

shehamb commented 26 minutes ago
Hello Maurice,
I am not sure if it could be a SD card power control issue because on the
same hardware, i am able to boot the UEFI bios (with UEFI shell) and see
the SD card mapping.
And yes, my bad. I meant, eMMC on UP2 board.
Regards,
Mahesh.
…

mauricema, 2020-05-29:
Submitted a patch #731.
Please look into SubmittSdcardPowerUp() in Stage2BoardInitLib.c on the changes required for SD card enabling for Intel CRB.  
Please note, it is for SD card only, not for eMMC.

shehamb, 2020-06-01:
> Submitted a patch #731.
> Please look into SubmittSdcardPowerUp() in Stage2BoardInitLib.c on the changes required for SD card enabling for Intel CRB.
> Please note, it is for SD card only, not for eMMC.

Hello Maurice, 
Thank you for the patch. So, as per your suggestions, i just took the changes in Stage2BoardInitLib.c. And with that, SD card is getting detected and eventually i am able to boot up the kernel from the SD card. But, i don't get the control of the UEFI shell. Generally, there will be timeout of 5 seconds on the shell before it executes startup.nsh. And moreover, i think i don't see any shell as such. Here are the last few lines of the log:
------------------------------------------------
[Bds]OsIndication: 0000000000000000
[Bds]=============Begin Load Options Dumping ...=============
  Driver Options:
  SysPrep Options:
  Boot Options:
    Boot0000: UiApp 		 0x0109
    Boot0001: UEFI SD Device 		 0x0001
    Boot0002: UEFI eMMC Device 		 0x0001
    Boot0003: UEFI eMMC Device 2 		 0x0001
    Boot0004: UEFI eMMC Device 3 		 0x0001
    Boot0005: UEFI Shell 		 0x0001
  PlatformRecovery Options:
    PlatformRecovery0000: Default PlatformRecovery 		 0x0001
[Bds]=============End Load Options Dumping=============
[Bds]BdsWait ...Zzzzzzzzzzzz...
[Bds]BdsWait(3)..Zzzz...
[Bds]BdsWait(2)..Zzzz...
[Bds]BdsWait(1)..Zzzz...
[Bds]Exit the waiting!
PROGRESS CODE: V03051007 I0
[Bds]Stop Hotkey Service!
[Bds]UnregisterKeyNotify: 000C/0000 Success
[Bds]UnregisterKeyNotify: 0002/0000 Success
[Bds]UnregisterKeyNotify: 0000/000D Success
PROGRESS CODE: V03051001 I0
Memory  Previous  Current    Next   
 Type    Pages     Pages     Pages  
======  ========  ========  ========
  09    00000008  00000000  00000008
  0A    00000004  00000000  00000004
  00    00000004  00000002  00000004
  06    000000C0  0000003F  000000C0
  05    00000080  0000002D  00000080
[Bds]Booting UEFI SD Device
 BlockSize : 512 
 LastBlock : 745FFF 
FSOpen: Open '\EFI\BOOT\BOOTX64.EFI' Success
[Bds] Expand PciRoot(0x0)/Pci(0x1B,0x0)/SD(0x0) -> PciRoot(0x0)/Pci(0x1B,0x0)/SD(0x0)/HD(1,MBR,0xE0E
27DFF,0x2000,0x744000)/\EFI\BOOT\BOOTX64.EFI
PROGRESS CODE: V03058000 I0
[Security] 3rd party image[0] can be loaded after EndOfDxe: PciRoot(0x0)/Pci(0x1B,0x0)/SD(0x0)/HD(1,
MBR,0xE0E27DFF,0x2000,0x744000)/\EFI\BOOT\BOOTX64.EFI.
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B 7880F040
Loading driver at 0x000787EB000 EntryPoint=0x000787F2F24 
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF 7880FE98
ProtectUefiImageCommon - 0x7880F040
  - 0x00000000787EB000 - 0x0000000000024000
PROGRESS CODE: V03058001 I0
PROGRESS CODE: V01040001 I0
I
PROGRESS CODE: V01040001 I0
PROGRESS CODE: V01040001 I0
Copyright 2012-2018 Wind River Systems

VxWorks UEFI Stage 1 Bootstrap (64-bit version)
Built on: Jul 18 2019/15:20:21

ACPI 2.0 table at: 000000007A004000

Loading from disk...
Opening [\EFI\BOOT\bootapp.cfg]... 
Opening file [\EFI\BOOT\bootapp.cfg] failed: Not Found
Opening [bootapp.sys]... 
Opening file [bootapp.sys] failed: Not Found
Opening [\EFI\BOOT\bootapp.sys]... FSOpen: Open '\EFI\BOOT\bootapp.sys' Success
Read 7455520 bytes.

Loading 64-bit ELF image.

------------------------------------------------------------------

Enclosing the complete log for your reference. Btw, i tried taking in rest of the changes from #731 as well.  but same behavior. 
[brc3-uefi.txt](https://github.com/slimbootloader/slimbootloader/files/4712373/brc3-uefi.txt)

mauricema, 2020-06-01:
Glad to know you have SD card showing up.
For the Shell control, it is part of the UEFI payload,  has nothing to do with SBL.
If you have a USB keyboard connected to your board, you will have to keep pressing 'F2' key after UEFI payload launching in order to boot into setup screen and then you can select boot to shell.

mauricema, 2020-06-03:
Can this be closed now ?

shehamb, 2020-06-04:
Yes, we can close this. Apologies for late response. I am accessing the
board remotely. So i had some technical issues accessing the board.
On other note, is network driver enabled by default for APL platform? When
i break into the shell and do "ifconfig -l", i don't see any n/w interface.
I tried to look into the packages to see if something needs to be enabled
for this to work. But couldn't really figure out.

mauricema, 2020-06-04:
OK, I will close it.
For network configuration, it is not on by default.  
You will have to pull in drivers required just as the normal UEFI BIOS.

closed 2020-06-04 by mauricema

shehamb, 2020-06-05:
Ok. And on another note, i think there is something amiss with the GPIO
configuration, for the SD card fix.
https://github.com/slimbootloader/slimbootloader/blob/master/Platform/ApollolakeBoardPkg/CfgData/CfgData_Gpio.dsc
As
per this file:
# !BSF SUBT:{GPIO_TMPL:GPIO_186 :0x0005e0c0 :*0x5c105004*}

And in the older version:
 # !BSF SUBT:{GPIO_TMPL:GPIO_186 :0x0005e0c0 :*0x5c105002*}

So, with this new version, sd card doesn't show in the mapping. But with
the reverted value of GPIO_186, it does. I realised this when i was doing a
fresh cloned build on windows.

mauricema, 2020-06-05:
Again, please use the latest code.  I think you are mixing changes.

The latest code uses:  
# !BSF SUBT:{GPIO_TMPL:GPIO_186 :0x0005e0c0 :*0x5c105004*}
And in the SD detection, we should use  SW_GPIO_177 for SD detection.   Please check SdcardPowerUp() and make sure your code matches the latest code.  This code has been tesetd on Leafhill and worked as expected.

If you use the old GPIO as
# !BSF SUBT:{GPIO_TMPL:GPIO_186 :0x0005e0c0 :*0x5c105002*}
then you have to use GPIO_186 for SD detection in SdcardPowerUp().  It also works.

Our recommendation is to use the same detection as our latest code in github.

PR #731 (2020-06-01): [APL] Fix SD card boot issue on Intel CRB boards
This patch fixed OsLoader boot from SD card issue on Intel APL CRB
borads. The SD/eMMC library was updated to follow the proper sequence
for SD card. Also platform code was updated to detect SD card and
apply SD card power.

It fixed #729.
commit 8eb31ee3 (2020-06-01): [APL] Fix SD card boot issue on Intel CRB boards

This patch fixed OsLoader boot from SD card issue on Intel APL CRB
borads. The SD/eMMC library was updated to follow the proper sequence
for SD card. Also platform code was updated to detect SD card and
apply SD card power using proper GPIO pins.

It fixed #729.