slimbootloader/slimbootloader #738: Build on Windows10 fails

shehamb, 2020-06-04:
Hi, 
i have set up the SBL on windows10 as per the instructions 
https://slimbootloader.github.io/getting-started/build-host-setup.html

but when i do the build (for APL), i get this error:
--------------------------------------------------------------------------------------------------------------
Building ... d:\git\sbl\slimbootloader-master\BootloaderCorePkg\Library\FspApiLib\FsptApiLib.inf [IA32]
StageLib.c
CpuExceptionHandler.c
d:\git\sbl\slimbootloader-master\Platform\CommonBoardPkg\AcpiTables\Tpm\Tpm2.aslc(44): error C2078: too many initializers
d:\git\sbl\slimbootloader-master\Platform\CommonBoardPkg\AcpiTables\Tpm\Tpm2.aslc(41): error C2078: too many initializers
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\Hostx86\x86\cl.exe"' : return code '0x2'
Stop.
Building ... d:\git\sbl\slimbootloader-master\Platform\ApollolakeBoardPkg\Library\Stage1ABoardInitLib\Stage1ABoardInitLib.inf [IA32]

build.py...
 : error 7000: Failed to execute command
        C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\Hostx86\x86\nmake.exe /nologo tbuild [d:\edk2\Build\BootloaderCorePkg\DEBUG_VS2017\IA32\Platform\ApollolakeBoardPkg\AcpiTables\AcpiTables]

build.py...
 : error 7000: Failed to execute command
        C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.10.25017\bin\Hostx86\x86\nmake.exe /nologo tbuild [d:\edk2\Build\BootloaderCorePkg\DEBUG_VS2017\IA32\BootloaderCorePkg\Library\FspApiLib\FspmApiLib]

build.py...
 : error F002: Failed to build module
        d:\git\sbl\slimbootloader-master\Platform\ApollolakeBoardPkg\AcpiTables\AcpiTables.inf [IA32, VS2017, DEBUG]

- Failed -
Build end time: 10:10:31, Jun.04 2020
Build total time: 00:00:14

Error in running process:
  build.bat --platform BootloaderCorePkg\BootloaderCorePkg.dsc -b DEBUG --arch IA32 --tagname VS2017 -n 12 -y Report.log -Y PCD -Y FLASH -Y LIBRARY
--------------------------------------------------------------------------------------------------------------

[win10-build-log.txt](https://github.com/slimbootloader/slimbootloader/files/4727612/win10-build-log.txt)

On another note, can i make a suggestion here? it would be a good idea to include a note (on 
https://slimbootloader.github.io/getting-started/build-host-setup.html#building-on-windows )
that one needs to follow these steps before building SBL
https://github.com/tianocore/tianocore.github.io/wiki/UDK2018-How-to-Build#how-to-build-windows-system

rprangar, 2020-06-04:
Can you please try a fresh clone and build from it? From the build log, it looks like this is not a clean build.

shehamb, 2020-06-05:
Fresh clone and build worked! Thank you.

closed 2020-06-05 by rprangar