slimbootloader/slimbootloader #769: Capsule image upgrade failed with error "Image verification failed, Security Violation!"

jns-git, 2020-07-08:
I create a capsule update image, and put the image into a USB disk, then enter SBL shell, use command “fwupdate” to trigger capsule image upgrade. Then I meet error when system resume from a reset. 
### Error log:

```
....
Payload entry: 0x7AAE72B0

Jump to payload

Payload startup
ACPI PmTimer Base: 0x408
PCI Express  Base: 0xE0000000
Starting Firmware Update

=================Read Capsule Image==============
BootMediumPciBase(0xE00A8000)
Init USB XHCI - Success
XHCI: Don't support multi-TT feature for Hub now. (force to disable MTT)
XhcPeiCheckUrbResult: STALL_ERROR! Completecode = 6
XhcPeiControlTransfer: error - Device Error, transfer - 2
Enumerate Bus - Success
Found 5 USB devices on bus
Use the 1st mass storage device
Found 1 mass storage devices
find boot partition
Partition type: MBR  (1 logical partitions)
Find partion success
Find partition
Detected FAT on HwDev 0 Part 0
Capsule Image found, ImageSize=0x6FE28C
First 256Bytes of capsule image
  00000000: 58 AE 3E 1A 80 B5 EF 4F-AC A3 A1 6D 9E 00 DF 5F  *X.>....O...m..._*
  00000010: 40 00 00 00 01 00 00 00-00 00 00 00 80 E1 6F 00  *@.............o.*
  00000020: 0C 01 00 00 40 00 00 00-38 E0 6F 00 78 E0 6F 00  *....@...8.o.x.o.*
  00000030: 08 01 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000040: 01 00 00 00 00 00 01 00-10 00 00 00 00 00 00 00  *................*
  00000050: 02 00 00 00 13 68 5C 60-C7 C2 42 42 9C 27 50 A4  *.....h\`..BB.'P.*
  00000060: C3 63 DB A4 01 00 00 00-00 E0 6F 00 00 00 00 00  *.c........o.....*
  00000070: 42 49 4F 53 00 00 00 00-AA 55 00 00 0D 00 01 00  *BIOS.....U......*
  00000080: 00 00 00 00 00 00 00 00-03 00 00 00 0A 00 69 04  *..............i.*
  00000090: 09 00 00 00 00 00 00 00-00 00 00 00 0A 00 00 00  *................*
  000000A0: 00 02 00 00 10 00 00 00-05 00 00 00 00 40 14 00  *.............@..*
  000000B0: 00 80 10 00 01 00 00 00-00 50 00 00 00 A0 00 00  *.........P......*
  000000C0: 0C 00 00 00 00 00 00 00-00 00 00 00 0D 00 00 00  *................*
  000000D0: 00 00 00 00 00 00 00 00-02 00 00 00 00 F0 01 00  *................*
  000000E0: 00 C0 05 00 11 00 00 00-10 02 00 00 08 01 00 00  *................*
  000000F0: 00 00 00 00 00 10 00 00-00 40 00 00 0E 00 00 00  *.........@......*
CapsuleImage: 0x793A9010, CapsuleSize: 0x6FE28C
HASH verification for usage (0x00000400) with Hash Alg (0x1): Security Violation
First 32 Bytes Input Data
  00000000: CD F8 69 85 D0 50 93 B9-03 3A E5 F6 B6 F6 EE 89  *..i..P...:......*
  00000010: 7B 2B B5 AF FA 31 4A 45-58 89 33 51 7E 14 93 55  *.+...1JEX.3Q...U*
Last 32 Bytes Input Data
  00000000: 15 63 76 6A 8A 05 5C E9-68 1A CB 03 75 32 6C 3A  *.cvj..\.h...u2l:*
  00000010: C8 D5 C7 E8 8A 87 51 F4-8B 33 80 17 00 01 00 01  *......Q..3......*
Image Digest
  00000000: D7 72 84 C6 31 57 05 C4-A1 AD 33 17 D5 B0 68 5F  *.r..1W....3...h_*
  00000010: 07 D8 37 21 D2 C4 7C CA-E2 83 46 CF 86 7A 7E 40  *..7!......F..z.@*
HashStore Digest
Image verification failed, Security Violation!
AuthenticateCapsule, Status = 0x8000001A
Firmware update failed with Status = Security Violation
Firmware update Done! clear CSE flag to normal boot mode.
HECI SecMode 0
Group    =00000020
Command  =00000007
IsRespone=00000001
Result   =00000000
Rebooting ...
BtGuard: VB : 0, MB : 0 
...
```

###  The environment I use
Intel CRB board: Apollo Lake Leaf Hill.

### How I build SBL image:

I use command to build SBL image. python BuildLoader.py build apl
But I meet below error:

```
!!! ERROR: SBL_KEY_DIR is set to directory ../SblKeys/ does not exist!!! 

!!! PRE-REQUISITE: Path to SBL_KEY_DIR has to be set with SBL KEYS DIRECTORY !!! 
!!! Generate keys using GenerateKeys.py available in BootloaderCorePkg/Tools directory !!! 
!!! Run $python BootloaderCorePkg/Tools/GenerateKeys.py -k $PATH_TO_SBL_KEY_DIR !!!
!!! Set SBL_KEY_DIR environ with path to SBL KEYS DIR !!!
!!! Windows $set    SBL_KEY_DIR=$PATH_TO_SBL_KEY_DIR !!!
!!! Linux   $export SBL_KEY_DIR=$PATH_TO_SBL_KEY_DIR !!!

```
So I create Sblkeys folder, and use GenerateKeys.py to generate keys.
```
mm@mm-VirtualBox:~/code/slimbootloader$ cd ..
mm@mm-VirtualBox:~/code$ mkdir Sblkeys
mm@mm-VirtualBox:~/code$ cd slimbootloader/
mm@mm-VirtualBox:~/code/slimbootloader$ python BootloaderCorePkg/Tools/GenerateKeys.py -k "../Sblkeys"
BootloaderCorePkg/Tools/GenerateKeys.py:118: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if args.KeySize is 'ALL':
Generating RSA private key, 2048 bit long modulus (2 primes)
.+++++
.........................................................+++++
e is 65537 (0x010001)
Generating RSA private key, 2048 bit long modulus (2 primes)
...............................................................................................................................................................+++++
..................................................................................+++++
e is 65537 (0x010001)
Generating RSA private key, 2048 bit long modulus (2 primes)
...........................................+++++
.....+++++
e is 65537 (0x010001)
Generating RSA private key, 2048 bit long modulus (2 primes)
....................................................................................+++++
.................................+++++
e is 65537 (0x010001)
Generating RSA private key, 2048 bit long modulus (2 primes)
.........................................................................+++++
.........+++++
e is 65537 (0x010001)
../Sblkeys/OS1_TestKey_Pub_RSA2048_priv.pem
Generating RSA private key, 2048 bit long modulus (2 primes)
................+++++
.................................................................................................................+++++
e is 65537 (0x010001)
../Sblkeys/OS1_TestKey_Pub_RSA2048.pem
writing RSA key
Generating RSA private key, 3072 bit long modulus (2 primes)
........................................++++
..........................++++
e is 65537 (0x010001)
Generating RSA private key, 3072 bit long modulus (2 primes)
................................................................................................++++
..................++++
e is 65537 (0x010001)
Generating RSA private key, 3072 bit long modulus (2 primes)
.....................................................................++++
..................................++++
e is 65537 (0x010001)
Generating RSA private key, 3072 bit long modulus (2 primes)
................................................................................................................................++++
...................++++
e is 65537 (0x010001)
Generating RSA private key, 3072 bit long modulus (2 primes)
.........................................................................................................++++
.........++++
e is 65537 (0x010001)
../Sblkeys/OS1_TestKey_Pub_RSA3072_priv.pem
Generating RSA private key, 3072 bit long modulus (2 primes)
........................++++
.................................................................................................++++
e is 65537 (0x010001)
../Sblkeys/OS1_TestKey_Pub_RSA3072.pem
writing RSA key
mm@mm-VirtualBox:~/code/slimbootloader$
```

Then build apl SBL image again, and it is successful.
`python BuildLoader.py build apl`

### How I stitching SBL image with Apollo Lake platform firmware image.
python Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i LEAFHILD.X64.0071.R01.1809030849.bin -s Outputs/apl/Stitch_Components.zip -o sbl_lfh_ifwi.bin

### How I create capsule image.
`python Platform/ApollolakeBoardPkg/Script/StitchLoader.py -b sbl.bios.bin -i LEAFHILD.X64.0071.R01.1809030849.bin -s Outputs/apl/Stitch_Components.zip -o sbl_lfh_ifwi.bin`

`python ./BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS sbl.bios.bin  -k ./Sblkeys/FirmwareUpdateTestKey_Priv_RSA2048.pem -o FwuImage.bin`

Then I use FwuImage.bin to do capsule upgrade, I meet the error as before. 
I try to use other *.pem keys, I meet the same error.
What's wrong, How can I go ahead?

mauricema, 2020-07-08:
I did not see anything wrong for your process.  The error message indicates the key used for capsule signing does not match the one built into SBL image. 

For command  line "python ./BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS  sbl.bios.bin  -k ./Sblkeys/FirmwareUpdateTestKey_Priv_RSA2048.pem -o FwuImage.bin",  are you using the correct key path? 
If your current directory is under slimbootloader, then it should be "../Sblkeys/FirmwareUpdateTestKey_Priv_RSA2048.pem" rather than "./Sblkeys/FirmwareUpdateTestKey_Priv_RSA2048.pem".   Please note the difference between ".." and "."   

Also, you can try to use KEY ID for signing as below.  In this way it will automatically try to find the key pointed by your env variable SBL_KEY_DIR.
python ./BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS sbl.bios.bin -k  KEY_ID_FIRMWAREUPDATE_RSA2048  -o FwuImage.bin

mauricema, 2020-07-10:
Do you still have issue ?   Is it addressed ?

jns-git, 2020-07-13:
Yes, It is workable.  Thanks~

closed 2020-07-13 by jns-git

Jiayan5877, 2021-02-02:
Dear Sir,

I have the same error "Image verification failed, Security Violation!" when triggering Firmware Update on EHL platform.
BTW, I make sure the key path should be right.
Here is my command referred to https://slimbootloader.github.io/supported-hardware/ehl-crb.html?highlight=fwupdate
python ./SblOpen/BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS ./Outputs/ehl/SlimBootloader.bin -k ./SblKeys/MasterTestKey_Priv_RSA2048.pem -o FwuImage.bin
or
python ./SblOpen/BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS ./Outputs/ehl/SlimBootloader.bin -k ./SblKeys/FirmwareUpdateTestKey_Priv_RSA2048.pem -o FwuImage.bin
Is my command wrong?

Here is my error log.
[fwupdate_Security Violation.log](https://github.com/slimbootloader/slimbootloader/files/5907921/fwupdate_Security.Violation.log)

May I know which key in SubKeys directory used for fwupdate?
[image]

Best Regards,

raghavag, 2021-02-02:
Hello,

EHL uses 3k signing format. Please use following command to create capsule

python ./SblOpen/BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS ./Outputs/ehl/SlimBootloader.bin -k ./SblKeys/FirmwareUpdateTestKey_Priv_RSA3072.pem -o FwuImage.bin

Thanks,
Raghava

[Image removed by sender. Sblkeys]<https://user-images.githubusercontent.com/53555872/106539281-a2de8c80-6538-11eb-9ba1-737474bf3049.jpg>

Jiayan5877, 2021-02-02:
Dear Raghava,

Thank for your comment.
When I use FirmwareUpdateTestKey_Priv_RSA3072.pem, Fwupdate can work but system cannot be powered on after resetting.
I am trying to fix this problem.
Is it possible SlimBootloader.bin causes this problem?

[fwupdate_but cannot boot.log](https://github.com/slimbootloader/slimbootloader/files/5908841/fwupdate_but.cannot.boot.log)

BTW, May I know why EHL uses 3k signing format?

Best Regards,

Jiayan5877, 2021-02-03:
Dear Raghava,

To execute following command, Fwupdate can work and system can be powered on after resetting.
python ./SblOpen/BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS ./ifwi_stitch/Temp/BiosRegion.bin -k ./SblKeys/FirmwareUpdateTestKey_Priv_RSA3072.pem -o FwuImage.bin

May I know how to make out EHL use 3k or 2k signing format?

Best Regards,

lsubash, 2021-02-03:
For turning on RSA 2K in EHL verified boot follow below steps.

In Platform\ElkhartlakeBoardPkg\BoardConfig.py,
        # RSA2048 or RSA3072
        self._RSA_SIGN_TYPE          = 'RSA3072'   **==>** "RSA2048'

        # 'SHA2_256' or 'SHA2_384'
        self._SIGN_HASH              = 'SHA2_384'   **==>** "SHA2_256"

For Fwupdate,
python ./SblOpen/BootloaderCorePkg/Tools/GenCapsuleFirmware.py -p BIOS ./ifwi_stitch/Temp/BiosRegion.bin -k ./SblKeys/FirmwareUpdateTestKey_Priv_RSA**2048**.pem -o FwuImage.bin

We use RSA3K/SHA2384 in EHL by default. This is better security. Also its recommendation from Intel crypto.

Jiayan5877, 2021-02-03:
Dear Subash,

Thank for your explanation.
I don't have other problem.

Best Regards,