slimbootloader/slimbootloader #777: APL: PCIe RP04 is not visible in "pci" command from sbl shell

shehamb, 2020-07-15:
Hello, 
We're using the custom hardware based on Leaf Hill CRB. We need to use the RP04 (device ID 0x5AD9). But it seems to be missing when i check the list of PCI devices using "pci" command from the SBL shell. 

While i was browsing through "pcie" issues over here, i have noticed that there was similar issue (https://github.com/slimbootloader/slimbootloader/pull/216)

So, i added these in CfgData_Int_LeafHill.dlt
PCIE_RP_CFG_DATA.PcieRpFeatures0.En      | 0x1
PCIE_RP_CFG_DATA.PcieRpFeatures1.En      | 0x1
PCIE_RP_CFG_DATA.PcieRpFeatures2.En      | 0x1
PCIE_RP_CFG_DATA.PcieRpFeatures3.En      | 0x1
PCIE_RP_CFG_DATA.PcieRpFeatures4.En      | 0x1

But i still don't see the RP04 in the list. I am enclosing the console log for your reference. 

[sbl-pcie-rp.txt](https://github.com/slimbootloader/slimbootloader/files/4924286/sbl-pcie-rp.txt)
Here is the snapshot of the "pci" command output on the shell:

Shell> pci
00:00.0 8086:5AF0 Bridge Device
00:02.0 8086:5A85 Display Controller
00:03.0 8086:5A88 Multimedia Device
00:0E.0 8086:5A98 Multimedia Device
00:0F.0 8086:5A9A Simple Communications Controllers
00:0F.1 8086:5A9C Simple Communications Controllers
00:0F.2 8086:5A9E Simple Communications Controllers
00:11.0 8086:5AA2 Pre 2.0 device
00:12.0 8086:5AE3 Mass Storage Controller
00:13.0 8086:5AD8 Bridge Device
00:13.2 8086:5ADA Bridge Device
00:14.0 8086:5AD6 Bridge Device
00:14.1 8086:5AD7 Bridge Device
00:15.0 8086:5AA8 Serial Bus Controllers
00:15.1 8086:5AAA Serial Bus Controllers
00:16.0 8086:5AAC Data Acquisition & Signal Processing Controllers
00:16.1 8086:5AAE Data Acquisition & Signal Processing Controllers
00:16.2 8086:5AB0 Data Acquisition & Signal Processing Controllers
00:16.3 8086:5AB2 Data Acquisition & Signal Processing Controllers
00:17.0 8086:5AB4 Data Acquisition & Signal Processing Controllers
00:17.1 8086:5AB6 Data Acquisition & Signal Processing Controllers
00:17.2 8086:5AB8 Data Acquisition & Signal Processing Controllers
00:17.3 8086:5ABA Data Acquisition & Signal Processing Controllers
00:18.0 8086:5ABC Data Acquisition & Signal Processing Controllers
00:18.1 8086:5ABE Data Acquisition & Signal Processing Controllers
00:18.2 8086:5AC0 Data Acquisition & Signal Processing Controllers
00:18.3 8086:5AEE Data Acquisition & Signal Processing Controllers
00:19.0 8086:5AC2 Data Acquisition & Signal Processing Controllers
00:19.1 8086:5AC4 Data Acquisition & Signal Processing Controllers
00:19.2 8086:5AC6 Data Acquisition & Signal Processing Controllers
00:1B.0 8086:5ACA Base System Peripherals
00:1C.0 8086:5ACC Base System Peripherals
00:1E.0 8086:5AD0 Base System Peripherals
00:1F.0 8086:5AE8 Bridge Device
00:1F.1 8086:5AD4 Serial Bus Controllers
02:00.0 8086:1531 Network Controller

Regards, 
Mahesh.

mauricema, 2020-07-15:
PCIe Root Port will only be visible when there are PCI device detected behind it.  
If there is no card,  the Root Port will be hidden.
Also, please note the mapping below:  
RP0 ->  PCI 00.14.0 (0x8086:0x5ad6)  -> CPU Signal PCIE4
RP1 ->  PCI 00.14.1 (0x8086:0x5ad7)  -> CPU Signal PCIE5
RP2 ->  PCI 00.13.0 (0x8086:0x5ad8)  -> CPU Signal PCIE0
RP3 ->  PCI 00.13.1 (0x8086:0x5ad9)  -> CPU Signal PCIE1
RP4 ->  PCI 00.13.2 (0x8086:0x5ada)  -> CPU Signal PCIE2
RP5 ->  PCI 00.13.3 (0x8086:0x5adb)  -> CPU Signal PCIE3
So you are actually looking into RP3.

mauricema, 2020-07-15:
Also in FIT,  you might need to config your PCIe root port bifurcation correctly.  Leafhill used 1x22x1, so both RP2 and RP3 will be routed to a single x2 slot.  If you need to use RP2 and RP3 separately, it needs to be 1x4 instead.

shehamb, 2020-07-15:
Hello Maurice,

Thank you for your email. Actually, we have connected the FPGA on RP3.
[image: image.png]

And as per the section 3 of Intel Document 559811 *Apollo Lake Platform
Intel Architecture Firmware Specification (Volume 2 of 2)*.
[image: image.png]

I tried changing the 1x22x1 to 1x4 in FIT. But that did not work.

Regards,
Mahesh.

mauricema, 2020-07-15:
Hi, Mahesh,

Since your PCIe configuration is different from the standard LeafHill,  you might need to configure it accordingly.  For PCIe configuration,  other than enabling, there are also PCIe clk req, reset pin, power pin configuration.  Depending on your board design, those have to be configured to match your board design.  We don't have your hardware, so it is not easy for us to figure out what could be wrong on your side.  You will need to check those settings.

Thanks
Maurice

shehamb, 2020-07-16:
Hi Maurice,

Just to let you know, changing the  PCIe root port bifurcation in FIT from
1x22x1 to 4x1 (this is the only configuration wherein port-2 is not
disabled) worked. Now i am able to see the RP-03 on the SBL as well as
VxWorks shell.

Regards,
Mahesh.

mauricema, 2020-07-16:
Hi, Mahesh,

Thank you for the updates.  Glad to know it worked.  :)
Sorry, it stated incorrectly.  I meant to change it to 4x1 so that each root port is separated.  But in my reply,  somehow I mentioned 1x4.   

Regards,
Mahesh.

mauricema, 2020-07-16:
can we close this then ?

shehamb, 2020-07-17:
Yes, we can close this.

Regards,
Mahesh.

mauricema, 2020-07-17:
closed as per discussion.

closed 2020-07-17 by mauricema