slimbootloader/slimbootloader #778: Split IFWI binary to SpiChunkN.bin

andreyv1978, 2020-07-16:
Hi

I know this is not a right place to create an issue ticket, but I would appreciate of any help
From the latest check-in to the https://github.com/tianocore/edk2-platforms/tree/devel-IntelAtomProcessorE3900 it looks like you guys the last touched the source
I am still trying to disable SSC as is my usual  batlle field, and the image generated from the devel-IntelAtomProcessorE3900 seems is doing that in a correct format from UEFI bios menu. However, I have to change IFWI image with a FIT tool to match my board design.
The problem is that the FIT tool generates a single IFWI image, but the stitching process in edk2 requires following:
1. SpiChunk1.bin
2. SpiChunk1SpiAccessControl.bin
3. SpiChunk2.bin
4. SpiChunk4.bin
5. 8MB_SPI_IFWI_Layout.map

Also the document in pp19 section ‘Configuration via Intel FIT tool’
https://software.intel.com/content/dam/develop/external/us/en/documents/uefi-firmware-porting-guide-for-the-intel-atom-processor-e3900-series-820187.pdf
recommends to make an adjustment to required Flex I/O settings. 

Can someone help me to figure out the spliting process ?
 
Regards,
Andrey

mauricema, 2020-07-16:
Hi, Andrey,

This is a totally different project for EDK2 APL implementation.  
Our team is not involved in that project and not really understand how the stitching was done for that project.  So it might be difficult for us to help here.

Just some of my thoughts. In this case, can you first stitch using whatever way they recommended.  Once it is done, then you can use FIT tool to open this IFWI image,  make modification as required and then generate new IFWI image again using FIT.

Regards,
Maurice

andreyv1978, 2020-07-17:
thank you for the recommendations Maurice,  it actually made  a trick. Unfortunately, not made SSC disable :( 
We decided to make board changes to support SSC enable and with hope everything else would work
So, I would use SBL to make all other peripherals work as needed . 
Closing ticket

closed 2020-07-17 by andreyv1978