slimbootloader/slimbootloader #783: LeafHill APL GPIO settings not identical with Intel GPIO Conf Tool

andreyv1978, 2020-07-21:
Hi

I am trying to match the GPIO settings from the Intel GPIO Configuration Tool and it seems a very hard
Initially I have .bin file generated from Intel GPIO Configuration Tool, but I am not being able to set everything in the SBL 
How can I replace internal LeafHill GPIO settings with my own ?
Are GPIO LeafHill settings set in CfgData_Gpio.dsc and mapped in Stage2B ?
I see that the eMMC set in Stage1, but would it be replaced later in the Stage2 ?

Andrey

rprangar, 2020-07-21:
Hi

If you hardware design is close to Leaf Hill CRB, then you can start with our GPIO settings for Leaf Hill and make the modifications for the changes in your hardware using ConfigEditor.py. That would be the straight forward way.

However, if you want to convert from an existing implementation, then you may have to write a conversion script. A reference is available here - https://github.com/slimbootloader/slimbootloader/blob/master/Platform/CommonBoardPkg/Tools/GpioDataConvert.py

Thanks.

andreyv1978, 2020-07-22:
I see, there is no easy way to do that
I do not know how close my board design to Leafhill regarding to GPIO. The .bin I have is a full GPIO table for all settings.

rprangar, 2020-07-23:
Hi

We will see if we can add support for the GpioDataConvert tool to accept the output from the GPIO configuration tool.

Thanks
Ravi

andreyv1978, 2020-07-23:
I've god a board design GPIO spec (much less to configure), but some setting are different between Intel GPIO tool and SBL 
The SBL python configuration tool  has much more settings for each GPIO than an Intel GPIO tool

andreyv1978, 2020-07-23:
Just an example:
edk2 and SBL BXT_GPIO_PAD_CONF:
//
  //                  Group Pin#:  pad_name,       PMode,GPIO_Config,HostSw,GPO_STATE,INT_Trigger,Wake_Enabled, Term_H_L,Inverted,GPI_ROUT,IOSstae, IOSTerm,MMIO_Offset, Community
  //
BXT_GPIO_PAD_CONF(L"GPIO_34 PWM0",             M0   ,    Native   , GPIO_D,  NA    ,   NA       , Wake_Disabled, P_20K_L,   NA    ,    NA,     Last_Value   ,     SAME, GPIO_PADBAR+0x0110,  NORTH),//RSVD

Intel tool example of the same GPIO:
[image]

where GPIO_Config,HostSw,GPO_STATE are missing 

SBL python config tool:
[image]

andreyv1978, 2020-07-23:
In addition to that the file CfgData_Gpio.dsc have the set for the  DW0 Register 0:
  # !BSF SUBT:{GPIO_TMPL:GPIO_34                :0x000610c5     :0x44000101} --> this is loaded into python tool
but where DW1 Register 1 settings ? 
The Intel GPIO tool have the settings like:
0x610     0x44000101          0x614     0x0000103f     /*GPIO_NAME = PWM0*/

Is it something missing in  my understanding ?

mauricema, 2020-07-23:
Yes, you are right.  The GPIO configuration tool does not provide all info required for GPIO initialization as you noticed, such as HostSw, WakeEnable, etc.  And that is why we recommend to use ConfigEditor.py to configure the GPIO for SBL.

In SBL, it has GPIO DW0/DW1 values in DSC file.  But these values are not exactly the same as the DW0 and DW1 defined in the GPIO register.  It needs to be translated accordingly.  But as mentioned,  even doing the translation, it will not provide full info required because some information will be stored in other APL SOC registers instead of the GPIO DW0 and DW1.

mauricema, 2020-07-23:
For you convenience,  I wrote a python script to help convert the GPIO configuration tool output into SBL DSC or DLT format.  I will provide basic GPIO info matching the table you have.  However the missing info cannot be retrieved as explained above.

The script is attached here.  To use this script, you need to rename the script from txt to py. And then you can follow the usage info to run the script.  
If you are doing a new board GPIO configuration, it is not recommended to change anything inside your CfgDataDef.dsc or CfgData_Gpio.dsc.  Instead, you should put other in the dlt file.
This script can also help you generate dlt format if you specify a output file with '.dlt' extension. 
Something like below:
  GPIO_CFG_DATA.GPIO_0_Half0               | 0x000500C5
  GPIO_CFG_DATA.GPIO_0_Half1               | 0x1D008008
You can take a reference into  CfgData_Ext_MB3.dlt.

[apl_gpio_conv.txt](https://github.com/slimbootloader/slimbootloader/files/4967736/apl_gpio_conv.txt)

andreyv1978, 2020-07-24:
The tool is brilliant, thank you

closed 2020-07-24 by andreyv1978