slimbootloader/slimbootloader #785: eMMC mapping and boot failed in APL LeafHill

andreyv1978, 2020-07-23:
Hi

I am trying to configure our custom design board based LeafHill APL. The eMMC GPIO settings added in  EarlyBootDeviceInit Stage1BBoardInitLib in form of an array, table like:
BXT_GPIO_PAD_INIT  mGpioSqx2InitTblEMMC[] = {

  BXT_GPIO_PAD_CONF(L"GPIO_156 EMMC0_CLK",         M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_20K_L,   NA   ,    NA  ,D0RxDRx0I,SAME  ,   GPIO_PADBAR+0x0020 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_157 EMMC0_D0",          M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0028 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_158 EMMC0_D1",          M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0030 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_159 EMMC0_D2",          M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0038 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_160 EMMC0_D3",          M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0040 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_161 EMMC0_D4",          M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0048 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_162 EMMC0_D5",          M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0050 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_163 EMMC0_D6",          M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0058 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_164 EMMC0_D7",          M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0060 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_165 EMMC0_CMD",         M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_NONE,    NA   ,    NA  ,HizRx1I  ,SAME  ,   GPIO_PADBAR+0x0068 ,  SOUTHWEST),
  BXT_GPIO_PAD_CONF(L"GPIO_182 EMMC0_STROBE",      M1,     NA    ,   NA   ,  NA     ,   NA       ,Wake_Disabled, P_20K_L,   NA   ,    NA  ,HizRx0I  ,SAME  ,   GPIO_PADBAR+0x00E8 ,  SOUTHWEST),
};

Also the same table inited in GpioInit: Stage2BoardInitLib at the end of the GPIO global sets

Unfortunately, the eMMC is not able to boot, log file is attached
[sbl-logging.log](https://github.com/slimbootloader/slimbootloader/files/4966181/sbl-logging.log)

I am able to see BLK0-5 mapped in the shell, but they not accessible 
I've not configured anything with ConfigEditor for GPIO, because seen you need eMMC early setup in PreConfigInit phase
What am I doing wrong ?

Maybe errors can tell what is wrong?
Error reported by SDHCI
Interrupt status = 8001
Error interrupt status = A
TRB failed with CRC Error
SdCardIdentification: Executing Cmd8 fails with Device Error

Andrey

mauricema, 2020-07-23:
From your log, the eMMC seems to be initialized properly.  And it even detected partitions on eMMC.
Since you did not see FS0/FS1/etc,  I assume the partition has not been formatted into FAT format yet.  You can try to boot into Linux using USB, then format eMMC partition to FAT filesystem.
If no valid filesystem exists, you should still be able to do block I/O access.   You should be able to use "dblk" command to read the raw data from eMMC.  EX:  dblk blk1

andreyv1978, 2020-07-27:
I am a bit concerned of how my settings are working:
1. eMMC mapped with no drive letters is allocated only when I have USB stick (eMMC has Ubuntu and works well with old Phoenix BIOS)
2. if no USB stick, eMMC can be not mapped at all (in rare cases after a few restart eMMC appears as mapped)
3. I have only a single USB port from 5 working, only a first is working, do you know why ?

I still trying to boot a minimal yocto image, but with a lot of crashes, while booting. I am not be able to get to the logging because the board is going OFF, therefore not be able to format eMMC
I remember I've managed to boot the same image a few month ago, would try to go back to my previous old checkouts

andreyv1978, 2020-07-27:
adding the boot log:
the last line is crashing a board (powering it OFF)
[boot-log.log](https://github.com/slimbootloader/slimbootloader/files/4981980/boot-log.log)

mauricema, 2020-07-27:
From your log, even Linux has detected eMMC.  So I guess your eMMC is accessible.
mmcblk1boot0: mmc1:0001 8GME4R partition 1 4.00 MiB
mmcblk1boot1: mmc1:0001 8GME4R partition 2 4.00 MiB
mmcblk1rpmb: mmc1:0001 8GME4R partition 3 512 KiB, chardev (245:0)
 mmcblk1: p1 p2 < p5 >

For USB,  the only thing I can think of is the GPIO related configuration and HSIO configuration.  
In the USB related FSP configuration, you also need to ensure the ports are all enabled.

For board carshing power off,  I have no idea on what went wrong.  Since it is a customized board,  it might be hard for us to provide useful info here.

andreyv1978, 2020-07-28:
i see the emmc is allocated, but OS boot is failing 
closing this, because board crashing is a different issue

closed 2020-07-28 by andreyv1978