slimbootloader/slimbootloader #786: [APL] Yocto boot kernel report indicates broken DMAR tables on LeafHill custom board

andreyv1978, 2020-07-29:
Hi

I am not a specialist in DMAR tables and VT-d settings. Unfortunately, the yocto boot has a following kernel report
when VT-d settings enabled:
 BIOS vendor: Intel Corporation; Ver: SBL:001.000.005.001.000.19188.D-7C208B4421975D06-dirty; Product Version: 0.1
WARNING: CPU: 0 PID: 0 at drivers/iommu/dmar.c:848 warn_invalid_dmar.part.10+0x5c/0x70
Modules linked in:
CPU: 0 PID: 0 Comm: swapper Tainted: G          I       4.19.80-intel-pk-standard #1
Hardware name: Intel Corporation SQX2 Client Platform/SQX2 Board, BIOS SBL:001.000.005.001.000.19188.D-7C208B4421975D06-dirty Unknown date
RIP: 0010:warn_invalid_dmar.part.10+0x5c/0x70
Code: 2a 00 bf 01 00 00 00 49 89 c5 e8 2f a2 2a 00 4d 89 f1 4d 89 e8 4c 89 e2 48 89 c1 48 89 de 48 c7 c7 08 f1 45 95 e8 a6 95 a4 ff <0f> 0b 5b 41 5c 41 5d 41 5e 5d c3 66 0f 1f 84 00 00 00 00 00 e8 ab
RSP: 0000:ffffffff95603db0 EFLAGS: 00010082
RAX: 0000000000000000 RBX: 00000000fed64000 RCX: 0000000000000000
RDX: 0000000000000000 RSI: 0000000000000082 RDI: 00000000ffffffff
RBP: ffffffff95603dd0 R08: 0000000000000081 R09: 2e30203a6e6f6973
R10: 7372655620746375 R11: 312e30203a6e6f69 R12: ffffffff9545fa3e
R13: ffffffff95a00014 R14: ffffffff95a00088 R15: ffffffffff240240
FS:  0000000000000000(0000) GS:ffff9d5e3ba00000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: ffff9d5e3ffff000 CR3: 000000002660a000 CR4: 00000000000406b0
Call Trace:
 dmar_validate_one_drhd+0xb2/0xd0
 dmar_walk_remapping_entries+0xae/0x150
 detect_intel_iommu+0x66/0xe8
 ? clkdev_alloc+0x90/0x90
 pci_iommu_alloc+0x55/0x71
 mem_init+0xe/0x3b
 start_kernel+0x29c/0x505
 x86_64_start_reservations+0x24/0x26
 x86_64_start_kernel+0x6f/0x72
 secondary_startup_64+0xa4/0xb0
---[ end trace 9c2e703a09e35333 ]---

when VT-d settings disabled:
DMAR: IOMMU enabled
Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)
Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)
------------[ cut here ]------------
Your BIOS is broken; DMAR reported at address 0!
BIOS vendor: Intel Corporation; Ver: SBL:001.000.005.001.000.19238.D-CFB1CB913323B7E8-dirty; Product Version: 0.1
WARNING: CPU: 0 PID: 0 at drivers/iommu/dmar.c:848 warn_invalid_dmar.part.10+0x5c/0x70
Modules linked in:
CPU: 0 PID: 0 Comm: swapper Tainted: G          I       4.19.80-intel-pk-standard #1
Hardware name: Intel Corporation SQX2 Client Platform/SQX2 Board, BIOS SBL:001.000.005.001.000.19238.D-CFB1CB913323B7E8-dirty Unknown date
RIP: 0010:warn_invalid_dmar.part.10+0x5c/0x70
Code: 2a 00 bf 01 00 00 00 49 89 c5 e8 2f a2 2a 00 4d 89 f1 4d 89 e8 4c 89 e2 48 89 c1 48 89 de 48 c7 c7 08 f1 65 b2 e8 a6 95 a4 ff <0f> 0b 5b 41 5c 41 5d 41 5e 5d c3 66 0f 1f 84 00 00 00 00 00 e8 ab
RSP: 0000:ffffffffb2803db0 EFLAGS: 00010082
RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
RDX: 0000000000000000 RSI: 0000000000000082 RDI: 00000000ffffffff
RBP: ffffffffb2803dd0 R08: 0000000000000084 R09: 746e49203a726f64
R10: 646e657620534f49 R11: 65746e49203a726f R12: ffffffffb2611736
R13: ffffffffb2c00014 R14: ffffffffb2c00088 R15: ffffffffff240240
FS:  0000000000000000(0000) GS:ffff9de33ba00000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: ffff9de31e401000 CR3: 000000015e00a000 CR4: 00000000000406b0
Call Trace:
 dmar_validate_one_drhd+0xc7/0xd0
 dmar_walk_remapping_entries+0xae/0x150
 detect_intel_iommu+0x66/0xe8
 ? clkdev_alloc+0x90/0x90
 pci_iommu_alloc+0x55/0x71
 mem_init+0xe/0x3b
 start_kernel+0x29c/0x505
 x86_64_start_reservations+0x24/0x26
 x86_64_start_kernel+0x6f/0x72
 secondary_startup_64+0xa4/0xb0
---[ end trace 7fb68b17d85a6fd2 ]---

We have a custom board design based on LeafHill APL.
Can you please advise what should I change in SBL settings to eliminate this report ?

Attaching a full boot include Yocto. the kernel report is on line 2180
[boot-debug-vtd.log](https://github.com/slimbootloader/slimbootloader/files/4995294/boot-debug-vtd.log)

mauricema, 2020-07-30:
You will need to check your DMAR ACPI table.  Kernel reported some entry has address 0.
From your log,   I did not see any  DMAR table update message from SBL.   Can you check if function UpdateDmarAcpi() is called in your flow ?  It needs to be called for VT-d support.   I did see the message on Leafhill.

>

andreyv1978, 2020-07-30:
Do you mean AcpiTables->Dmar.aslc ? I am not changing it, using from SBL

andreyv1978, 2020-07-30:
'DMAR table update' message only when I am setting (self.ENABLE_VTD               = 1 ) in BoardConfig.py
Unfortunately, the kernel prints report like:
Your BIOS is broken; DMAR reported at address fed64000 returns all ones!
BIOS vendor: Intel Corporation; Ver: SBL:001.000.005.001.000.19263.D-7B497F6975FE0B44-dirty; Product Version: 0.1
WARNING: CPU: 0 PID: 0 at drivers/iommu/dmar.c:848 warn_invalid_dmar.part.10+0x5c/0x70
Modules linked in:
CPU: 0 PID: 0 Comm: swapper Tainted: G          I       4.19.80-intel-pk-standard #1
Hardware name: Intel Corporation SQX2 Client Platform/SQX2 Board, BIOS SBL:001.000.005.001.000.19263.D-7B497F6975FE0B44-dirty Unknown date
RIP: 0010:warn_invalid_dmar.part.10+0x5c/0x70
Code: 2a 00 bf 01 00 00 00 49 89 c5 e8 2f a2 2a 00 4d 89 f1 4d 89 e8 4c 89 e2 48 89 c1 48 89 de 48 c7 c7 08 f1 85 ac e8 a6 95 a4 ff <0f> 0b 5b 41 5c 41 5d 41 5e 5d c3 66 0f 1f 84 00 00 00 00 00 e8 ab
RSP: 0000:ffffffffaca03db0 EFLAGS: 00010082
RAX: 0000000000000000 RBX: 00000000fed64000 RCX: 0000000000000000
RDX: 0000000000000000 RSI: 0000000000000082 RDI: 00000000ffffffff
RBP: ffffffffaca03dd0 R08: 0000000000000081 R09: 2e30203a6e6f6973
R10: 7372655620746375 R11: 312e30203a6e6f69 R12: ffffffffac85fa3e
R13: fffffffface00014 R14: fffffffface00088 R15: ffffffffff240240
FS:  0000000000000000(0000) GS:ffff9f877ba00000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: ffff9f877ffff000 CR3: 000000003920a000 CR4: 00000000000406b0
Call Trace:
 dmar_validate_one_drhd+0xb2/0xd0
 dmar_walk_remapping_entries+0xae/0x150
 detect_intel_iommu+0x66/0xe8
 ? clkdev_alloc+0x90/0x90
 pci_iommu_alloc+0x55/0x71
 mem_init+0xe/0x3b
 start_kernel+0x29c/0x505
 x86_64_start_reservations+0x24/0x26
 x86_64_start_kernel+0x6f/0x72
 secondary_startup_64+0xa4/0xb0
---[ end trace b3efb11d8d7e2480 ]---

and later:
DMAR: Host address width 39
DMAR: DRHD base: 0x000000fed64000 flags: 0x0
DMAR: Failed to map dmar0
DMAR: Parse DMAR table failure.

I want to disable VTD

mauricema, 2020-07-30:
Loosk like when VT-d is enabled, some DMA remapping engine was disabled, so it shows all ones.  Is your silicon fused differently ? or a different stepping ? 
To disable VT-d, you will need to set self.ENABLE_VTD = 0 in BoardConfig.py, and then remove DMAR ACPI table.  
To remove the DMAR table, you can change the code in PlatformUpdateAcpiTable() to something like below:
if (FeaturePcdGet (PcdVtdEnabled)) {
    if (Table->Signature == EFI_ACPI_VTD_DMAR_TABLE_SIGNATURE) {
      PlatformData  = (PLATFORM_DATA *)GetPlatformDataPtr ();
      if (PlatformData->RmrrUsbAddress != 0) {
        Dmar = (EFI_ACPI_DMAR_TABLE *)Table;
        Dmar->RmrrHeci.RmrBaseAddress  = PlatformData->RmrrUsbAddress;
        Dmar->RmrrHeci.RmrLimitAddress = Dmar->RmrrHeci.RmrBaseAddress + VTD_RMRR_USB_LENGTH - 1;
      }
      UpdateDmarAcpi (Table);
    }
} else {
   Status = EFI_UNSUPPORTED;
}

andreyv1978, 2020-07-31:
the addition of EFI_UNSUPPORTED makes the STAGE_2 stuck with a following:
ASSERT_EFI_ERROR (Status = Unsupported)
ASSERT [Stage2] d:\vmshared\projects\sbl\git\sbl\Platform\ApollolakeBoardPkg\Library\Stage2BoardInitLib\Stage2BoardInitLib.c(2183): !EFI_ERROR (Status)

STAGE_2: System halted!

andreyv1978, 2020-07-31:
Just made a boot with old BIOS from Phoenix they used EDK2 and there is no DRAM table at all when VT-d disable:
ACPI: RSDP 0x000000007977F014 000024 (v02 PTEC  )
ACPI: XSDT 0x0000000079755188 0000F4 (v01 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: FACP 0x0000000079778000 00010C (v05 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI BIOS Warning (bug): 32/64X length mismatch in FADT/Pm1aEventBlock: 32/16 (20180810/tbfadt-569)
ACPI BIOS Warning (bug): Invalid length for FADT/Pm1aEventBlock: 16, using default 32 (20180810/tbfadt-674)
ACPI: DSDT 0x0000000079766000 00A0CF (v02 PTEC   EDK2     00000003 BRXT 0100000D)
ACPI: FACS 0x0000000079702000 000040
ACPI: UEFI 0x0000000079706000 000042 (v01 PTEC   OEMTABLE 00000002 PTEC 00000002)
ACPI: BDAT 0x000000007977D000 000030 (v02 PTEC   OEMTABLE 00000000 PTEC 00000002)
ACPI: PSDS 0x000000007977C000 00004D (v01 PTEC   OEMTABLE 00000005 PTEC 00000002)
ACPI: DBG2 0x000000007977B000 000072 (v00 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: DBGP 0x000000007977A000 000034 (v01 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: HPET 0x0000000079777000 000038 (v01 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: LPIT 0x0000000079776000 00005C (v01 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: APIC 0x0000000079775000 000084 (v03 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: MCFG 0x0000000079774000 00003C (v01 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: NPKT 0x0000000079773000 000065 (v01 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: PRAM 0x0000000079772000 000030 (v01 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: WSMT 0x0000000079771000 000028 (v01 PTEC   OEMTABLE 00000003 PTEC 00000002)
ACPI: SSDT 0x000000007975C000 004D38 (v02 PTEC   RVPRtd3  00000003 BRXT 0100000D)
ACPI: SSDT 0x000000007975B000 000B53 (v02 PTEC   UsbCTabl 00000003 BRXT 0100000D)
ACPI: SSDT 0x0000000079759000 0018EB (v01 PTEC   Platform 00001000 INTL 20150818)
ACPI: BATB 0x0000000079758000 00004A (v02 PTEC   OEMTABLE 00000000 PTEC 00000002)
ACPI: SSDT 0x0000000079757000 00072B (v02 PTEC   CpuSsdt  00003000 INTL 20150818)
ACPI: SSDT 0x0000000079756000 00032D (v02 PTEC   Cpu0Tst  00003000 INTL 20150818)
ACPI: SSDT 0x000000007977E000 00017C (v02 PTEC   ApTst    00003000 INTL 20150818)
ACPI: POAT 0x0000000079754000 000055 (v03 PTEC   OEMTABLE 00000000 PTEC 00000002)
ACPI: SSDT 0x0000000079751000 002760 (v02 PTEC   SaSsdt   00003000 INTL 20150818)
ACPI: FPDT 0x0000000079750000 000044 (v01 PTEC   OEMTABLE 00000002 PTEC 00000002)
ACPI: BGRT 0x000000007974F000 000038 (v01 PTEC   OEMTABLE 00000002 PTEC 00000002)
ACPI: WDAT 0x0000000079779000 000104 (v01 PTEC   OEMTABLE 00000000 PTEC 00000002)
ACPI: UEFI 0x00000000796FC000 00013E (v01 PTEC   OEMTABLE 00000001 PTEC 00000002)

In SBL boot, with VT-d disable self.ENABLE_VTD = 0:
CPI: RSDP 0x000000007A004000 000024 (v02 OEMID )
ACPI: XSDT 0x000000007A0040E0 00006C (v01 OEMID  OEMTABLE 00000005 CREA 0100000D)
ACPI: DMAR 0x000000007A004210 0000B8 (v01 INTEL  Slimboot 00000001 INTL 00000001)
ACPI: FACP 0x000000007A0042D0 00010C (v05 OEMID  OEMTABLE 00000005 CREA 0100000D)
ACPI BIOS Warning (bug): 32/64X length mismatch in FADT/Pm1aEventBlock: 32/16 (20180810/tbfadt-569)
ACPI BIOS Warning (bug): Invalid length for FADT/Pm1aEventBlock: 16, using default 32 (20180810/tbfadt-674)
ACPI: DSDT 0x000000007A004610 005C3B (v02 INTEL  BXT-SOC  00000000 INTL 20160422)
ACPI: FACS 0x000000007A0043E0 000040
ACPI: FACS 0x000000007A0043E0 000040
ACPI: HPET 0x000000007A004420 000038 (v01 OEMID  OEMTABLE 00000005 CREA 0100000D)
ACPI: APIC 0x000000007A004460 000072 (v03                 00000000      00000000)
ACPI: MCFG 0x000000007A0044E0 00003C (v01                 00000001      00000000)
ACPI: FPDT 0x000000007A004520 000044 (v01 INTEL  OEMTABLE 00000005 CREA 0100000D)
ACPI: TPM2 0x000000007A0045C0 00004C (v04 INTEL  OEMTABLE 00000005 CREA 0100000D)
ACPI: SSDT 0x000000007A00A250 0003BE (v02 INTEL  Tpm2Tabl 00001000 INTL 20160422)
ACPI: NHLT 0x000000007A00A610 00040E (v00 INTEL  NHLT     00000002 INTL 01000013)

mauricema, 2020-07-31:
Sorry, I did not paste the right code.    VT-d disable is an issue in the SBL code.
I have submitted a PR to address it.  
https://github.com/slimbootloader/slimbootloader/pull/789
Please use that patch.

andreyv1978, 2020-08-03:
sorted, thank you for the patch

closed 2020-08-03 by andreyv1978